gtxaspec/wz_mini_hacks #2: cannot create directory ‘./tmp_root/4.36.8.32_ext

Vendo232, 2022-04-04:
Hi
I`m very exited to try your hack! trying to compile the SD card FW part using Windows 10 and GIT installed and the code gives me this error.

I will try later today to run this in VM Ubuntu

`Copy locla utilities to SD_ROOT
Download latest rootfs firmware 4.36.3.19
./setup.sh: line 27: wget: command not found
mkdir: cannot create directory ‘./tmp_root/4.36.3.19_ext’: No such file or directory
tar: ./tmp_root/4.36.3.19.tar: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
Download latest stable firmware 4.36.8.32
mkdir: cannot create directory ‘./tmp_root/4.36.8.32_ext’: No such file or directory
./setup.sh: line 34: wget: command not found
tar: ./tmp_root/4.36.8.32.tar: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
Extract rootfs, prepare for modification
./setup.sh: line 38: unsquashfs: command not found
mkdir: cannot create directory ‘./tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/.ssh’: No such file or directory
Change unknown stock password to WYom2020 in /etc/shadow
./setup.sh: line 46: ./tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/etc/shadow: No such file or directory
chmod: cannot access './tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/etc/shadow': No such file or directory
Add run_mmc.sh to rootfs rcS init.d script
sed: can't read ./tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/etc/init.d/rcS: No such file or directory
repack rootfs for flashing
./setup.sh: line 53: mksquashfs: command not found
copy latest appfs and kernel to SD_ROOT
cp: cannot stat './tmp_root/4.36.8.32_ext/Upgrade/app': No such file or directory
cp: cannot stat './tmp_root/4.36.8.32_ext/Upgrade/kernel': No such file or directory
extract run_mmc.sh script to SD_ROOT

`

[image]

Vendo232, 2022-04-04:
I was able to install Wget by installing Chocolatey and following this guide [https://www.youtube.com/watch?v=wm72ToyK34Q]

then got this issue

`2022-04-04 10:34:33 (13.9 MB/s) - './tmp_root/4.36.8.32.tar' saved [5734400/5734400]

Extract rootfs, prepare for modification
./setup.sh: line 38: unsquashfs: command not found
mkdir: cannot create directory ‘./tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/.ssh’: No such file or directory
Change unknown stock password to WYom2020 in /etc/shadow
./setup.sh: line 46: ./tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/etc/shadow: No such file or directory
chmod: cannot access './tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/etc/shadow': No such file or directory
Add run_mmc.sh to rootfs rcS init.d script
sed: can't read ./tmp_root/4.36.3.19_ext/Upgrade/rootfs_dir/etc/init.d/rcS: No such file or directory
repack rootfs for flashing
./setup.sh: line 53: mksquashfs: command not found
copy latest appfs and kernel to SD_ROOT
extract run_mmc.sh script to SD_ROOT
`

gtxaspec, 2022-04-04:
I will add a prerequisites section, there are some utilities required before execution.

Vendo232, 2022-04-04:
here is the log from Linux/Ubuntu it seems it went OK, when I put files on SD nothing happens.
Should I put tmp_root folder on SD card?

`server@server-Standard-PC-i440FX-PIIX-1996:~/Downloads/wz_mini_hacks-master$ ./setup.sh compile
Create SD Card root directory
Copy test to SD_ROOT
Download utilities to SD_ROOT
--2022-04-04 16:23:17--  https://www.busybox.net/downloads/binaries/1.21.1/busybox-mipsel
Resolving www.busybox.net (www.busybox.net)... 140.211.167.122
Connecting to www.busybox.net (www.busybox.net)|140.211.167.122|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1576152 (1.5M)
Saving to: ‘SD_ROOT/busybox’

SD_ROOT/busybox     100%[===================>]   1.50M  6.31MB/s    in 0.2s    

2022-04-04 16:23:18 (6.31 MB/s) - ‘SD_ROOT/busybox’ saved [1576152/1576152]

Copy locla utilities to SD_ROOT
Download latest rootfs firmware 4.36.3.19
--2022-04-04 16:23:18--  https://s3-us-west-2.amazonaws.com/wuv2/upgrade/WYZE_CAKP2JFUS/firmware/4.36.3.19.tar
Resolving s3-us-west-2.amazonaws.com (s3-us-west-2.amazonaws.com)... 52.218.170.88
Connecting to s3-us-west-2.amazonaws.com (s3-us-west-2.amazonaws.com)|52.218.170.88|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5662720 (5.4M) [application/octet-stream]
Saving to: ‘./tmp_root/4.36.3.19.tar’

4.36.3.19.tar       100%[===================>]   5.40M  12.8MB/s    in 0.4s    

2022-04-04 16:23:18 (12.8 MB/s) - ‘./tmp_root/4.36.3.19.tar’ saved [5662720/5662720]

Download latest stable firmware 4.36.8.32
--2022-04-04 16:23:18--  https://s3-us-west-2.amazonaws.com/wuv2/upgrade/WYZE_CAKP2JFUS/firmware/4.36.8.32.tar
Resolving s3-us-west-2.amazonaws.com (s3-us-west-2.amazonaws.com)... 52.218.170.88
Connecting to s3-us-west-2.amazonaws.com (s3-us-west-2.amazonaws.com)|52.218.170.88|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5734400 (5.5M) [application/octet-stream]
Saving to: ‘./tmp_root/4.36.8.32.tar’

4.36.8.32.tar       100%[===================>]   5.47M  13.8MB/s    in 0.4s    

2022-04-04 16:23:19 (13.8 MB/s) - ‘./tmp_root/4.36.8.32.tar’ saved [5734400/5734400]

Extract rootfs, prepare for modification
Parallel unsquashfs: Using 1 processor
351 inodes (423 blocks) to write

[===============================================================/] 423/423 100%

created 64 files
created 33 directories
created 287 symlinks
created 0 devices
created 0 fifos
Change unknown stock password to WYom2020 in /etc/shadow
Add run_mmc.sh to rootfs rcS init.d script
repack rootfs for flashing
Parallel mksquashfs: Using 1 processor
Creating 4.0 filesystem on SD_ROOT/rootfs2.bin, block size 131072.
[===============================================================|] 136/136 100%

Exportable Squashfs 4.0 filesystem, xz compressed, data block size 131072
	compressed data, compressed metadata, compressed fragments,
	compressed xattrs, compressed ids
	duplicates are removed
Filesystem size 3763.45 Kbytes (3.68 Mbytes)
	32.94% of uncompressed filesystem size (11424.94 Kbytes)
Inode table size 2188 bytes (2.14 Kbytes)
	15.36% of uncompressed inode table size (14244 bytes)
Directory table size 3238 bytes (3.16 Kbytes)
	52.98% of uncompressed directory table size (6112 bytes)
Number of duplicate files found 2
Number of inodes 385
Number of files 64
Number of fragments 14
Number of symbolic links  287
Number of device nodes 0
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 34
Number of ids (unique uids + gids) 1
Number of uids 1
	root (0)
Number of gids 1
	root (0)
copy latest appfs and kernel to SD_ROOT
extract run_mmc.sh script to SD_ROOT
server@server-Standard-PC-i440FX-PIIX-1996:~/Downloads/wz_mini_hacks-master$
`

Vendo232, 2022-04-04:
these are the files in SD_ROOT

[image]

Vendo232, 2022-04-04:
tried several SD card, cards I know work with V3 and nothing happens

gtxaspec, 2022-04-05:
I may not have been clear during the last update. turn camera off. insert memory card and then plug camera in.  let me know

Vendo232, 2022-04-05:
> I may not have been clear during the last update. turn camera off. insert memory card and then plug camera in. let me know

Hello, that is what I have been doing, Copy files on SD Card, turn off V3, insert SD Card and turn on V3.

the cam does something but after ~5 min it goes back online . FW remains the same 4.36.0.228

Vendo232, 2022-04-05:
one more suggestion. The compilation script is design to pull latest FW for future. Could you perhaps create and share the SD_ROOT files now/today to avoid the compilation part?

gtxaspec, 2022-04-05:
I will flash the same FW version you have and test and get back to you

Vendo232, 2022-04-05:
> @Vendo232 just flashed the firmware you have, and git cloned the repo, and install works fine for me... tried on two different cameras...perhaps i could recommend re-flashing .228 firmware and trying again?

after you flashed the lower FW, did you install wyzehack using DNS spoofing?

gtxaspec, 2022-04-05:
No I did not, as DNS spoofing is not required for this.

Vendo232, 2022-04-05:
> No I did not, as DNS spoofing is not required for thi

Vendo232, 2022-04-05:
So you do not need telnet prior your hack? then the steps would be:

1. downgrade to 4.36.0.228
2. git clone your repo
3. in Linux OS run ./setup.sh compile
4. copy SD_ROOT content on SD Card
5. tun off V3 , insert the SD card and turn it ON.
6. wait for FW update until LED blinks rapidly
7. SSH to the V3

correct?

closed 2022-04-05 by Vendo232

Vendo232, 2022-04-05:
> right!

my use case for this is to install the HTTP stream hack which allows me to use V3 in Frigate w/o any issue the RTSP FW is not working well.

it would help me a lot if you could try installing this http hack on the latest FW via SSH and see if the installation works.

`cd /configs
wget crb.users.sonic.net/current/install.sh
chmod +x install.sh
./install.sh
You should see msg “ Installed Successfully”`

https://github.com/FiveLeavesLeft/WyzeCameraLiveStream

it would be awesome if it works

gtxaspec, 2022-04-05:
I will test this for you shortly

gtxaspec, 2022-04-05:
It does install, but @fiveleavesleft may have to do an update, as it looks like he based it off wyzehacks, and wz_mini_hacks currently doesn't work wyzehacks at the moment.

Vendo232, 2022-04-05:
awesome! thank you the test, so you got this msg “Installed Successfully”

gtxaspec, 2022-04-05:
`[root@WCV3_spare_test:configs]# ./install.sh 
Connecting to crb.users.sonic.net (209.204.190.65:80)
saving to '/configs/stream_hack/libimp.so'
'/configs/stream_hack/libimp.so' saved
Moving /configs/wyze_hack.sh to /configs/wyze_hack.sh.orig
Stream hack installed.
`

Vendo232, 2022-04-05:
do you need to hold SETUP button when you enter the SD card with the mini hacks to V3?

gtxaspec, 2022-04-05:
No, at boot time the camera reads Test.tar and executes the scripts to initialize installation.  Do you still have telnet access?

Vendo232, 2022-04-05:
it WORKED!! I can SSH into the V3 with root access. GREAT JOB! 

unfortunately for me the stream hack installed but does not work :-(

gtxaspec, 2022-04-05:
hopefully the streamhack will be updated soon!

Vendo232, 2022-04-05:
just FYI, flashed the RTSP firmware 4.61.0.3 via standard SD card ( copied the demo_wcv3.bin on sd card ) left all your mini hack files on SD card. The FW was updated but SSH is gone

gtxaspec, 2022-04-05:
right, when you reflash the firmware manually, you have to re-install mini hacks

Vendo232, 2022-04-05:
I think some folks might appreciate having RTSP firmware + SSH root access. Not sure if that would be possible.

Vendo232, 2022-04-05:
that worked, but it will force standard FW 4.36.8.32

here is the link for the latest RTSP Firmware , probably the setup.sh will require adjustment for RTSP branch
https://download.wyzecam.com/firmware/rtsp/demo_v3_RTSP_4.61.0.1.zip

gtxaspec, 2022-04-05:
ah, i forgot about the rtsp, i will make a new version that will let you select RTSP or not.

FiveLeavesLeft, 2022-04-05:
I am watching this with much interest but I am really busy with other things and don't know if I will have time to update the stream hack. I did add it to my list of things to do and I think what is happening here is really cool and thanks for all the work.