gtxaspec/wz_mini_hacks #4: ssh root@<deviceIP> -> Connection closed by <deviceIP> port 22

evanheckert, 2022-04-13:
First of all, HUGE thanks for your efforts here. Excited about the potential!

Device Model: `Wyze Cam v3`
MAC: `7C7...`
Firmware Version: `4.61.0.3` (Factory RTSP firmware)
Plugin Version: `2.29.7`

```
> ssh root@<deviceIP>
Connection closed by <deviceIP> port 22
```

Did a git clone today (4/13 at 3pm CST) and copied contents of SD ROOT to the root of SD card. 

I didn't wipe the card first - had WyzeHacks on the same firmware prior to a factory reset. Attempted this hack after the factory reset.

The camera booted right up - didn't seem to take more time to install anything. RTSP at the <deviceIP> is functioning. All signs seem to point to the auto-installation not happening.

gtxaspec, 2022-04-13:
is there a log inside of the wz_mini_hacks/log directory?

evanheckert, 2022-04-13:
Thanks for the reply! Here's the result:

wz_mini/wz_mini_hacks.log
```
udhcpc (v1.22.1) started
Sending discover...
Sending discover...
Sending discover...
Sending select for 192.168.1.184...
Lease of 192.168.1.184 obtained, lease time 6504
deleting routers
adding dns 192.168.1.1
adding dns 1.1.1.1
usb direct disabled
set hostname
Run dropbear ssh server
```

May I suggest inserting a timestamp to prefix each line of the log?

gtxaspec, 2022-04-13:
thanks for the log, i figured out the bug.  pull the latest git and you should be fine now.

evanheckert, 2022-04-13:
Fixed! Fantastic, thank you. Few interesting things I observed:

* With ethernet enabled, I could no longer ssh via wifi, only using the ip I assigned to the ethernet MAC address
    * this isn't a problem, just an observation. is wifi disabled then?
* The factory RTSP is using the ethernet IP for its RTSP URL. 
    * This is great news, as enabling ethernet with wyze hacks would still result in wifi RTSP unless wifi was killed altogether. 
* `ifconfig` still only shows `lo` and `wlan0`
    * When I'd enabled using wyze-hacks, it caused `eth0` to show up there. I do see that the video feed is causing the wlan0 TX bytes to go up, so I guess wlan0 is remapped to use the ethernet in place of wifi?

You mentioned wifi was needed for setup - does this mean every time it boots, or will the ethernet configuration persist across boots? Looking to use it in car without wifi, streaming display over ethernet via RTSP.

Thanks again! Feel free to close issue, just didn't know where else to put the question.

gtxaspec, 2022-04-13:
1: right, wifi is disabled now
2: yep
3: in order to trick iCamera into not panicking over the wifi connection, we rename wlan0 to wlan0old and eth0 to wlan0, wlan0 is actually eth0, just renamed.

wifi is only needed for the initial setup, registration with the wyze app. after that, if the wifi network is not present, and you have ENABLE_USB_ETH="true", the camera will work through the ethernet adapter just fine with no wifi at all.

evanheckert, 2022-04-13:
Can I buy you a coffee? Paypal, venmo, etc?

closed 2022-04-13 by evanheckert

gtxaspec, 2022-04-13:
it's the thought that counts! but seriously, these two guys from japan blew the whole thing wide open with their research and work, and this would NOT have been possible with out them!

 https://github.com/mnakada/atomcam_tools
 https://github.com/bakueikozo/atomcam_tools

if you can donate, donate to those fine folks!!

evanheckert, 2022-04-13:
So I'm curious if my idea is possible (let me know if somewhere else is more appropriate for idea discussion):

* Turn car on, powers router, camera, and tablet.
* Tablet connects to RTSP stream over ethernet to display in-car cam in real time
* Camera *also* records to SD card

I believe the above will now work, but here's the part I'm curious about:
* When I arrive home (geofence from tablet GPS?), connect to home wifi and upload footage to NAS via SMB/NFS and chime or flash camera LED when completed (so I can turn car off)

gtxaspec, 2022-04-13:
if the tablet travels with you, using something like tasker to detect a geofence, then set a script to connect to the camera via ssh, disable Ethernet, enable wifi,and copy the files to the nas through NFS.

haven't used tasker in a while, but it seems feasible.