gtxaspec/wz_mini_hacks #5: support for LAN - Realtek RTL8153

Vendo232, 2022-04-13:
adding this info just in case it might be possible to add support for Realtek RTL8153

Bus 001 Device 006: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Eth                                                                         ernet Adapter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8153 RTL8153 Gigabit Ethernet Adapter
  bcdDevice           30.00
  iManufacturer           1 Realtek
  iProduct                2 USB 10/100/1000 LAN
  iSerial                 6 000000
  bNumConfigurations      2
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              350mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0050
    bNumInterfaces          2
    bConfigurationValue     2
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              350mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      6 Ethernet Networking
      bInterfaceProtocol      0
      iInterface              5
      CDC Header:
        bcdCDC               1.10
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1
      CDC Ethernet:
        iMacAddress                      3 (??)
        bmEthernetStatistics    0x00000000
        wMaxSegmentSize               1514
        wNumberMCFilters            0x0000
        bNumberPowerFilters              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               8
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

gtxaspec, 2022-04-13:
sure,let me see.

gtxaspec, 2022-04-13:
the built in rtl8152 driver should support this already in the wz_mini_hacks kernel.

Vendo232, 2022-04-13:
can`t get it to work, I`m on RTSP 4.61.0.3, the can has this IP on WIFI 192.168.1.56, do you get the same IP ( MAC ) when you connect over LAN ?

ordered this guy in the meantime, it has 50% coupon btw : 
https://www.amazon.com/dp/B09H19TK6B?psc=1&smid=A3MSS5ZN97PBY3&ref_=chk_typ_imgToDp

gtxaspec, 2022-04-13:
if you can SSH into the camera, plug the usb adapter in, and then do:

dmesg

then
lsusb

then
ifconfig eth0

and paste the output here i can try to find out why it doesnt work.

Vendo232, 2022-04-13:
ha, I can not SSH, that might be the problem. Will try to re-flash it

Vendo232, 2022-04-14:
flashed standard FW: demo_wcv3_4.36.8.32
when ETH enabled via ENABLE_USB_ETH="true" the cam would not connect so no SSH
when ENABLE_USB_ETH="false" and the USB LAN connected it will connect via WIFI and here is the output of those commands:

`[root@WCV3_spare_test:~]# dmesg
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_swan_1.0__ (root@5997f1cc9594) (gcc version 4.7.4 (Buildroot 2016.02) ) #28 PREEMPT Wed Apr 13 06:47:11 UTC 2022
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:00521C25
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 0, n = 0, reg val = 0x081000ff
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 0, n = 0, reg val = 0x081000ff
[    0.000000] CCLK:1392MHz L2CLK:696Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 005c5000 @ 00010000 (usable)
[    0.000000]  memory: 0007b000 @ 005d5000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 05000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x04ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x04ffffff]
[    0.000000] On node 0 totalpages: 20480
[    0.000000] free_area_init_node: node 0, pgdat 805d1a00, node_mem_map 81000000
[    0.000000]   Normal zone: 160 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 20480 pages, LIFO batch:3
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 20320
[    0.000000] Kernel command line: console=ttyS1,115200n8 mem=80M@0x0 rmem=48M@0x5000000 rdinit=/init mtdparts=jz_sfc:256K(boot),1984K(kernel),3904K(rootfs),3904K(app),1984K(kback),3904K(aback),384K(cfg),64K(para)
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 72860k/81920k available (4584k kernel code, 9060k reserved, 1321k data, 492k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:358
[    0.000000] clockevents_config_and_register success.
[    0.000014] Calibrating delay loop... 1391.00 BogoMIPS (lpj=6955008)
[    0.087830] pid_max: default: 32768 minimum: 301
[    0.092689] Mount-cache hash table entries: 512
[    0.097606] Initializing cgroup subsys debug
[    0.101863] Initializing cgroup subsys freezer
[    0.107310] devtmpfs: initialized
[    0.111472] regulator-dummy: no parameters
[    0.115719] NET: Registered protocol family 16
[    0.131156] bio: create slab <bio-0> at 0
[    0.136649] jz-dma jz-dma: JZ SoC DMA initialized
[    0.141568] usbcore: registered new interface driver usbfs
[    0.147154] usbcore: registered new interface driver hub
[    0.152551] usbcore: registered new device driver usb
[    0.157718]  (null): set:249  hold:250 dev=100000000 h=500 l=500
[    0.163802] media: Linux media interface: v0.10
[    0.168369] Linux video capture interface: v2.00
[    0.173176] Advanced Linux Sound Architecture Driver Initialized.
[    0.180503] Switching to clocksource jz_clocksource
[    0.185431] cfg80211: Calling CRDA to update world regulatory domain
[    0.192481] dwc2 otg probe start
[    0.192506] jz-dwc2 jz-dwc2: cgu clk gate get error
[    0.197426] DWC IN OTG MODE
[    0.200823] dwc2 dwc2: Keep PHY ON
[... 143 lines cut ...]
[    7.755635] codec_codec_ctl: set sample rate...
[    7.755953] codec_codec_ctl: set device...
[    7.991897] codec_set_device: set device: speaker...
[    8.004132] SPEAKER CTL MODE3 !
[    9.395413] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   10.866262] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   13.399223] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   13.935578] RTL871X: start auth
[   13.948265] RTL871X: auth success, start assoc
[   13.957474] RTL871X: assoc success
[   13.961114] RTL871X: recv eapol packet
[   13.965100] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   13.977530] RTL871X: send eapol packet
[   14.001378] RTL871X: recv eapol packet
[   14.012094] RTL871X: send eapol packet
[   14.017752] RTL871X: set pairwise key camid:4, addr:14:cc:20:da:4b:89, kid:0, type:AES
[   14.028860] RTL871X: set group key camid:5, addr:14:cc:20:da:4b:89, kid:2, type:AES
[   32.242203] run_mmc.sh start
[root@WCV3_spare_test:~]# lsusb
Bus 001 Device 002: ID 0bda:8153
Bus 001 Device 001: ID 1d6b:0002
[root@WCV3_spare_test:~]# ifconfig eth0
ifconfig: eth0: error fetching interface information: Device not found
[root@WCV3_spare_test:~]#
`

gtxaspec, 2022-04-14:
is this dmesg log with your usb adapter connected to the camera?

Vendo232, 2022-04-14:
yes

Vendo232, 2022-04-14:
also, I`m using RTSP FW 4.61.0.3 and the SSH does not work

Vendo232, 2022-04-14:
tried the 4/14 version, but cam will not connect when ENABLE_USB_ETH="true"

[image]

closed 2022-04-14 by Vendo232