gtxaspec/wz_mini_hacks #6: USB Ethernet - How to connect

cessna, 2022-04-13:
Dumb question, how exactly are you connecting a USB Ethernet adaptor to the WyzeCam v3? Are you using like a hub to connect both power and the adapter to the single port in the back?

gtxaspec, 2022-04-13:
2 Pack Micro USB Host OTG Cable with Micro USB Power Compatible with Amazon Fire TV Stick, Compatible with LG HTC, Kindle Fire Tablets https://www.amazon.com/dp/B07S23NKK7

evanheckert, 2022-04-14:
FWIW, this is what I'm using if you don't already have a USB-A ethernet adapter. https://www.amazon.com/gp/product/B0876N3PMZ

I'm using the Wyze power source along side plain ethernet, but apparently you can use power over ethernet if you've got it.

cessna, 2022-04-14:
I was wondering about POE. 

Thanks!