gtxaspec/wz_mini_hacks #7: ASIX AX88179 USB 3.0 Gigibit Ethernet

Vendo232, 2022-04-14:
RTSP: 4.61.03
adapter:  ASIX AX88179 USB 3.0 Gigibit Ethernet
dmesg:

[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_swan_1.0__ (root@5997f1cc9594) (gcc version 4.7.4 (Buildroot 2016.02) ) #28 PREEMPT Wed Apr 13 06:47:11 UTC 2022
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:005214B5
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 0, n = 0, reg val = 0x081000ff
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 0, n = 0, reg val = 0x081000ff
[    0.000000] CCLK:1392MHz L2CLK:696Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 005c5000 @ 00010000 (usable)
[    0.000000]  memory: 0007b000 @ 005d5000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 05000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x04ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x04ffffff]
[    0.000000] On node 0 totalpages: 20480
[    0.000000] free_area_init_node: node 0, pgdat 805d1a00, node_mem_map 81000000
[    0.000000]   Normal zone: 160 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 20480 pages, LIFO batch:3
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 20320
[    0.000000] Kernel command line: console=ttyS1,115200n8 mem=80M@0x0 rmem=48M@0x5000000 rdinit=/init mtdparts=jz_sfc:256K(boot),1984K(kernel),3904K(rootfs),3904K(app),1984K(kback),3904K(aback),384K(cfg),64K(para)
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Memory: 72860k/81920k available (4584k kernel code, 9060k reserved, 1321k data, 492k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:358
[    0.000000] clockevents_config_and_register success.
[    0.000014] Calibrating delay loop... 1391.00 BogoMIPS (lpj=6955008)
[    0.087830] pid_max: default: 32768 minimum: 301
[    0.092689] Mount-cache hash table entries: 512
[    0.097608] Initializing cgroup subsys debug
[    0.101864] Initializing cgroup subsys freezer
[    0.107315] devtmpfs: initialized
[    0.111478] regulator-dummy: no parameters
[    0.115724] NET: Registered protocol family 16
[    0.131160] bio: create slab <bio-0> at 0
[    0.136652] jz-dma jz-dma: JZ SoC DMA initialized
[    0.141572] usbcore: registered new interface driver usbfs
[    0.147158] usbcore: registered new interface driver hub
[    0.152555] usbcore: registered new device driver usb
[    0.157723]  (null): set:249  hold:250 dev=100000000 h=500 l=500
[    0.163806] media: Linux media interface: v0.10
[    0.168374] Linux video capture interface: v2.00
[    0.173182] Advanced Linux Sound Architecture Driver Initialized.
[    0.180510] Switching to clocksource jz_clocksource
[    0.185438] cfg80211: Calling CRDA to update world regulatory domain
[    0.192489] dwc2 otg probe start
[    0.192514] jz-dwc2 jz-dwc2: cgu clk gate get error
[    0.197434] DWC IN OTG MODE
[    0.200830] dwc2 dwc2: Keep PHY ON
[    0.204208] dwc2 dwc2: Using Buffer DMA mode
[... 185 lines cut ...]
[   21.398193] RTL871X: set group key camid:5, addr:14:cc:20:da:4b:89, kid:1, type:AES
[   23.201986] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0001: -145
[   28.212632] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0001: -145
[   32.361749] run_mmc.sh start
[   33.241968] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0001: -145
[   38.252309] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0001: -145
[   43.261990] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0002: -145
[   48.281998] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0001: -145
[   53.292026] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0002: -145
[   58.301975] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to read reg index 0x0001: -145
[   63.311944] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0001: -145
[   68.331967] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to read reg index 0x0000: -145
[   68.346470] ax88179_178a 1-1:2.1 eth1: register 'ax88179_178a' at usb-dwc2-1, ASIX AX88179 USB 3.0 Gigibit Ethernet, 92:61:30:2e:c6:c0
[   68.362227] usb 1-1: USB disconnect, device number 2
[   68.377613] ax88179_178a 1-1:2.0 eth0: unregister 'ax88179_178a' usb-dwc2-1, ASIX AX88179 USB 3.0 Gigibit Ethernet
[   68.412003] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -19
[   68.432259] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0001: -19
[   68.445930] ax88179_178a 1-1:2.0 (unregistered net_device): Failed to write reg index 0x0002: -19
[   68.457336] ax88179_178a 1-1:2.1 eth1: unregister 'ax88179_178a' usb-dwc2-1, ASIX AX88179 USB 3.0 Gigibit Ethernet
[   68.502058] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0002: -19
[   68.511703] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0001: -19
[   68.532280] ax88179_178a 1-1:2.1 (unregistered net_device): Failed to write reg index 0x0002: -19
[   68.891957] usb 1-1: new high-speed USB device number 3 using dwc2
[   69.893582] ax88179_178a 1-1:1.0 (unregistered net_device): Failed to read reg index 0x0040: -32
[   70.026910] ax88179_178a 1-1:1.0 eth0: register 'ax88179_178a' at usb-dwc2-1, ASIX AX88179 USB 3.0 Gigibit Ethernet, f8:e4:3b:2d:c1:17

Vendo232, 2022-04-14:
lsusb
Bus 001 Device 003: ID 0b95:1790
Bus 001 Device 001: ID 1d6b:0002

ifconfig eth0
eth0      Link encap:Ethernet  HWaddr F8:E4:3B:2D:C1:17
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

[root@WCV3:~]#

gtxaspec, 2022-04-14:
ok let me look into this.

gtxaspec, 2022-04-15:
i ordered the USB adapter to test, stand by

gtxaspec, 2022-04-17:
received the adapter, I was able to have make it work.  I'll include the fixes in the next release. @Vendo232

gtxaspec, 2022-04-20:
this usb adapter is strange, it doesn't work properly when the plug is connected all the way in, I.E. the the usb 3.0 pins.  if you leave the connector just pushed in a bit, it works.  There are various reports online describing other issues with this adapter.  

I would recommend sticking with a USB 2.0 adapter for the camera's.  If you really want it to work ,don't push in the connector all the way and use the latest release.

closed 2022-04-20 by gtxaspec

Vendo232, 2022-04-20:
thank you very much @gtxaspec