gtxaspec/wz_mini_hacks #10: Status of WyzeCamLiveStream hack on V3.

FiveLeavesLeft, 2022-04-16:
I know some people were interested in getting my hack running again on V3.

The hack relies on changing LD_LIBRARY_PATH and intercepting
calls to IMP_Encoder_GetStream which lives in /system/lib/libimp.so.
This sort of worked on V3 under WyzeHacks but it was tenuous. I remember
being convinced at one time that iCamera relied on C "undefined behavior".

I can't currently get it to work reliably on V3. If I create a libimp.so
that contains only this...

        #define LIB "/system/lib/libimp.so"

        static void *sym = 0;

        void  __attribute__((constructor)) foo(void) {
                printf("CRBMIN constructor");
                if(sym) return;
                sym = dlopen(LIB, RTLD_GLOBAL|RTLD_NOW);
                printf("CRBMIN sym = %p", sym);
        }

        void _init(void) {
                printf("CRBMIN _init");
                foo();
        }

.. the camera runs for awhile but eventually gets errors and reboots.  
What could account for this? I have some ideas:

        1. My toolchain is not exactly compatible with the toolchain iCamera was compiled with.
        2. Bug in the dynamic loader that ships with the camera. ( Doubtful. )
        3. iCamera relying on undefined behavior.

I don't think it's failing because of anything Wyze the company is doing intentionally. I can look at
the stdout of iCamera with and without the hack above and see minor differences in error reporting. So
this hack that shouldn't change any behavior is causing iCamera to fail. A long time ago, on V3 under
WyazHacks I think I observed the dynamic loader stepping on the memory of static strings.

The loader that ships with the camera doesn't have the usual LD_DEBUG stuff compiled in. That would
have been nice.

That's the status, I don't know if anyone out there is interested enough and/or has the expertise
to take this to the next level. I am open to suggestions. I will continue to think about it.

Thanks,

Charlie ( FiveLeavesLeft  )

ps: Thanks for the great work.

gtxaspec, 2022-04-17:
Charlie,

I wasn't too familiar with your project before but I've had many folks contact me regarding the functionality with wz_mini, so maybe we can figure out a way to make it work.

You for sure need the right toolchain.  What I would start with is heading to https://github.com/mnakada/atomcam_tools/ and git cloning his repo.  Follow Build.md (with google translate) to setup a docker environment with the toolchain and a bunch of other stuff.  I was able to sucessfully compile everything for wz_mini using this, and more.  It should also be possible to replace the built-in loader as well.  

Have you looked at the libcallback sources at https://github.com/mnakada/atomcam_tools/blob/main/libcallback/video_callback.c and audio_callback.c?  Maybe you could integrate this method into your livestream hack as well?

FiveLeavesLeft, 2022-04-18:
Thanks for the links. I did a make.sh on atomcam_tools last night and it completed. Wow. A lot to unpack here. Thanks.

FiveLeavesLeft, 2022-04-20:
@gtxaspec In the old version of wz_mini_hacks I had modified setup.sh to modify /etc/rcS to change PATH and LD_LIBRARY_PATH. In looking at the new version of wz_mini_hacks, I don't see a way to do this. Could I get a copy of the old setup.sh ? ( I lost mine. ) Or maybe you could change wz_mini_hacks to put /configs/bin and /configs/lib on the front of PATH and LD_LIBRARY_PATH as this may be generally useful for future hacks. Thanks.

gtxaspec, 2022-04-20:
try https://github.com/gtxaspec/wz_mini_hacks/tree/fe9e712c3894d8e3fc49be86c3b2f2bfb484c6be

gtxaspec, 2022-04-20:
check out the latest release, you, I bind over rcS when using RTSP, maybe you can hook into this.

gtxaspec, 2022-04-20:
another new version, more hooks in different places, you can change fstab, rcS, and some other stuff too.. check it out if you like.

FiveLeavesLeft, 2022-04-20:
Thanks, build.md is very helpful too. You are making great progress at blinding speed.

gtxaspec, 2022-04-25:
any luck?  Also have you checked out libcallback, and seen how he hooked in to iCamera to get the streams?   It seems to work but break the live view and microsd recording when enabled

FiveLeavesLeft, 2022-04-25:
I've been really busy with other things and in the process of moving. Not sure when I will be able to get back to this. You can close this if you like. Thanks.

gtxaspec, 2022-04-26:
@FiveLeavesLeft thanks for the info.  I have a few questions if you have time, I am trying to get the wyzecamlivestream compiled and loaded to test.  

I compiled the code, my questions are:

1.do you replace the libimp.so in /system/lib ?  is iCamera supposed to use this instead?

2. or do you leave that alone and LD_PRELOAD it?

FiveLeavesLeft, 2022-04-26:
I’ve never used LD_PRELOAD but I was going to try that option sometime soon.

I changed LD_LIBRARY_PATH to look in /configs/stream_hack before /system/lib so iCamera would load libimp.so from there. Then I dlopen’ed /system/lib/libimp.so inside my code which is in /configs/stream_hack/libimp.so . This worked on V2 but had some strange side effects on V3 that I don’t understand. 

Using LD_PRELOAD is an interesting approach that I never tried. 

I would like to get back to this, and I feel sort of bad that you are using your precious time on this, but I won’t be able to look at it for at least a week. 

As an aside, in my recent effort I also changed PATH and I created a fake “dmon” that looked like this:

#!/bin/sh

if [ $6 = "/system/bin/iCamera" ]
then
       
        /system/bin/iCamera >& /media/mmc/icam.log &
else
        
        /system/bin/dmon $@
fi

That let me look iCamera’s stdout and stderr. Maybe you have a better way to do that.

gtxaspec, 2022-04-26:
thank you for the detailed explenation,  i have some extra time so no worries.

also, once the library is loaded, how do you trigger the stream? or is it automatic once loaded

gtxaspec, 2022-04-26:
got it partially working with wz_mini, I change the LD_LIBRARY path as you mentioned, and it works briefly but then iCamera reboots the camera...
``` [init.c:1432][sdk_major_error_handle] errCode:1
```
there is hope!

FiveLeavesLeft, 2022-04-26:
That’s what I was seeing. Are you saying it’s working now and you can stream video from it?

gtxaspec, 2022-04-27:
for reference, here is an updated SDK targeted to the T31
https://github.com/BiatuAutMiahn/IVSP-T31-1.1.1-20200508/tree/main/software/sdk/Ingenic-SDK-T31-1.1.1-20200508/samples/libimp-samples

gtxaspec, 2022-04-30:
I have compiled a new version of ld-uClibc.so.0 and libuClibc-1.0.40.so with LD_DEBUG enabled for development

FiveLeavesLeft, 2022-05-09:
I’m finally getting around to looking at this again. Downloaded the latest version, copied the files over to the SDCARD. When it powered up I heard it say something about “creating swap”. I can ping it, but I can’t ssh into it.

My wz_mini/etc/ssh/authorized_keys looks like:

#PLACE AUTHORIZED KEYS BELOW
ssh-rsa AAAAB3NzaC1yc2EAA …..  +P9b30= ***@***.***

I tried ssh <ip>, ssh -l root <ip> and ssh -l crb <ip> . All say "Connection closed by 192.168.8.143 port 22”.

Any ideas? Thanks.

gtxaspec, 2022-05-09:
try ssh -v root@ip, this will enable verbose mode, see if it's trying the key at all.

FiveLeavesLeft, 2022-05-09:
> try ssh -v ***@***.***, this will enable verbose mode, see if it's trying the key at all.

Thanks for the quick reply! Here’s what it says. I am on an MacBook air.

air:wz_mini_hacks-master crb$ ssh -v ***@***.***
OpenSSH_8.6p1, LibreSSL 2.8.3
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 21: include /etc/ssh/ssh_config.d/* matched no files
debug1: /etc/ssh/ssh_config line 54: Applying options for *
debug1: Authenticator provider $SSH_SK_PROVIDER did not resolve; disabling
debug1: Connecting to 192.168.8.143 [192.168.8.143] port 22.
debug1: Connection established.
debug1: identity file /Users/crb/.ssh/id_rsa type 0
debug1: identity file /Users/crb/.ssh/id_rsa-cert type -1
debug1: identity file /Users/crb/.ssh/id_dsa type -1
debug1: identity file /Users/crb/.ssh/id_dsa-cert type -1
debug1: identity file /Users/crb/.ssh/id_ecdsa type -1
debug1: identity file /Users/crb/.ssh/id_ecdsa-cert type -1
debug1: identity file /Users/crb/.ssh/id_ecdsa_sk type -1
debug1: identity file /Users/crb/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /Users/crb/.ssh/id_ed25519 type -1
debug1: identity file /Users/crb/.ssh/id_ed25519-cert type -1
debug1: identity file /Users/crb/.ssh/id_ed25519_sk type -1
debug1: identity file /Users/crb/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /Users/crb/.ssh/id_xmss type -1
debug1: identity file /Users/crb/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.6
debug1: Remote protocol version 2.0, remote software version dropbear_2022.82
debug1: compat_banner: no match: dropbear_2022.82
debug1: Authenticating to 192.168.8.143:22 as 'root'
debug1: load_hostkeys: fopen /Users/crb/.ssh/known_hosts2: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts: No such file or directory
debug1: load_hostkeys: fopen /etc/ssh/ssh_known_hosts2: No such file or directory
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ssh-ed25519
debug1: kex: server->client cipher: ***@***.*** MAC: <implicit> compression: none
debug1: kex: client->server cipher: ***@***.*** MAC: <implicit> compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
Connection closed by 192.168.8.143 port 22
air:wz_mini_hacks-master crb$

gtxaspec, 2022-05-09:
i think your mac ssh client doesn't like dropbear ssh.  you can disable the ssh only keys requirement, edit `/opt/wz_mini/init.d/v3_init.sh` and delete the `-s -g` from the dropbear command and then save and reboot.

raelbsd, 2022-05-11:
Hi, thanks for hard work! I have 4 wyze cams, I made it work in one, but in the three left had no luck. I'm trying to ssh and I have this, the first time a kex_exchange_identification error and then "Connection refused" all the times. I've already tried to delete the -s -g, and it's the same. In the v3_init.log I don't see any errors or warnings related.

Maybe I should open a new issue but looks it's related to the latest error in thread.
What else could I try?
Thanks!

root@calisto:~# ssh -v 192.168.1.30
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug1: Connecting to 192.168.1.30 [192.168.1.30] port 22.
debug1: Connection established.
debug1: identity file /root/.ssh/id_rsa type -1
debug1: identity file /root/.ssh/id_rsa-cert type -1
debug1: identity file /root/.ssh/id_dsa type -1
debug1: identity file /root/.ssh/id_dsa-cert type -1
debug1: identity file /root/.ssh/id_ecdsa type -1
debug1: identity file /root/.ssh/id_ecdsa-cert type -1
debug1: identity file /root/.ssh/id_ecdsa_sk type -1
debug1: identity file /root/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /root/.ssh/id_ed25519 type -1
debug1: identity file /root/.ssh/id_ed25519-cert type -1
debug1: identity file /root/.ssh/id_ed25519_sk type -1
debug1: identity file /root/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /root/.ssh/id_xmss type -1
debug1: identity file /root/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.2p1 Ubuntu-4ubuntu0.4
kex_exchange_identification: read: Connection reset by peer
root@calisto:~# ssh -v 192.168.1.30
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug1: Connecting to 192.168.1.30 [192.168.1.30] port 22.
debug1: connect to address 192.168.1.30 port 22: Connection refused
ssh: connect to host 192.168.1.30 port 22: Connection refused

gtxaspec, 2022-05-11:
@raelbsd this could mean that the memory card is corrupted, or the authorized_keys file is corrupted.  Have you tried reformatting your cards and copying everything fresh?

raelbsd, 2022-05-12:
Hi, I'll try with a brand new card, I've formatted all the 4 cards and I did a fresh install several times. 
Also I've tried with a normal user with the keys generated, and when I've tried as root, I've change the dropbear parameters to allow password logins.
I'll try with a new card to see if I got luck.
Thanks!

FiveLeavesLeft, 2022-05-12:
I finally did get the WyzeCamLiveStream to work under wz_mini_hacks. A minor code change and use of LD_PRELOAD and LD_LIBRARY_PATH. Do you want to incorporate it into wz_mini_hacks or shall I maintain a separate GitHub repo for it? It’s a couple of C files and a Makefile. 

Thanks.

gtxaspec, 2022-05-12:
I would love to include it in the project!

@FiveLeavesLeft question, what video stream does it tap into? the current solution In the project uses the same bitrate used to stream in the app I believe, and users complain the bitrate is too low compared to the stock rtsp fw.

gtxaspec, 2022-05-12:
also, I wonder if it would be possible to direct the video stream to a /dev/video loopback device so v4l2rtspserver can handle the audio as well, or does your code do audio as well?

FiveLeavesLeft, 2022-05-13:
The code interposes a function in libimp.so that gets called for each frame of h264 video. I don’t know what the bitrate is, my code is very naive and just writes a frame each time the function is called. I never tried to throttle anything.

The code just writes to file descriptor so maybe I could write to /dev/video or something similar. A named piped maybe? 

No audio at all. 

Wyze runs iCamera with LD_PRELOAD=libsetunbuf.so, does wz_mini_hacks do that also? 

Thanks.

gtxaspec, 2022-05-13:
the libsetunbuf is only needed for the wyze dmon logging, though wz mini only runs without it when running the rtsp server, as libcallback has it built in I believe.

FiveLeavesLeft, 2022-05-14:
If you want to try the new live stream version, ssh into the camera and:

cd /configs
wget www.sonic.net/~crb/dev/get
source ./get
reboot

For this to work. you must modify LD_LIBRARY_PATH to have /configs/lib_hack in front of everything else eg

export LD_LIBRARY_PATH=/configs/lib_hack:$LD_LIBRARY_PATH

How you do that is up to you. 

This version depends on LD_PRELOAD=libsetunbuf.so ( as is the case in app_init.sh ). 

To stream try ( where $(IP) is the ip address of the camera ).
	vlc http://$(IP):12345 —demux h264
or
	mpv http://$(IP):12345

If this works for you, we can discuss how best to integrate it into wz_mini_hacks and I can get you the Makefile and source code.

Thanks.

gtxaspec, 2022-05-14:
tested, it works!

gtxaspec, 2022-05-14:
if you want to update your repo I can get the source from there, if that's easiest for you

endertable, 2022-05-30:
Was this actually integrated in to wz_mini already, I didnt’t notice the update.   Or is it integrated into the rtsp server?

FiveLeavesLeft, 2022-05-31:
It wasn't integrated. Do you think there is reason to do so? Is there something that this adds that isn't covered by everything else running so well? 

The files I mentioned below are no longer up to date. If there's a reason to integrate I will need to update them. Otherwise we should close this issue. Thanks.

gtxaspec, 2022-05-31:
@FiveLeavesLeft maybe we could keep your solution in mind as a back up just in case, if you agree.  It's always good to have more than one way to achieve something, especially on such a popular feature.

FiveLeavesLeft, 2022-05-31:
Sounds good to me.

gtxaspec, 2022-05-31:
great to hear, and thank you very much!

closing for now.

closed 2022-05-31 by gtxaspec

gtxaspec, 2022-06-28:
@FiveLeavesLeft Hello!  I wanted to include your h264 solution in an upcoming release of wz_mini as an alternative to v4l2rtspserver for users, would you mind sharing the source to integrate it?

thanks!

FiveLeavesLeft, 2022-10-11:
I override a routine, IMP_Encoder_GetStream that gets called for each frame. I call the original version of the routine and if someone has opened a connection to my little server I send the frame over that connection. 

So I have some C code that waits for a connection. That runs in it’s own little thread.

FiveLeavesLeft, 2022-10-11:
Great. The Makefile and source code and toolchain are in www.sonic.net/~crb/dev/src.tgz . I forget where exactly I got this toolchain, but it’s probably the same one you are using. 

Integrate this into wz_mini_hacks however you want. I chose /configs because it’s writable and persistent on V2 and V3. ( I haven’t made this version work with V2 yet. ) 

The code compiles into libcrb.so which is copied into /configs/lib_hack/libsetunbuf.so . 

How it works:

The routine IMP_Encoder_GetStream is called for each h264 frame. It it used internally by iCamera and lives in /system/lib/libimp.so . My code intercepts the calls to IMP_Encoder_GetStream, calls the version in libimp.so and also writes the h264 frame to a client ( file descriptor ) if there is one. 

Changing it to write to some other file descriptor would be trivial. I’ve never done anything with /dev/video and I don’t know anything about it. 

Let me know if you have any questions. Thanks.

ps: Thinking about this,  this hack is persistent and will live on even without a sdcard.