gtxaspec/wz_mini_hacks #8: RTSP Server?

yeahme49, 2022-04-16:
I didn't see it mentioned so I'm assuming not, but does this enable RTSP output from the camera without needing Wyze's RTSP firmware?

gtxaspec, 2022-04-16:
Not yet, somethings to figure out and work on with the Wyze firmware.

virmaior, 2022-04-16:
the atomcam team found that RTSP support was unstable when they added it directly to their package. Assuming we've got the same fundamental hardware, we face the same limit.

evanheckert, 2022-04-16:
FWIW, I'm on the factory RTSP firmware with this hack working as desired. @yeahme49 & @virmaior

gtxaspec, 2022-04-16:
I've been unable to use the atomcam team' build to output RTSP.  I've tried to compile the binary, then use the loopback module, no go so far.  Maybe someone has more info on it, looks like some stuff may be different in the Wyze FW compared to the atomcam

virmaior, 2022-04-16:
@gtxaspec I can buy them here in Japan... The company's website and amazon-jp won't ship direct to US, but I can always buy one and then ship it.

gtxaspec, 2022-04-16:
@virmaior i ported over the atomcam tools to the wyze hardware, and rtsp does not work.  it seems that the atomcam team uses a special library, libcallback.so, and this is tuned to the iCamera_app for the atomcam only.

I was wondering, perhaps you could be our liason to @mnakada...maybe they can analyze our iCamera app, or we could arrange to send him a wyzecam v3 for free so he could study it?  If they are willing

let me know what you think

mnakada, 2022-04-16:
I don't use WyzeCam so I can't answer correctly.
Have you looked at the following?
Great hack for WyzeCam, although V3 doesn't seem to be supported.
https://github.com/openmiko/openmiko

Check the following to get RTSP working
- libcallback/video_callback.c intercepts the callback where CameraApp is video encode using local_sdk.
  Is the callback here being called?
　I think local_sdk is probably the manufacturer's library used by both AtomCam and WyzeCam.
　If it is not called, this needs to be addressed first.
- Put audio aside for the moment and concentrate on getting Video through.
- I am outputting the intercepted VIDEO to V4L2. are the VIDEO related settings in the kernel config correct?
Compare with configs/kernel.config.
- The v4l2rtspserver is brought from openmiko. I didn't need any modifications.

I do not need WyzeCam.
In Japan, the radio law is strictly regulated and any device that is not through TELEC certification is illegal.

gtxaspec, 2022-04-16:
@mnakada thank you for the information, I have a working prototype  of RTSP server working with the Wyze Camera now

gtxaspec, 2022-04-17:
if someone wants to test,

https://github.com/gtxaspec/wz_mini_hacks_rtsp

<ip address>:8554/unicast

video only at the moment

gtxaspec, 2022-04-17:
Using https://github.com/mnakada/atomcam_tools/tree/main/libcallback as base, I removed everything except video_callback.c and audio_callback.c, and compiled libcallback.so. You can LD_PRELOAD this to iCamera and then run /media/mmc/v4l2rtspserver.  Some issues:

1. Live view only works with 360p resolution
2. no audio
3. choppy

Vendo232, 2022-04-17:
> if someone wants to test,
> 
> https://github.com/gtxaspec/wz_mini_hacks_rtsp
> 
> IP_ADRESS:8554/unicast
> 
> video only at the moment

it is Christmas? so cool what you are doing.

I`m testing it, flashed 4.36.8.32
Copied files from https://github.com/gtxaspec/wz_mini_hacks_rtsp on empty SD Card , but rtsp://192.168.1.56:8554/unicast is not working in VLC

gtxaspec, 2022-04-17:
just updated a bit ago, was missing a file

Vendo232, 2022-04-17:
> just updated a bit ago, was missing a file, and btw VLC doesn't work for some reason on PC, but it works on my phone.

I have it, just downloaded it like 10min ago, these are my files on SD Card

[image]

Vendo232, 2022-04-17:
it is WORKING ! , it takes a minute to start probably

gtxaspec, 2022-04-17:
cool, let me know how the quality is for you

Vendo232, 2022-04-17:
it is evening here, but I will put it outside and on Frigate tomorrow and will let you know. 

This is HISTORIC !!! 

[image]

gtxaspec, 2022-04-17:
@mnakada thank you for your guidance.  I have achieved good video stable video results, but audio is very garbled, sounds very slow... do you have any ideas?  command line: v4l2rtspserver -A 8000 -C 1 -a S16_LE /dev/video1,hw:Loopback,0

gtxaspec, 2022-04-17:
```
/media/mmc/wz_mini/bin/v4l2rtspserver [-v[v]] [-Q queueSize] [-O file]
	          [-I interface] [-P RTSP port] [-p RTSP/HTTP port] [-m multicast url] [-u unicast url] [-M multicast addr] [-c] [-t timeout] [-T] [-S[duration]]
	          [-r] [-w] [-s] [-f[format] [-W width] [-H height] [-F fps] [device] [device]
	 -v               : verbose
	 -vv              : very verbose
	 -Q <length>      : Number of frame queue  (default 5)
	 -O <output>      : Copy captured frame to a file or a V4L2 device
	 -b <webroot>     : path to webroot
	 RTSP/RTP options
	 -I <addr>        : RTSP interface (default autodetect)
	 -P <port>        : RTSP port (default 8554)
	 -p <port>        : RTSP over HTTP port (default 0)
	 -U <user>:<pass> : RTSP user and password
	 -R <realm>       : use md5 password 'md5(<username>:<realm>:<password>')
	 -u <url>         : unicast url (default unicast)
	 -m <url>         : multicast url (default multicast)
	 -M <addr>        : multicast group:port (default is random_address:20000)
	 -c               : don't repeat config (default repeat config before IDR frame)
	 -t <timeout>     : RTCP expiration timeout in seconds (default 65)
	 -S[<duration>]   : enable HLS & MPEG-DASH with segment duration  in seconds (default 2)
	 -x <sslkeycert>  : enable RTSPS & SRTP
	 V4L2 options
	 -r               : V4L2 capture using read interface (default use memory mapped buffers)
	 -w               : V4L2 capture using write interface (default use memory mapped buffers)
	 -B               : V4L2 capture using blocking mode (default use non-blocking mode)
	 -s               : V4L2 capture using live555 mainloop (default use a reader thread)
	 -f               : V4L2 capture using current capture format (-W,-H,-F are ignored)
	 -f<format>       : V4L2 capture using format (-W,-H,-F are used)
	 -W <width>       : V4L2 capture width (default 0)
	 -H <height>      : V4L2 capture height (default 0)
	 -F <fps>         : V4L2 capture framerate (default 25)
	 -G <w>x<h>[x<f>] : V4L2 capture format (default 0x0x25)
	 ALSA options
	 -A freq          : ALSA capture frequency and channel (default 44100)
	 -C channels      : ALSA capture channels (default 2)
	 -a fmt           : ALSA capture audio format (default S16_BE)
	 Devices :
	 [V4L2 device][,ALSA device] : V4L2 capture device or/and ALSA capture device (default /dev/video0,/dev/video0)

```

mnakada, 2022-04-17:
Congrats!!!
You were quick to put the RTSP video through.

Since you said you are getting sound, below is how I did it when I checked AtomCam's audio format.

1. When audio_callback is called, it prints the gettimeofday time, the size of the data and the first 32 bytes or so.
2. Calculate the data rate from the period and data size.
3. Check the endian, bit width, and Stereo/Mono from the data dump.
4. Set the format and bit rate based on that.

Maybe that will work.

Vendo232, 2022-04-17:
hello, here are some pics from the rtsp , it seems moving objects is an issue.

[image]
[image]
[image]
[image]

Vendo232, 2022-04-17:
judging by this discussion : https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/issues/1427

it seems the command will need some finetuning. @gtxaspec is it a right approach to edit this line and reboot to see an effect?

[image]

gtxaspec, 2022-04-17:
yes

gtxaspec, 2022-04-17:
@virmaior i was able to duplicate the pixelation you are experiencing when recording using FFMPEG or playing on desktop using celluloid... surprisingly when I use VLC on android to view, the pixelation is not present, whatsoever.  The stream is perfect.  

Not too experienced in capturing RTSP, as I never used it prior, but just an observation.

Vendo232, 2022-04-18:
this should be the command but it is not working 

`v4l2rtspserver -W 1920 -H 1080 -F 20 -P 8554 -fH264 /dev/video1`

also it seems v4L2 supports also http stream with command -S but I was unable to make it work

gtxaspec, 2022-04-18:
@Vendo232, not working, as in you are setting the command line but nothing is happening?  also don't forget the & at the end after /dev/video1

Vendo232, 2022-04-18:
> 

[image]

Vendo232, 2022-04-18:
[image]

just realized I had a typo in -P argument, after removing Port part it still produces a lot of pixelation when moving objects

`/media/mmc/v4l2rtspserver -W 1920 -H 1080 -F 20 -fH264 /dev/video1 &`

gtxaspec, 2022-04-18:
do you have a mobile device? can you try viewing from Android or iOS VLC and see if the pixelation still happens?

Vendo232, 2022-04-18:
will give it a try

virmaior, 2022-04-18:
my guess would be that the wyze's processor can't handle it at that resolution. Try it at 960x540 and see if the artefact rate drops.

gtxaspec, 2022-04-18:
@mnakada I have a question for you if you don't mind and have the time.  In your libcallback source, for video_callback.c, is it possible to modify this to output to a file (a socket?) instead of a video device (/dev/videoX)?  We have rtsp server from LIVE555  (http://www.live555.com/mediaServer/) which is more efficient, and what Wyze uses in their official firmware. in their firmware, the wyze iCamera produces these files:

```
srwxr-xr-x    1 root     root             0 Dec  6 00:02 live.264
srwxr-xr-x    1 root     root             0 Dec  6 00:02 live.g711
```
which can then be streamed by the livemediaserver.

mnakada, 2022-04-18:
There are several possible ways to pass video data from the iCamera application to the RTSP Server.
- Via /dev/video using V4L2
- Via named pipe
- Via socket
- Via shared memory
I think that memory copying will occur in buffer-cache when using V4L2, named pipe, or via socket.
I think the least amount of memory-to-memory copy is shared memory.

I think all of them are feasible.
But I don't know which one is really efficient.

Vendo232, 2022-04-18:
changed the resolution and I run this command and it had no effect to the streaming format. This explain why all my attempts yesterday were not successful. :-) 

[image]

the stream remains 1920x1080, 20fps ( which is the original Wyze format ) 
[image]

gtxaspec, 2022-04-18:
@Vendo232 I experienced the same result, changing command line parameters for the video settings don't do anything.  Probably because the rtspserver doesn't do any transcoding or modification of the video stream itself?

evanheckert, 2022-04-19:
Curious, as I'm currently using this repo with the factory RTSP firmware. What's the benefit of adding RTSP through the hack instead of using the factory firmware?

gtxaspec, 2022-04-19:
the RTSP official firmware is based off an old build of the official firmware, and is missing the on device AI edge detection stuff I believe, among other stuff.

virmaior, 2022-04-19:
@evanheckert I was also under the impression that the wyze corporate RTSP requires you to use the wyze app periodically to activate it.
At least for me, I'd like to find ways to use my cameras that don't involve wyze at all after they installed an update without my consent to my cameras.

Vendo232, 2022-04-19:
on top of all mentioned here the wyze corporate RTSP firmware is unstable build to many and the cam requires manual reset quite often. In my case I would have to manually unplug the cam every 48h. Having the same cam in the same spot with the same wifi signal but with 4.36.0.228 + HTTP fiveleavesleft stream hack the cam could go for months not missing a beat in Frigate. 

With the wz_mini + [4.36.8.32]  (March 17, 2022) you should have the latest from wyze plus all the goodies and perks from @gtxaspec like LAN, spotlight, USB direct...

Vendo232, 2022-04-19:
> @Vendo232 I experienced the same result, changing command line parameters for the video settings don't do anything. Probably because the rtspserver doesn't do any transcoding or modification of the video stream itself?

while the  stream appears to be working OK in VLC even on the PC , it fails in mpv player ( windows ) and Frigate the NVR for the pixelation of moving objects. It has probably something to do with live555 configuration but I`m not a rtsp expert, not sure what I`m looking for.  Videos from the stream downloaded from Frigate which does not transcode streams has 39fps. 

[image]

gtxaspec, 2022-04-20:
latest release has RTSP built in, supports audio and video now.

evanheckert, 2022-04-20:
> latest release has RTSP built in, supports audio and video now.

So use this repo instead of the rtsp specific one?

gtxaspec, 2022-04-20:
yes, the rtsp specific repo is deprecated now.

capncybo, 2022-04-20:
I understand the RTSP Server was just merged into Main & I was about to try this but...
I'm confused by the external directory of: /libcallback_avonly/
Reading the former installation instructions... we are to install the contents of SD_CARD onto the "SD-Card"
But "/libcallback_avonly/" is External/Above/Outside... the SD_CARD Directory.
Why is it there, & do we have to do anything with it?
Perhaps the installation has changed slightly?

Vendo232, 2022-04-20:
there is no change in installation. The folder /libcallback_avonly I believe contains uncompiled code of the callback files in case you would like to make your own modification.

gtxaspec, 2022-04-21:
still trying to solve the issue of the app not working in HD or SD mode when the rtsp server libcallback is being used, if anyone has an idea feel free to chime in!

RogueDried, 2022-04-23:
Excuse my ignorance but is the video streaming issue localized to the wz_mini_hacks or does this probablem stem from parts of the other projects ie dfangs? I am trying to get a better understanding of how this is all tied together.

gtxaspec, 2022-04-23:
the issues with RTSP are specific to the implementation used within this project.  
libcallback is used to hook the video.  the sources are included in the repo.

RogueDried, 2022-04-23:
Okay if you dont mind me asking for clarification both video_callback. and audio_callback.c are combined together into libcallback.so. Comparing these to other intterations the only difference I have noticed is the logic for enabling audio / video and setting the CaptureEnable variable to 1 by default. Hopefully I am on track here. Both files refense the liblocalsdk.so. Is this all correct?

gtxaspec, 2022-04-23:
right

gtxaspec, 2022-04-24:
if we switch to the second channel, ```  if(ch == 1) {``` in video_callback.c, then the RTSP server will output 360p, and all functions work fine, recording to SD card, live view, etc, but the resolution is lower of course.  Is this something that people are OK with?  Or are you all OK with the current RTSP situation

virmaior, 2022-04-24:
For me, optimal would be SD card recording + full res RTSP.
360p RTSP and SD would be second best.

capncybo, 2022-04-24:
> if we switch to the second video channel output from the t31 image signal processor, ` if(ch == 1) {` in video_callback.c, then the RTSP server will output 360p, and all functions work fine, recording to SD card, live view, etc, but the resolution is lower of course. Is this something that people are OK with? Or are you all OK with the current RTSP situation

It's your project (& an Excellent one), Thank you for asking for our input.  I totally understand there may be times when a compromise must be made.  But to clarify... would the WYZE app stream also be 360p & only the RTSP Stream be lower res?

gtxaspec, 2022-04-24:
only rtsp would be 360p...I will probably add it as an option so everyone has the most options.

ankurvdev, 2022-04-28:
i'm curious. Can we not simply lift up the binary
`/system/bin/liveMediaServer` from the rtsp firmware and drop it into the sdcard and run it
Is that not the binary responsible for rtsp streaming ?

Would that binary not work on non-rtsp firmwares ?

gtxaspec, 2022-04-28:
no, that rtsp server doesn't work with v4l2 devices, which are utilized in this mod at the moment.

gtxaspec, 2022-05-07:
please test the latest version for RTSP fixes.

Vendo232, 2022-05-07:
> please test the latest version for RTSP fixes.

Will test it later today

gtxaspec, 2022-05-09:
rtsp seems to work great now, running over 24 hours on a v3 and panv2, no issues.  also, ffmpeg seems to work fine now.

`ffmpeg -rtsp_transport tcp -i rtsp://admin:password@x.x.x.x:8554/unicast test.mp4` is what i tested.  

let me know and i'll open this back up if any other issues arise.

closed 2022-05-09 by gtxaspec

Vendo232, 2022-05-11:
sorry for delay, I`m up and running it for 24hrs now and all looks good, running the latest FW 4.36.9.131

Question: Is there an option to increase bitrade beyond 1000kbps?

[image]

Vendo232, 2022-05-12:
we will need to bum up the bitrade, the WYZE RTSP FW has it above 1000kbps. At ~300kpbs the picture is at low quality 

WYZE RTSP
[image]

[8 more replies]