gtxaspec/wz_mini_hacks #9: Working on 4.36.8.32?

yeahme49, 2022-04-16:
Shut down my V3, pulled the SD card, put into my computer and copied the files to the root of the SD card, put back in camera and turned it on. Unable to connect to SSH, connection refused. Pulled card and the wz_mini_hacks.log file is empty. Put the card back in and tried again, same issue. Does anyone have this working on 4.36.8.32 (latest version)?

gtxaspec, 2022-04-16:
try reflashing the latest stock firmware, and make sure that the factory file and the wz_mini directory are the only items in the root of the sd card.

virmaior, 2022-04-17:
yeahme -> how big is your SD card what is the format?
wyze's support for larger SD cards is a bit buggy and might not work at the stage where this hack happens.

yeahme49, 2022-04-17:
128gb card. I'll try a smaller card and see if that works better when I get a chance.

yeahme49, 2022-04-19:
32GB card fat32 format worked. The 128gb card was exfat and could not get that to work. Don't have a 64gb card to test so not sure if 32 or 64 is the max that works for now.

gtxaspec, 2022-04-20:
looks like it works well, closing issue.

closed 2022-04-20 by gtxaspec