gtxaspec/wz_mini_hacks #11: Camera will not boot when files are present

AscendedGravity, 2022-04-21:
WyzeCam v3 running firmware 4.36.8.32 and a 32gb SD card formatted for FAT32. 

Camera will boot normally without an SD card installed or with a newly formatted empty card. Upon adding the wz_mini folder and the factory_t31_ZMC6tiIDQN to the root directory the camera fails to connect to WiFi and a solid red light appears. I've re-flashed the latest firmware as well to "start fresh". The log file also is empty.

I've waited as long as 30 minutes and nothing.

gtxaspec, 2022-04-21:
when you have the red light, press the setup button on the camera, do you get the "ready to connect prompt"?  if so, try running the setup in the app again.  sometimes the wi-fi settings don't load properly, stock firmware bug.

AscendedGravity, 2022-04-21:
Pressing the setup button does not prompt the Ready to Connect cue. I re-did the setup process regardless, but still stuck. Comparing the SD card afterwards versus the contents from the build zip file, it does appear that the camera is at least unpacking the swapfile and terminfo archives. So something is happening.

gtxaspec, 2022-04-21:
what is the model of your camera and first 3 groups of the Mac address?

AscendedGravity, 2022-04-21:
Model: WYZEC3
Mac: 2C:AA:8E

gtxaspec, 2022-04-21:
I'll review the logging functions shortly so we can get some more info...stand by

gtxaspec, 2022-04-21:
question, this is with the stock run_mmc.sh config? or have you changed any of the configuration?

AscendedGravity, 2022-04-21:
The first time I only attempted to change the hostname. Attempts after that were stock.

gtxaspec, 2022-04-21:
in the file ```wz_mini/etc/init.d/v3_init.sh```

add the following:

```exec >/opt/logfile.txt 2>&1```right above the 5th line, above ```echo '```
so that exec should be line #5

then boot the camera, when it stalls, wait 30 seconds, take the card out and then look at logfile.txt in the root of the sd card, feel free to paste that here

AscendedGravity, 2022-04-21:
[logfile.txt](https://github.com/gtxaspec/wz_mini_hacks/files/8535626/logfile.txt)

gtxaspec, 2022-04-21:
looks like ```mount: mounting tmpfs on /opt/wz_mini/tmp failed: No such file or directory``` is the problem.  inside your wz_mini, is there a tmp directory?

gtxaspec, 2022-04-21:
No, there wouldn't be, because the git is missing this directory.  Strange.  i will fix this in a moment.  Create a folder named tmp and then you should be able to boot

gtxaspec, 2022-04-21:
git repo has been updated, let me know if it works for you.

AscendedGravity, 2022-04-21:
Adding the tmp folder did the trick! Confirmed with a successful SSH connection.

closed 2022-04-21 by gtxaspec