gtxaspec/wz_mini_hacks #12: Feature Request: Fall back to wifi if ethernet fails

evanheckert, 2022-04-22:
First of all, this repo (and its ongoing evolution) is amazing and appreciated. This request would be icing on the cake, not at all a complaint!

I'm troubleshooting my ethernet connection (pretty sure it's my micro USB extension that's the problem since it works without). Unfortunately if ethernet fails to connect for any reason, I can't SSH in to debug since there's no network connection, requiring me to get to the mounted camera and get the card out, change the variable in run_mmc.sh, and connect via wifi.

Would it be possible to attempt the ethernet connection, then after a number of seconds (configurable in run_mmc.sh, perhaps?) reset the wlan0 to its original setup?

Thanks again!

endertable, 2022-04-26:
Not quite sure how ethernet works on these cams as I run all mine on Wifi.  Do you plug the ethernet adapter it into the power cord microUSB?  I can try to give you a little start up of how you might be able to crank up your wifi if your ether goes haywire:

At the very end of run_mmc.sh paste the following between the 2 lines:

-------------------------------------------------------------------------------------------------------
startWiFI(){
mySID="SID"
myPSK="Password"

cat <<- EOS > /tmp/wpa_supplicant.conf
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
        ssid="${nySID}"
        key_mgmt=WPA-PSK
        pairwise=CCMP TKIP
        group=CCMP TKIP WEP104 WEP40
        psk="${myPSK}"
        scan_ssid=1
        priority=2
}
EOS

killall -9 wpa_supplicant
killall -9 udhcpc
ifconfig wlan0 down 1>/dev/null 2>&1
sleep 2
ifconfig wlan0 up  1>/dev/null 2>&1
/system/bin/restart_wlan0.sh
wpa_supplicant -D nl80211 -iwlan0 -c/tmp/wpa_supplicant.conf -B  1>/dev/null 2>&1 &
wpa_cli blacklist clear  1>/dev/null 2>&1
udhcpc -i wlan0 -p /var/run/udhcpc.pid -b  1>/dev/null 2>&1 &
sleep 10
LOOP="get_TF_out"
}

MAXTRIES=2
CNT=0
while [ "$LOOP" = "" ]
do
 	#  Place an IP on ethernet you can test ping from this cam
	ping 192.168.1.1 1>/dev/null 2>&1
	if [ $? != 0 ]
	then
		if [ $CNT -ge $MAXTRIES ]
		then
			startWiFI
		else
			CNT=$((CNT+1))
		fi
	fi
	sleep 5
done

-------------------------------------------------------------------------------------------------------

What this will do is every 10 seconds ping a device on your ether connection.  Upon 2 failures (MAXTRIES) it will run startWiFi
function which will start up your WiFi (make sure you adjust "SID" and "Password").  If it doesn't work, you may have to ifconfig down the ether device first, which I have no idea what the interface name is for that, but you can run "ifconfig" while connected to ether to find out.

This repo, like WyzeHack, and Fiveleaves are amazing for the Wyze cam community!!!  Thanks @gtxaspec !!

Good luck and let me know if it works.
ender

gtxaspec, 2022-06-17:
i think this is what @skilau is working on: https://github.com/gtxaspec/wz_mini_hacks/pull/89

gtxaspec, 2022-06-23:
feature release in https://github.com/gtxaspec/wz_mini_hacks/commit/81df1b22bbcd1adea0605c03fbb7b74d1a646fea should satisfy this!

closed 2022-06-23 by gtxaspec