gtxaspec/wz_mini_hacks #13: v2 support?

gluedots, 2022-04-22:
Saw this posted on reddit where someone said the v2 was also supported, but I don't see any mention of the v2.

Would I just follow the instructions for the v3?

gtxaspec, 2022-04-22:
No, it doesn't work for V2.

See https://github.com/Dafang-Hacks/Xiaomi-Dafang-Hacks for the V2

gluedots, 2022-04-22:
Could support for the v2 be added in the future?

virmaior, 2022-04-22:
for v2, why not https://github.com/openmiko/openmiko ?
I think the pedigree from this project winds up being partially derived from that because atomcam_tools is partially derived form that and this is partially derived from atomcam_tools

gtxaspec, 2022-04-24:
will look into this at a later time.

closed 2022-04-24 by gtxaspec

gtxaspec, 2022-05-28:
@gluedots v2 now supported.

gluedots, 2022-05-29:
Thanks, will give it a try.