gtxaspec/wz_mini_hacks #14: How to add user for non-root NFS mounts

fabriziorizzo, 2022-04-22:
remote NFS v4 volume requires specific UID/GID (1007/1007), does not allow root.  How can NFS mounts be configured / executed from your script  with the correct UID/GID to allow access?

endertable, 2022-04-25:
At the bottom of run_mmc.sh script (before the NFS mount command) you can add something like the following:

cp /etc/passwd /opt/wz_mini/tmp/.storage/passwd
mount --bind /opt/wz_mini/tmp/.storage/passwd /etc/passwd

grep -q 'nfsuser:x:1007:1007:' /etc/passwd
if [ $? != 0 ]; then
  echo 'nfsuser:x:1007:1007:NFS Pseudo User:/:/bin/sh'   >> /etc/passwd
  # echo 'nfsuser:1007:x' >> /etc/group
fi

Not sure if you would have to populate the /etc/group file also?!?  But if you do, bind it like with /etc/passwd, then uncomment the line in the if-then routine.

gtxaspec, 2022-05-01:
@fabriziorizzo I don't think I have nfsv4 enabled in the kernel...is this something you still need?

fabriziorizzo, 2022-05-01:
> @fabriziorizzo I don't think I have nfsv4 enabled in the kernel...is this something you still need?

Yes please.

gtxaspec, 2022-05-08:
@fabriziorizzo check the latest release for NFSv4 support, let me know if it works for you.

gtxaspec, 2022-05-14:
let me know if you have any problems, closing for now. @fabriziorizzo

closed 2022-05-14 by gtxaspec

fabriziorizzo, 2022-05-14:
Thank you.  have not had time to test yet.