gtxaspec/wz_mini_hacks #15: Trouble installing, pls help I'm confused

capncybo, 2022-04-22:
Wyze Cam V4, firmware: 4.36.8.3
- 1st Attempts realized my SD-Card was likely TOO Big (It was 128GB & formatted via extFAT)
- Located another SD-Card (16GB), used Gparted to delete previous Raspberry Pi NOOBS partitions & then reformatted in FAT32
I did this after reading the success of the previous closed Issues pertaining to Wyze firmware: 4.36.8.3
Were not supposed to format the SDcard in ext3 or something are we???

Also noticed that secure ssh was just added overnight (while I slept).
So I'm not sure if the old method of...
logging in via SSH on port 22 as "root" & password "WYom2020" still works?
Or is it now ONLY possible to ssh in after adding the secure SSH?

FOUND: 
SSH is enabled, but is secured using public key authentication for security. Edit the file wz_mini/etc/ssh/authorized_keys and enter your public key here. If you need a simple guide, [how to use public key authentication](https://averagelinuxuser.com/how-to-use-public-key-authentication/)

I was expecting to be able to at least see a closed port 22 (But perhaps that's not how secure SSH works & it fails silently).
Any advise would be appreciated
Thanks

memcgregor, 2022-04-22:
Here is an alternative for the sshkeygen for linux if your on windows https://www.ssh.com/academy/ssh/putty/windows/puttygen 

additionally this was continueing to fail for more. I  tried multiple different SD cards of different sizes and was having issues. Additionally tried multiple camera firmwares from offical and beta. I found the issue was due to me utilizing the files from my local git repo. Once i downloaded the zip directly and exported over to the microsd card with only the contents of SD_ROOT on the drive and that worked for me.

RogueDried, 2022-04-22:
Additionally if SSH is providing a failure to connect / port closed then it failed to load from the SD card.  I

capncybo, 2022-04-23:
> Here is an alternative for the sshkeygen for linux if your on windows https://www.ssh.com/academy/ssh/putty/windows/puttygen
>
> additionally this was continueing to fail for more. I tried multiple [...]

Thanks for the added info...
Unfortunately, I still cannot get this to work.  Hence (if you don't mind) I'll ask some more questions.
Perhaps these (answers/details) could be added to the **PREREQUISITES** Section ???
-Do we have to actually start fresh by... Removing any previously WYZE registered Cameras for the the WYZE app/account?
 So far... I've been trying to add the files to the root directory of the SD-Card, but the Camera was still associated as previous.
 (In my case I abandoned using a Larger 128GB SD-Card & tried a16G SD-Card formatted using FAT32)
-What SD-Card format types are compatible for this to work?
-Does the Volume label matter, & should it actually be called SD_ROOT (Or can the Card's Volume label be anything)?
-Previously I had the WYZE V3 Camera MAC address use a static IP address (& it still is) do the scripts require DHCP enabled?
-When using secure SSH do you still add the "root"username such as, for example...  ssh @root@192.168.50.22
-Which firmware versions have been verified to be compatible. (I am currently on Wyze Cam V4, firmware: 4.36.8.3)

I think I read a closed-issue conversation where @gtxaspec mentioned pressing the setup button when the camera led was RED.
To myself this would seem to indicate, any previously associated wyze cams should be deleted or removed from the WYZE app/account prior to attempting install this hack.
+
Perhaps in some cases, the firmware needs to be rolled backwards to a prior version?
Sorry for the Lengthy POST but...
I'm "typically" fairly good with these types of firmware/software upgrades.
Perhaps I have an incompatible SD-Card or just tried after recent WYZE firmware disabled this HACK???
Note:
I also typically have the [docker-wyze-bridge](https://github.com/mrlt8/docker-wyze-bridge):
running on a pi400 so perhaps it was interfering with this HACK somehow

Anyways I'm batting a BIG ZERO on this one.
Not feeling very clever.
Whatsoever LOL
HELP
Pls.

RogueDried, 2022-04-23:
> Author

- You do not need to remove the cameras or modify them or your account in any way.
- FAT32 and exFAT are support by the camera. I personally only had luck with FAT32
- [ ] SD_ROOT label is what i had success with. I have not validated it differently
- No DHCP does not matter this takes place during boot.
- Yes you need to ssh root@192.168.50.22
- This does not require a specific version to operate (no down grading is required)
- You only need to press the setup button if for some reason the wireless settings get wipped out.
- No change is required of the firmware

Try formatting the card via the camera and ensuring the label is SD_ROOT and then copy the files over. Is the log populating in wz_mini/log/wz_mini_hacks.log? If this is zero its not launching. Additionally did you download the zip or clone the repo?

capncybo, 2022-04-24:
Thank you (ever-so-much) for the detailed response & suggestions.  "FINALLY" was able to ssh into my WYZE Cam V3.
But I wanted to share what I learned to assist others & save them from wasting so much time.
I used FAT32 on a 16G SD-Card (Partitioned the entire thing as one Partition & used the Volume Label: SD_ROOT
Note: I was using a Raspberry Pi400 with the Raspbian-OS & here's a WARNING...
-The default text editor "Mousepad" was apparently truncating the end of the public key that I was trying to paste into:
wz_mini/etc/ssh/authorized_keys

But aside from the one change to the "authorized_keys file"...
I noticed there was a hidden .gitignore file inside the: /SD_ROOT/wz_mini/tmp/
I decided to delete it & try again... But I didn't make any other changes.

However this time instead of Doing a HARD Reboot by physically unplugging & re-plugging in the camera...
I used my phone & the WYZE app to do the RESTART Camera (at the bottom of: Settings)
And... immediately after doing (the Software RESTART), I was on my pi400 at a terminal entering:
root@192.168.50.22
It timed out quickly & said no host... But... I could just hit the up arrow & enter to try again, & again.
After about 3-4 attempts... The SSH key message displayed & the Camera actually connected ;-)
Note: the first time my terminal session glitched & I couldn't type anything (Could have been my pi400).
But after closing the window & re-sending another: root@192.168.50.22
Finally, I could type & I was connected to the Camera :-)

Anyways, perhaps I was misunderstanding the installation instructions but I wasn't expecting the time-frame of SSH-login to be brief.  Meaning in my previous testing, whenever I plugged in the camera...
My Camera seemed to boot Normally & there wasn't ever a SSH daemon running to connect with.

closed 2022-04-25 by capncybo