gtxaspec/wz_mini_hacks #17: View Playback does not work with this installed

frugala, 2022-04-23:
First, this hack is amazing, I am rtsp-ing to VLC as I type this!   But, the normal wyze app will not record to the SD card now.  Is this a known thing?  Or is it a normal wyze bug and I need a different SD card or something?  Can anyone confirm that they have the continuous record to SD card while using this hack?  THANKS!

gtxaspec, 2022-04-23:
its an issue with the rtsp...when you enable RTSP, pretty much using an external playback client is the best way to view and record.

frugala, 2022-04-23:
OK.  Is there a way to disable RTSP and re-enable record to SD via SSH with this hack?  Then I could remotely choose whichever I want...

gtxaspec, 2022-04-23:
if you follow the read me, it explains how to change the variables, the file is run_mmc.sh on the memory card /media/mmc/wz_mini

frugala, 2022-04-23:
Thank you.  This is amazing, impressive, and fun!

frugala, 2022-04-23:
I found this on reddit because of a post in the wyze subreddit, but that post is gone now.  Is it against the rules to talk about this in there?  Can we ask questions and share info about this hack in that subreddit?

gtxaspec, 2022-04-23:
it's most likely against their rules, I wouldn't do it, they may ban you lol

skilau, 2022-04-23:
Any thought to investigating whether the SD card could be partitioned in such a way that the added files the boot from exist on 1 partition while having another partition that is exposed to the Wyze side for their SD record storage?
Not sure if it is possible, but thought I would throw it out there?

frugala, 2022-04-23:
I can confirm that i was able use SSH and edit the run_mmc.sh file using vi, and I simply disabled RTSP, and now the camera is doing the continuous record to SD card.  

New question:  Last night when I had RTSP running, i was able to type in a web browser the address directly to a .mp4 file in the record directory, and it would download to my PC.  So I was going to write a script to grab the new file every minute.  But now either I am not typing the path correctly, or maybe it required something that RTSP enabled, so I can't get those files when RTSP is not running.  ?  thanks

gtxaspec, 2022-04-24:
@frugala yes, there is a lot of potential with this mod as to what you can do.  the sky is the limit.

@skilau The problem with the RTSP server is the way the libcallback hack modifies the stock firmware, nothing to do with the sd card.  Need more investigation on this. 

closing for now

closed 2022-04-24 by gtxaspec

unoukujou, 2024-02-22:
> its an issue with the rtsp...when you enable RTSP, pretty much using an external playback client is the best way to view and record.

Quick question... I have RTSP enabled and continuous recording to the SD card is working fine for me. According to what I'm reading in this issue, it should not be working? Just want to say it is indeed working for me at least... ??

After I read the comments here I was going to uninstall wz_mini because I absolutely need continuous recording to the camera's SD card. But after checking it myself, it is indeed working and so is RTSP. Is there any issue I need to look out for? Because the continuous recording is saving the dats to the SD card that wz_mini is installed on. What happens when it consumes all the free space? There will be no free space left for wz_mini to use?

frugala, 2024-02-23:
> >  
> But after checking it myself, it is indeed working and so is RTSP. Is there any issue I need to look out for? Because the continuous recording is saving the data to the SD card that wz_mini is installed on. What happens when it consumes all the free space? There will be no free space left for wz_mini to use?

Regarding your "it works" comment, yes!  It is awesome.  They have been supporting and updating this a lot since that was posted 2 years ago!   Regarding your next comment, I'm not sure about the memory management, but considering how good these dev's are, I bet they have it covered and it is OK.   My guess: the mini hacks themselves do not need any extra memory to run, so it is OK to use it all up for video. (??)

unoukujou, 2024-02-23:
Cool, so then it's an actual feature and not some weird thing that happened only for me.

So anyway, I think wz_mini is running some kind of Linux OS off the SD card? I would expect any OS needs some free space to breathe and not have problems. Definitely concerned about that.

frugala, 2024-02-23:
I found this other thread that suggests it keeps about 1GB free, but that might just be a guess too https://github.com/gtxaspec/wz_mini_hacks/discussions/122

unoukujou, 2024-02-23:
> I found this other thread that suggests it keeps about 1GB free, but that might just be a guess too #122

Good to see it has been asked. I don't find a definite answer though. The last sentence in this comment https://github.com/gtxaspec/wz_mini_hacks/discussions/122#discussioncomment-3014077 is exactly my concern. If the OS has no free space, files will get corrupted eventually.

Another question I have is if it is possible to stream the saved SD card video via RTSP? That would be useful as I find the wyze app is laggy for me.