gtxaspec/wz_mini_hacks #19: PanV2:  Vertical Tilt only not stable

gtxaspec, 2022-04-24:
Using our custom kernel, Vertical Tilt on the PANv2 is unstable...unless you have the motor speed in the app set to 9, it barely moves.  Confirmed this is not a HW issue, using the stock kernel it works fine.

Horizontal movement works fine.

Is this an I2C problem?  Need to investigate how the motor kernel module interfaces with the kernel.  The I2C settings in our kernel config match the stock config supplied by wyze...

Grntrenchman, 2023-08-08:
Outside of the app, trying to move using "cmd move" works fine for horizontal, but hangs for vertical, and Ctrl-C leaves it in an error state (running again produces "error"..)

I'm not sure if it's possible to use the higher motor speed from here (without the app), or where it's set.

gtxaspec, 2023-11-14:
found out that PB_27 is claimed by the kernel for MMC detect, which is not applicable to the Panv2, this should be fixed soon by a kernel patch.

Tested on a Pan v2, vertical tilt is super smooth now.

wokawoka, 2024-04-02:
> found out that PB_27 is claimed by the kernel for MMC detect, which is not applicable to the Panv2, this should be fixed soon by a kernel patch.
> 
> Tested on a Pan v2, vertical tilt is super smooth now.

Has this problem already been resolved with the last public version of wz_mini_hacks ?
Not long ago I tried a recent version of wz_mini_hacks and I was still  experiencing the problem on my PAN V2.
Thank you!

gtxaspec, 2024-11-05:
not feasable to fix on stock firmware; fixed in thingino

closed 2024-11-05 by gtxaspec

wokawoka, 2024-11-05:
> not feasable to fix on stock firmware; fixed in thingino

thank you for your answer