gtxaspec/wz_mini_hacks #20: Possible to setup this camera on wifi without a Wyze app account?

westerncanada, 2022-04-24:
I'm wondering if this is possible to setup without a Wyze account.
I tried creating a wpa_supplicant.conf and placing it inside the /wz_mini/etc/ directory but no joy.

endertable, 2022-04-25:
What you are doing won't work, per se, because the network initialization is done by the Wyze binaries and they create  wpa_supplicant.conf off of files in the /configs directory.   Just for the fun of it I did the following with a fresh out of the box cam:

1.  Copied /configs/.user_config from another cam onto the SD card into the \wz_mini directory and named it usrconfig
2.  Populated my  \wz_mini\etc\ssh\authorized_keys  file (for ssh access)
3.  Added the following to the bottom of the run_mmc.sh fille:

if ! [ -f /configs/.user_config ]
then
    echo -n "mySID" > /configs/.wifissid
    echo -n "myPasswd" > /configs/.wifipasswd
    rm -f /configs/.wifipasswd.enc
    cp /media/mmc/wz_mini/usrconfig /configs/.user_config
    echo "1650922495" > /configs/.time_checkpoint
    reboot
fi

Voila!!  After reboot, it came up on the first try. Of course, put your SID and Password.  Side effects:  It was not recording video and the RTSP was YIKES.
BUT, in all fairness, the cam was on FW 4.36.3.18 (this was literally out of the box).  I believe the old firmwares wanted a date from NTP before they would record video, but I think the newer firmwares will use /configs/.time_checkpoint to get a date AND trigger iCamera to record?!?  I believe this RTSP on this hack was also built off the new firmwares video so will probably have better results on everything once you upgrade the FW.  I'm not a fan of any RTSP, I would use fiveLeave h264 hack for NVR recording and for live streaming view I use the app on iPad (4 cams on one screen) or their Web viewer which I have great results with.

Let me know how it goes...
ender

virmaior, 2022-05-05:
at least from 4.36.2.5, it uses /configs/.time_checkpoint
but notably that file does not update and matches whatever time it was at firmware compilation (consequently for reboots, it will revert to that time until it connects to wyze's services and updates the time).

gtxaspec, 2022-05-18:
This would require not using iCamera, and developing a custom interface to the camera hardware, which is outside of the scope of this project for now.

closed 2022-05-18 by gtxaspec