gtxaspec/wz_mini_hacks #21: USB hubs/switches/doohickes for USB CDC for multiple cameras

virmaior, 2022-04-27:
Working off of:

> - USB gadget support, connect the camera directly to a supported router to get an internet connection, no USB Ethernet Adapter required, using USB CDC_NCM

I've been trying to figure out if it'd be possible to switch all of my cameras to using a USB hub with ethernet over usb functionality.

I can find the spec but I can't figure out if any hub supports this (i.e. plug the hub directly into a RPI and give each port CDC-based ethernet over USB).

Can anyone point me in the right direction?

gtxaspec, 2022-04-27:
I can test this for you, I'll plug a hub into my router and a few cameras to the hub

gtxaspec, 2022-04-28:
@virmaior yes this works, I tested using 3 V3 cameras plugged into a USB 3 hub pluuged into a router running openwrt 21.02.  

Plug the cameras in, make sure you use the usb_direct kernel, add the USB interfaces to the bridge of br-lan, save the config, then power cycle the camera's and they grab an IP automatically.   if you wan't them to have static IP's make sure you set the unique MAC's in run_mmc.sh or they use a random mac on every boot.

virmaior, 2022-04-28:
that is amazing!

looks like I will be buying a powered usb hub to switch my 5 cameras over to this! (this might take quite a while for me to get to due to life circumstances -- but when I do I will write it up)

gtxaspec, 2022-05-02:
great to hear, looking forward to reading!

closed 2022-05-02 by gtxaspec

f4llenz, 2023-04-30:
Hi, I've been trying to set this up but I can't seem to make it work. This is how my br-lan interface looks like:

[image]

When I power cycle the camera, I get this on the kernel log:

```
[10708.768540] usb 1-1.3: new high-speed USB device number 8 using ehci-platform
[10708.948943] cdc_ncm 1-1.3:1.0: MAC-Address: 02:3c:e1:db:7d:cc
[10708.950101] cdc_ncm 1-1.3:1.0 usb0: register 'cdc_ncm' at usb-1b000000.usb-1.3, CDC NCM, 02:3c:e1:db:7d:cc
[10708.959149] br-lan: port 4(usb0) entered blocking state
[10708.959163] br-lan: port 4(usb0) entered disabled state
[10708.959526] device usb0 entered promiscuous mode
[10708.960324] cdc_ncm 1-1.3:1.0 usb0: 425 mbit/s downlink 425 mbit/s uplink
```
usb_direct is enabled on the camera, I have all the usb cdc_ncm kernel modules enabled on my openwrt router. I'm using a data capable micro usb cable. Is there something I'm missing?

virmaior, 2023-05-01:
> Hi, I've been trying to set this up but I can't seem to make it work. This is how my br-lan interface looks like:
> 
> usb_direct is enabled on the camera, I have all the usb cdc_ncm kernel modules enabled on my openwrt router. I'm using a data capable micro usb cable. Is there something I'm missing?

I can probably look at it in about 12 hours and report back on how my RPI is configured to do this.

Things off the top of my head:
1. it's necessary to set the MAC address for the USB-direct device on the camera
2. As you're doing, it's necessary to bridge the USB-direct network on the router/server side
3. As you're doing, it's necessary to have a data-cable micro usb cable
4. Presumably your openWRT router can give enough juice vis microusb or else we wouldn't be seeing kernel log entries

Questions
1. How are you assigning the IP address for the USB direct device?
2. Was the camera previously set up?
3. this is a WyzeV3?
4. What firmware is the wyze running and what version wz_mini_hacks ?