gtxaspec/wz_mini_hacks #23: MQTT Support for homeassistant integration

ankurvdev, 2022-04-28:
What would it take to add MQTT support 

Publishing and subscribing to MQTT bindings will make 

Looking at something similiar in Dafang-Hacks 
It seems they have a binary reading state from shared memory that fires off a shell script on motion detection

https://github.com/EliasKotlyar/Xiaomi-Dafang-Software/blob/1cfed9e1735c3bd376f86195318a9067161cb3d3/v4l2rtspserver-master/src/ImpEncoder.cpp#L723

gtxaspec, 2022-04-29:
will need some help on this one...

Vendo232, 2022-05-05:
I vote for MQTT, it would be very cool feature which would turn V3 into sensor device.

gtxaspec, 2022-05-07:
With the latest build, if you enable rtsp, ```cmd``` becomes available:

```cmd waitmotion <timeout>```

so, cmd waitmotion 1000 waits for motion, for 1000 seconds, then exits once motion is detected.  it outputs the coordinates of the motion detected. 

you could so something like:
```while true; do cmd waitmotion 1000; sleep 0.1; done``` to get continuous output for use in a script.  Checkout the sources in libcallback if you want more info on how it works.

Matthew-Whitlock, 2022-05-25:
A workaround to full MQTT support is to use the webhook trigger in homeassistant (which I think makes more sense anyway) - I figured I'd post it here, as it took a little tinkering:

```
#!/opt/wz_mini/bin/bash

while true
do
  ret=`cmd waitmotion 1000 | grep "timeout" | wc -l`

  if [ $ret -eq 0  ] ; then
    echo "Reporting motion!"
    wget --post-data="motion=1" http://your.homeassistant.com/api/webhook/your_webhook -O/dev/null 2>/dev/null
  else
    #optionally actively report timeouts
    echo "No motion found!"
  fi
done
```

An important note is that the version of wget provided doesn't support https, so you'll have to expose http access if you aren't now. The post-data isn't important, and you can send any number of var1=data&var2=data2 within, which can be accessed from homeassistant.

gtxaspec, 2022-05-25:
i'll try to compile a static version of wget with https, i forgot about that.

gtxaspec, 2022-05-25:
wget with https support will be included in next release!

gtxaspec, 2022-05-25:
release out now: ```/opt/wz_mini/bin/wget```