gtxaspec/wz_mini_hacks #24: usb-storage devices don't work, no support in kernel

gtxaspec, 2022-04-29:
usb-storage is missing from kernel, unable to use USB flash drives/ssd/hdd as storage.  
need to include usb-storage, probably scsi subsystem too.

capncybo, 2022-04-29:
In your opinion...
is adding "usb-storage"  going to be Difficult or Very time consuming to implement?
Or do you think it should be doable & well within your current knowledge-level & existing skill-set?

gtxaspec, 2022-04-29:
it's already done, lol.  I'll have an updates release within the day.

capncybo, 2022-04-29:
OMG, wow.  Your answer certainly exceeded my expectations... Very impressive,  Looking forward to your update.
And Hopefully, Many many more,
Awesome work!!!

gtxaspec, 2022-04-30:
new version including support now released.

closed 2022-04-30 by gtxaspec

capncybo, 2022-04-30:
Just Flashed the NEW version... Super Job!!!
IMO The audio playback is a really professional addition.
+
It worked 1st attempt even though I decided to try my BIGGER 128G SD-CARD.
Awesome stuff,
Thx & keep up the Jaw Dropping Work ;-)