gtxaspec/wz_mini_hacks #25: USB OTG Doesn't power HL_PANv2

gtxaspec, 2022-04-30:
Curiously, using a USB OTG Cable, or a microusb hub with power pass thru, doesn't power the PanV2 at all.  

Inspecting the unit inside, the connector has 5 wires coming from the base to the board.  if the ID pin is shorted for OTG, could this be preventing the camera from booting?

gtxaspec, 2022-05-01:
Looks like I was right!  Removing the ID pin does allow the USB OTG cable / adapter / hub to power the device.  The USB devices work too!

If you have a PANv2 and you want to use USB OTG, for a HUB, flash drive, ethernet adapter or whatever, see the attached photos.

You need to remove the 2nd red cable from the connector, or just snip it and tape the ends off.  It is not needed.

gtxaspec, 2022-05-01:
[image]
[image]

capncybo, 2022-05-01:
Thank you for sharing this information, this makes me really ant to purchase one.

closed 2022-05-07 by gtxaspec