gtxaspec/wz_mini_hacks #27: scp - not found, (Maybe could be added ???) + Best method to upgrade firmware

capncybo, 2022-05-01:
I flashed one v3 Camera's SD-Card (& of course... you upgraded about an hour afterwards).
But rather than.. Get the ladder-out again, pull down the camera, remove SD-Card etc etc...
I thought...
Perhaps I can just use ssh & scp the most recently altered files?
Obviously I soon noticed there was no scp but that led to me thinking...
 'What is the easiest or best method to remotely update your firmware,  WITHOUT removing the SD-Card from EACH Camera???"
I noticed there was already wget but thought I'd ask the author himself.
+ I did already make some changes remotely to certain files via ssh & using nano...
but what if the changes require copying some binary files?
Any wisdom would be appreciated.
Thx

sbn-purchark, 2022-05-01:
<code>ssh user@host "cat > remote" < file</code>

[How to copy a file without using scp inside an ssh session?](https://superuser.com/questions/291423/how-to-copy-a-file-without-using-scp-inside-an-ssh-session)

capncybo, 2022-05-01:
> `ssh user@host "cat > remotefile" < file`
> 
> [How to copy a file without using scp inside an ssh session?](https://superuser.com/questions/291423/how-to-copy-a-file-without-using-scp-inside-an-ssh-session)

Cool tip, thanks for sharing

gtxaspec, 2022-05-02:
scp is already included in the 'dropbearmulti' executable.  I will add a symlink in PATH to enable this.

you can do this on an already installed system by running:

```ln -s /opt/wz_mini/bin/dropbearmulti /opt/wz_mini/tmp/.bin/scp```

gtxaspec, 2022-05-02:
latest repo has the individual dropbear binaries, let me know if it works for you.

capncybo, 2022-05-02:
Wow that was quick, you really put the ...zing in: Ama-zing ;-) 
It worked perfectly, tested via:

scp root@192.168.50.22:/opt/factory_t31_ZMC6tiIDQN factory_t31_ZMC6tiIDQN
+
scp test_file root@192.168.50.22:/opt/test_file
& recursively via:
scp -r root@192.168.50.22:/opt/wz_mini/etc/init.d testdir

Thx again

capncybo, 2022-05-02:
scp (working properly) closing issue