gtxaspec/wz_mini_hacks #28: Problems getting started. Nothing is executing.

a904guy, 2022-05-01:
Camera:
Wyze Pan V2
Firmware: 4.49.1.20 and tried the original firmware the device had out of the box.

I've copied the SD_ROOT to the SD card (exFAT Format)

SD Card Specs: ScanDisk Ultra Micro SDxc I, Class 10 1u (~128gb)

Fresh Install Filesystem looked like:
/wz_mini/
/factory_t31_ZMC6tiIDQN

Letting the camera boot, I'm able to access from the wyze app, see it's recognized the micro card, recording is turned off.
Nothing happens that is supposed too. Port scan (nmap, and nettools on android) shows no port 22, and attempting to connect is futile. 

I've even modified the /wz_mini/run_mmc.sh script to try and see if it was executing at all. I put a `curl https://myserver.com/hello-from-wyze;` above the syslog message command `echo  "run_mmc.sh start" > /dev/kmsg`. The curl call never made it. (Assuming busybox has curl enabled)

After boot and examining the drive again the usual suspects show up file wise.
/log/...txt
/time_lapse/
/wz_mini/
/factory_t31_ZMC6tiIDQN
/time.ini

I'm not sure how to get a serial command line going, I connected the camera over USB to my machine, and neither Putty or screen could make a serial connection.

This is cool development for these cameras. and am really excited for the work that has been done by everyone to get here. 

I'm unsure how to proceed / debug what is wrong.
Any insights / hints in to how to get started would be greatly appreciated. Wrong partition type?

gtxaspec, 2022-05-02:
Hi, I haven't tested with that firmware version, try updating to the latest one and running it.  It's safe to update.  Also, the memory card typically must be formatted as FAT32, I havent tested exfat.  

I will test this now.  stand by

gtxaspec, 2022-05-02:
The bootloader does not support exFAT, so the custom kernel never gets loaded.  At the moment, formatting the memory card with fat32 is a requirement.  If you need exFAT, I would suggest making a small partition, at least 256mb fat32, then a second partition as exfat.  You could then mount the exfat partition to store recordings or other stuff.

a904guy, 2022-05-02:
Thank you! I'll format FAT32 and report back.

a904guy, 2022-05-02:
That did it, on boot I'm hearing the TTS voice.

Thank you!

a904guy, 2022-05-02:
SSH is available. Awesome.

closed 2022-05-02 by a904guy

gtxaspec, 2022-05-02:
glad this worked!  have fun.