gtxaspec/wz_mini_hacks #30: tailscale support

gtxaspec, 2022-05-07:
a few users have mentioned tailscale support.  Wireguard is built in, and they have mipsle binaries available:

endertable, 2022-05-07:
I went to the website to see what tailscale is and it looks like some sort of VPN software?  It looks interesting, but I was just thinking in my mind what and or how would this apply to these cams?  Remote connections to the cam from outside of your home?
Just curious. 😀

gtxaspec, 2022-05-07:
for example, you tether the camera to your phone, or use a public wifi or guest network, you would use the VPN to connect back to your home network to save videos or photos etc.

gtxaspec, 2022-05-07:
yes it does work, though the ```tailscaled``` binary seems to need ```iptables``` which isn't included yet.  for now a fake file like:

```
#!/bin/sh
echo "iptables v1.8.7 (nf_tables): no command specified"
```
at /opt/wz_mini/bin/iptables

and launching the daemon with the args: ```tailscaled -statedir /opt/wz_mini/tmp``` and then running ```tailscale up``` 

enables connectivity.  the wireguard kernel module looks like it's not required for tailscale to work.

gtxaspec, 2022-05-08:
iptables is now included in the latest release, you can run tailscale with:

```tailscaled -statedir /opt/wz_mini/tmp``` and then running ```tailscale up```

closed 2022-05-08 by gtxaspec

endertable, 2022-05-08:
Nice work!!  And I had just perfected a poor mans VPN with an ssh tunnel, hehe.  I take it the iPhone has a client for this tailscale prog?  Or am I looking at this wrong?

gtxaspec, 2022-05-08:
yes, apparently tailscale clients are available for almost every platform.

akivacp, 2023-02-01:
I successfully got this to work in discussion https://github.com/gtxaspec/wz_mini_hacks/discussions/409