gtxaspec/wz_mini_hacks #31: Camera will not boot with wz_mini_hacks

brentfraser, 2022-05-07:
First of all, thanks for the great project!  I have a new Wyze Cam v3 (MAC D0:3F:27:xx:xx:xx), but it will not boot with wz_mini_hacks (commit 26af3aa). It shows the red light for 5 seconds then blinks, shows the red light etc.

I used a16 GB microsd card, formatted fat32, 
- added the content of SD_ROOT so the root of the card contains
```
    wz_mini (folder)
    factory_t31_ZMC6tiIDQN
```
- set RTSP_ENABLED="true" in run_mmc.sh
- turned off the camera, inserted the microSD card, turned on camera

but it just does the red light , blink 5 second cycle.  After 10 minutes, I turned it off.

Am I missing something?   Thanks!

brentfraser, 2022-05-07:
It seems Windows 10 was the problem.  I had formatted the micro SD card on Windows 10 (as fat32).  After the camera boot problem, I moved the micro SD card on to an Ubuntu machine and deleted all the files (including the pesky "System Volume Information" folder).  Then using Ubuntu I copied the SD_ROOT contents on to the sd card.  

The camera now boots!

brentfraser, 2022-05-07:
It works great!  I've tested the RTSP stream with clients on Windows (VLC), Ubuntu (VLC, ffplay), and on a QNAP NAS.  All good.

closed 2022-05-07 by brentfraser

gtxaspec, 2022-05-07:
great to hear!

crossan007, 2024-02-18:
I had the same experience.  Windows 10 prepared SD card caused my V3 camera to go into a "red light" boot loop.

Preparing the SD Card using a raspberry pi seems to have created a usable wz_mini_hacks card 😎 

> It seems Windows 10 was the problem. I had formatted the micro SD card on Windows 10 (as fat32). After the camera boot problem, I moved the micro SD card on to an Ubuntu machine and deleted all the [...]

zsmitchell, 2024-06-15:
For anybody else that is struggling with this...what worked for me was formatting my SD card OUTSIDE of windows (I used the download from SDCARD.org) and it booted right up after that.