gtxaspec/wz_mini_hacks #32: Doorbell (non pro) compatability

gtxaspec, 2022-05-09:
2k image sensor

- resolution is not 2K, currently scaled to 1728x1296
- `############## framesource 0 ###############
chan status: running
output pixformat: NV12
output resolution: 1728 * 1296
scaler : enable
scaler width: 1728
scaler height: 1296
crop : disable
the state of buffers:
queue count: 3
queue addr: 0x06301100
queue addr: 0x06635300
queue addr: 0x06969500
done count: 0
the output buffers is: 0
the losted buffers is: 0
ch0_pre_dequeue_drop is: 0
ch0_pre_dequeue_intc_ahead_cnt is: 0
pdq cnt: 0, pdq intc cnt:0, pdq intc cnt1:0
---------------------------
buf:0, state:3, buf_state:30, buf_qcnt:342, buf_dqcnt:341
buf:1, state:3, buf_state:30, buf_qcnt:342, buf_dqcnt:341
buf:2, state:3, buf_state:30, buf_qcnt:342, buf_dqcnt:341
`
`****************** ISP INFO **********************
Software Version : H20210615a
SENSOR NAME : sc4236
SENSOR OUTPUT WIDTH : 2048
SENSOR OUTPUT HEIGHT : 1536
ISP OUTPUT FPS : 20 / 1
SENSOR OUTPUT RAW PATTERN : BGGR
ISP Top Value : 0xb570224d
ISP Runing Mode : Night
ISP Custom Mode : Disable
ISP WDR Mode : Disable
SENSOR Integration Time : 1596 lines
SENSOR Max Integration Time : 1596 lines
SENSOR analog gain : 48
MAX SENSOR analog gain : 125
SENSOR digital gain : 0
MAX SENSOR digital gain : 0
ISP digital gain : 0
MAX ISP digital gain : 64
ISP Tgain DB : 1
ISP EV value: 4646
ISP EV value log2: 798358
ISP EV value us: 49875
ISP EV min int: 3
ISP EV min again: 1024
ISP WB weighted rgain: 245
ISP WB weighted bgain: 356
ISP WB color temperature: 3010
ISP AWB Start rgain 175: bgain 239
Saturation : 128
Saturation : 128
Sharpness : 128
Contrast : 128
Brightness : 128
Antiflicker : 0
Mirror: Disable, Flip: Disable
debug : ch0 done 1403,ip done 1404,0,0,0,0,0,0
debug1 : 0,0,900
`

gtxaspec, 2022-05-09:
@bakueikozo, I read on your blog that you also own a doorbell.  Were you able to do any modifications to it?

Thanks!

gtxaspec, 2022-07-13:
boot log
```
U-Boot SPL 2013.07 (Aug 24 2020 - 09:59:00)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=116 nr = 1 od0 = 1 od1 = 2
cppcr is 07405100
CPM_CPAPCR 0740510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPMPCR 0640510d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a7b5510
apll_freq 1392000000 
mpll_freq 1200000000 
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 600000000
cclk  1392000000
l2clk 696000000
h0clk 240000000
h2clk 240000000
pclk  120000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 0000001e
T31_194: 0000001c
jz-04 :  0x00000051 
jz-08 :  0x000000a0 
jz-28 :  0x00000024 
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
DDRC_STATUS         0x80000001
DDRC_CFG            0x0ae88a42
DDRC_CTRL           0x0000891c
DDRC_LMR            0x00400008
DDRC_DLP            0x00000000
[... 227 lines cut ...]
[    0.644661] NET: Registered protocol family 17
[    0.649885] input: gpio-keys as /devices/platform/gpio-keys/input/input0
[    0.657011] drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
[    0.667645] VFS: Mounted root (squashfs filesystem) readonly on device 31:2.
[    0.675397] Freeing unused kernel memory: 224K (804d8000 - 80510000)
mdev is ok......
No version info

hualai login: [    2.811022] @@@@ tx-isp-probe ok(version H20210615a), compiler date=Jun 18 2021 @@@@@
[    2.846324] jz_pwm_probe[218] d_name = tcu_chn0
[    2.851410] jz_pwm_probe[218] d_name = tcu_chn2
[    2.857981] jz_pwm_probe[218] d_name = tcu_chn3
[    2.864390] The version of PWM driver is H20180309a
[    2.875722] request pwm channel 0 successfully
[    2.880310] request pwm channel 2 successfully
[    2.886528] request pwm channel 3 successfully
[    2.892970] pwm-jz pwm-jz: jz_pwm_probe register ok !
[    2.967183] jz_codec_register: probe() successful!
[    3.374883] dma dma0chan24: Channel 24 have been requested.(phy id 7,type 0x06 desc a064d000)
[    3.384059] dma dma0chan25: Channel 25 have been requested.(phy id 6,type 0x06 desc a13cf000)
[    3.393326] dma dma0chan26: Channel 26 have been requested.(phy id 5,type 0x04 desc a0762000)
[    3.662375] RTL871X: module init start
[    3.666312] RTL871X: rtl8189ftv v4.3.24.7_21113.20170208.nova.1.01
[    3.672687] RTL871X: build time: Jul 13 2020 20:01:05
```

gtxaspec, 2022-07-13:
possible microSD interface
[image]

endertable, 2022-07-14:
Nice meticulous breakdown of the doorbell.  It's good they find to keep they're hardware very similar.  Also makes it easier for us.  :)

gtxaspec, 2023-07-25:
iCamera is forever broken on this device.  Moving towards migrating this device to OpenIPC.  Test build working at 2k res.

closed 2023-07-25 by gtxaspec