gtxaspec/wz_mini_hacks #34: V2 compatibility

gtxaspec, 2022-05-10:
uboot sources support loading kernel factory file, but is disabled on production models.

also, compiling uboot from source doesn't boot on camera, need to reflash via spi programmer.

hmm... has anyone compiled uboot successfully for the v2?

endertable, 2022-05-11:
You may have already seen this and it might not offer you anything but I just want to make sure since he dealt mostly with the V2:

https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/hacks/flashinguboot.md

gtxaspec, 2022-05-11:
yep thats the one I tried, and the one that bricked my camera.  Luckily I have the tools and know how to repair =D

endertable, 2022-05-11:
I remember you asking @BiatuAutMiahn or was his project specific to the V3 only?

gtxaspec, 2022-05-17:
v3 only, still looking for someone that has a working v2 bootloader from dafang

gtxaspec, 2022-05-17:
kernel compiled OK, tested good.  right now installation would have to reflash the camera, so not as easy as the V3s.  If we had a working u-boot, flash once and then boot from the sd card.

gtxaspec, 2022-05-24:
latest release includes v2 support.

closed 2022-05-24 by gtxaspec

endertable, 2022-05-24:
Fantastic work!!  Thanks for all your efforts!