gtxaspec/wz_mini_hacks #35: RTSP - low bitrate

gtxaspec, 2022-05-12:
researching

gtxaspec, 2022-05-13:
@vendo232 what are you using to measure the bitrate?

gtxaspec, 2022-05-13:
Can you try to `ffmpeg -i <file` to compare between RTSP fw and wz_mini?

this is what i show from wz_mini:

```
  Duration: 00:01:00.05, start: 0.000000, bitrate: 1059 kb/s
  Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080, 927 kb/s, 20 fps, 20 tbr, 1200k tbn, 40 tbc (default)
```

gtxaspec, 2022-05-13:
there may be a bug in the v4l2rtspserver when audio is enabled the bitrate is much lower and lots of stream errors, at least for me.

i will try to recompile, and also test a different version of the rtspserver.

gtxaspec, 2022-05-14:
this is fixed now

closed 2022-05-14 by gtxaspec

Vendo232, 2022-05-14:
> Can you try to `ffmpeg -i <file>` to compare between RTSP fw and wz_mini?
> 
> this is what i show from wz_mini:
> 
> ```
> Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'test.mp4':
>   Metadata:
>     major_brand     : isom
>     minor_version   : 512
>     compatible_brands: isomiso2avc1mp41
>     title           : LIVE555 Streaming Media v2022.04.15
>     encoder         : Lavf58.76.100
>     comment         : LIVE555 Streaming Media v2022.04.15
>   Duration: 00:00:34.06, start: 0.000000, bitrate: 982 kb/s
>   Stream #0:0(und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080, 980 kb/s, 19.99 fps, 20 tbr, 90k tbn, 40 tbc (default)
>     Metadata:
>       handler_name    : VideoHandler
>       vendor_id       : [0][0][0][0]
> ```
> 
> you can use the command `ffmpeg -i rtsp://user:pass@ip:8554/unicast -acodec copy -vcodec copy test.mp4` to record

here is the RTSP WZ HACK file

' ffmpeg version 2022-05-12-git-30e2bb0f64-full_build-www.gyan.dev Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 11.3.0 (Rev1, Built by MSYS2 project)
  configuration: --enable-gpl --enable-version3 --enable-static --disable-w32threads --disable-autodetect --enable-fontconfig --enable-iconv --enable-gnutls --enable-libxml2 --enable-gmp --enable-bzlib --enable-lzma --enable-libsnappy --enable-zlib --enable-librist --enable-libsrt --enable-libssh --enable-libzmq --enable-avisynth --enable-libbluray --enable-libcaca --enable-sdl2 --enable-libdav1d --enable-libdavs2 --enable-libuavs3d --enable-libzvbi --enable-librav1e --enable-libsvtav1 --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxavs2 --enable-libxvid --enable-libaom --enable-libjxl --enable-libopenjpeg --enable-libvpx --enable-mediafoundation --enable-libass --enable-frei0r --enable-libfreetype --enable-libfribidi --enable-liblensfun --enable-libvidstab --enable-libvmaf --enable-libzimg --enable-amf --enable-cuda-llvm --enable-cuvid --enable-ffnvcodec --enable-nvdec --enable-nvenc --enable-d3d11va --enable-dxva2 --enable-libmfx --enable-libshaderc --enable-vulkan --enable-libplacebo --enable-opencl --enable-libcdio --enable-libgme --enable-libmodplug --enable-libopenmpt --enable-libopencore-amrwb --enable-libmp3lame --enable-libshine --enable-libtheora --enable-libtwolame --enable-libvo-amrwbenc --enable-libilbc --enable-libgsm --enable-libopencore-amrnb --enable-libopus --enable-libspeex --enable-libvorbis --enable-ladspa --enable-libbs2b --enable-libflite --enable-libmysofa --enable-librubberband --enable-libsoxr --enable-chromaprint
  libavutil      57. 24.101 / 57. 24.101
  libavcodec     59. 28.100 / 59. 28.100
  libavformat    59. 23.100 / 59. 23.100
  libavdevice    59.  6.100 / 59.  6.100
  libavfilter     8. 38.100 /  8. 38.100
  libswscale      6.  6.100 /  6.  6.100
  libswresample   4.  6.100 /  4.  6.100
  libpostproc    56.  5.100 / 56.  5.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'b.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.45.100
  Duration: 00:02:26.03, start: 0.000000, bitrate: 287 kb/s
  Stream #0:0[0x1](und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709, progressive), 1920x1080, 285 kb/s, 20 fps, 20 tbr, 10240 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0] '

and here is from wyze rtsp fw

' ffmpeg version 2022-05-12-git-30e2bb0f64-full_build-www.gyan.dev Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 11.3.0 (Rev1, Built by MSYS2 project)
  configuration: --enable-gpl --enable-version3 --enable-static --disable-w32threads --disable-autodetect --enable-fontconfig --enable-iconv --enable-gnutls --enable-libxml2 --enable-gmp --enable-bzlib --enable-lzma --enable-libsnappy --enable-zlib --enable-librist --enable-libsrt --enable-libssh --enable-libzmq --enable-avisynth --enable-libbluray --enable-libcaca --enable-sdl2 --enable-libdav1d --enable-libdavs2 --enable-libuavs3d --enable-libzvbi --enable-librav1e --enable-libsvtav1 --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxavs2 --enable-libxvid --enable-libaom --enable-libjxl --enable-libopenjpeg --enable-libvpx --enable-mediafoundation --enable-libass --enable-frei0r --enable-libfreetype --enable-libfribidi --enable-liblensfun --enable-libvidstab --enable-libvmaf --enable-libzimg --enable-amf --enable-cuda-llvm --enable-cuvid --enable-ffnvcodec --enable-nvdec --enable-nvenc --enable-d3d11va --enable-dxva2 --enable-libmfx --enable-libshaderc --enable-vulkan --enable-libplacebo --enable-opencl --enable-libcdio --enable-libgme --enable-libmodplug --enable-libopenmpt --enable-libopencore-amrwb --enable-libmp3lame --enable-libshine --enable-libtheora --enable-libtwolame --enable-libvo-amrwbenc --enable-libilbc --enable-libgsm --enable-libopencore-amrnb --enable-libopus --enable-libspeex --enable-libvorbis --enable-ladspa --enable-libbs2b --enable-libflite --enable-libmysofa --enable-librubberband --enable-libsoxr --enable-chromaprint
  libavutil      57. 24.101 / 57. 24.101
  libavcodec     59. 28.100 / 59. 28.100
  libavformat    59. 23.100 / 59. 23.100
  libavdevice    59.  6.100 / 59.  6.100
  libavfilter     8. 38.100 /  8. 38.100
  libswscale      6.  6.100 /  6.  6.100
  libswresample   4.  6.100 /  4.  6.100
  libpostproc    56.  5.100 / 56.  5.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'b.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.45.100
  Duration: 00:02:26.03, start: 0.000000, bitrate: 287 kb/s
  Stream #0:0[0x1](und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, bt709, progressive), 1920x1080, 285 kb/s, 20 fps, 20 tbr, 10240 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0] '

Vendo232, 2022-05-14:
you can achieve the same in VLC player

[image]

Vendo232, 2022-05-14:
I`m going to test the new version today

gtxaspec, 2022-05-14:
@Vendo232 can you test something for me on your existing install?

Vendo232, 2022-05-14:
If course

gtxaspec, 2022-05-14:
open up vlc and start watching the stream, then connect with ssh, and run the following commands on the camera:

```
/system/bin/impdbg --enc_rc_s 0:48:4:2048
/system/bin/impdbg --enc_rc_s 0:52:4:2048
```

Vendo232, 2022-05-15:
should I look for bitrate?

gtxaspec, 2022-05-15:
look at the general image quality and see if it is ok for you compared to stock rtsp and anything else

gtxaspec, 2022-05-15:
you can raise the 2048 number up to 8096 if you want

Vendo232, 2022-05-15:
what is the difference between command with 48 and 52? should they be executed toghether everytime?

gtxaspec, 2022-05-15:
first one is average and second is max bitrate

Vendo232, 2022-05-15:
I`m running the RTSP FW + mini RTSP and the cam is very slow.

gtxaspec, 2022-05-15:
Oh, i see, yes you are running 2 servers it will slow down since the quality goes down.  try stopping one, and run the commands and see if your quality goes up

gtxaspec, 2022-05-15:
(the image quality should be identical between the two servers anyway)

gtxaspec, 2022-05-15:
[image]

gtxaspec, 2022-05-15:
[image]

gtxaspec, 2022-05-15:
[image]

gtxaspec, 2022-05-15:
[image]

Vendo232, 2022-05-15:
trying 8096 on both, it is looking really good

[image]

gtxaspec, 2022-05-15:
nice.  i will issue a new release soon