gtxaspec/wz_mini_hacks #36: Stuck at debug1: expecting SSH2_MSG_KEX_ECDH_REPLY

raelbsd, 2022-05-12:
Hi, finally I made a fresh install, in a new SD formatted as FAT. After the boot, I've wait at least 5 minutes to give time to the ssh server to load, and when I try to ssh it gives me this error.

```
juan@calisto:~$ ssh 192.168.1.30 -vvv
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.1.30 is address
debug2: ssh_connect_direct
debug1: Connecting to 192.168.1.30 [192.168.1.30] port 22.
debug1: Connection established.
debug1: identity file /home/juan/.ssh/id_rsa type 0
debug1: identity file /home/juan/.ssh/id_rsa-cert type -1
debug1: identity file /home/juan/.ssh/id_dsa type -1
debug1: identity file /home/juan/.ssh/id_dsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa type -1
debug1: identity file /home/juan/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519 type -1
debug1: identity file /home/juan/.ssh/id_ed25519-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_xmss type -1
debug1: identity file /home/juan/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.2p1 Ubuntu-4ubuntu0.4
debug1: Remote protocol version 2.0, remote software version dropbear_2022.82
debug1: no match: dropbear_2022.82
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to 192.168.1.30:22 as 'juan'
debug3: hostkeys_foreach: reading file "/home/juan/.ssh/known_hosts"
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group14-sha256,ext-info-c
debug2: host key algorithms: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,sk-ecdsa-sha2-nistp256-cert-v01@openssh.com,ssh-ed25519-cert-v01@openssh.com,sk-ssh-ed25519-cert-v01@openssh.com,rsa-sha2-512-cert-v01@openssh.com,rsa-sha2-256-cert-v01@openssh.com,ssh-rsa-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,sk-ecdsa-sha2-nistp256@openssh.com,ssh-ed25519,sk-ssh-ed25519@openssh.com,rsa-sha2-512,rsa-sha2-256,ssh-rsa
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: MACs ctos: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: MACs stoc: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group14-sha256,diffie-hellman-group14-sha1,kexguess2@matt.ucc.asn.au
debug2: host key algorithms: ssh-ed25519,ecdsa-sha2-nistp256,rsa-sha2-256,ssh-rsa,ssh-dss
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes256-ctr
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes256-ctr
debug2: MACs ctos: hmac-sha1,hmac-sha2-256
debug2: MACs stoc: hmac-sha1,hmac-sha2-256
debug2: compression ctos: none
debug2: compression stoc: none
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ecdsa-sha2-nistp256
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug3: send packet: type 30
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
Connection closed by 192.168.1.30 port 22
```
I've tried changing some HostKeyAlgorithms but no luck, do you have any idea?
Thanks!

gtxaspec, 2022-05-12:
I'll test the same version you are using and get back to you

gtxaspec, 2022-05-12:
can you try `ssh -o KexAlgorithms=ecdh-sha2-nistp521 root@192.168.1.30`

raelbsd, 2022-05-13:
This is the output:

```
juan@calisto:~$ ssh -o KexAlgorithms=ecdh-sha2-nistp521 root@192.168.1.30 -vvv
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.1.30 is address
debug2: ssh_connect_direct
debug1: Connecting to 192.168.1.30 [192.168.1.30] port 22.
debug1: Connection established.
debug1: identity file /home/juan/.ssh/id_rsa type 0
debug1: identity file /home/juan/.ssh/id_rsa-cert type -1
debug1: identity file /home/juan/.ssh/id_dsa type -1
debug1: identity file /home/juan/.ssh/id_dsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa type -1
debug1: identity file /home/juan/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519 type -1
debug1: identity file /home/juan/.ssh/id_ed25519-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_xmss type -1
debug1: identity file /home/juan/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.2p1 Ubuntu-4ubuntu0.4
debug1: Remote protocol version 2.0, remote software version dropbear_2022.82
debug1: no match: dropbear_2022.82
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to 192.168.1.30:22 as 'root'
debug3: hostkeys_foreach: reading file "/home/juan/.ssh/known_hosts"
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: ecdh-sha2-nistp521,ext-info-c
debug2: host key algorithms: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,sk-ecdsa-sha2-nistp256-cert-v01@openssh.com,ssh-ed25519-cert-v01@openssh.com,sk-ssh-ed25519-cert-v01@openssh.com,rsa-sha2-512-cert-v01@openssh.com,rsa-sha2-256-cert-v01@openssh.com,ssh-rsa-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,sk-ecdsa-sha2-nistp256@openssh.com,ssh-ed25519,sk-ssh-ed25519@openssh.com,rsa-sha2-512,rsa-sha2-256,ssh-rsa
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: MACs ctos: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: MACs stoc: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group14-sha256,diffie-hellman-group14-sha1,kexguess2@matt.ucc.asn.au
debug2: host key algorithms: ssh-ed25519,ecdsa-sha2-nistp256,rsa-sha2-256,ssh-rsa,ssh-dss
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes256-ctr
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes256-ctr
debug2: MACs ctos: hmac-sha1,hmac-sha2-256
debug2: MACs stoc: hmac-sha1,hmac-sha2-256
debug2: compression ctos: none
debug2: compression stoc: none
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug1: kex: algorithm: ecdh-sha2-nistp521
debug1: kex: host key algorithm: ecdsa-sha2-nistp256
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug3: send packet: type 30
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
Connection closed by 192.168.1.30 port 22

```

gtxaspec, 2022-05-13:
try ssh -o KexAlgorithms=ecdsa-sha2-nistp256 root@192.168.1.30

raelbsd, 2022-05-13:
juan@calisto:~$ ssh -o KexAlgorithms=ecdsa-sha2-nistp256 root@192.168.1.30 -vvv
Unsupported KEX algorithm "ecdsa-sha2-nistp256"
command-line line 0: Bad SSH2 KexAlgorithms 'ecdsa-sha2-nistp256'.

gtxaspec, 2022-05-13:
I'm unable to reproduce, what version of Ubuntu are you using?

raelbsd, 2022-05-13:
I'm using Ubuntu 20.04.4 LTS Focal (Desktop). I have ssh installed from repositories, and this is the version:
```
juan@calisto:~$ ssh -V
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
```

raelbsd, 2022-05-13:
I've tried also to remove the `-s -g` options of dropbear in /opt/wz_mini/init.d/v3_init.sh. After that, this is the output:

```
juan@calisto:~$ ssh root@192.168.1.30 -vvv
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.1.30 is address
debug2: ssh_connect_direct
debug1: Connecting to 192.168.1.30 [192.168.1.30] port 22.
debug1: Connection established.
debug1: identity file /home/juan/.ssh/id_rsa type 0
debug1: identity file /home/juan/.ssh/id_rsa-cert type -1
debug1: identity file /home/juan/.ssh/id_dsa type -1
debug1: identity file /home/juan/.ssh/id_dsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa type -1
debug1: identity file /home/juan/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519 type -1
debug1: identity file /home/juan/.ssh/id_ed25519-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_xmss type -1
debug1: identity file /home/juan/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.2p1 Ubuntu-4ubuntu0.4
kex_exchange_identification: read: Connection reset by peer
juan@calisto:~$ ssh root@192.168.1.30 -vvv
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.1.30 is address
debug2: ssh_connect_direct
debug1: Connecting to 192.168.1.30 [192.168.1.30] port 22.
debug1: connect to address 192.168.1.30 port 22: Connection refused
ssh: connect to host 192.168.1.30 port 22: Connection refused

```
Any try before this gives me "Connection refused". Maybe the dropbear ssh is down after the kex_exchange_identification warning?

Thanks

raelbsd, 2022-05-13:
Ok, I'm not sure where is the issue. I made a rollback of the deleted dropbear options, so now the only change is on the authorized_keys file. I've added another public key on the file, from a rpi4 running Debian 11. To add the key I do a cat xxx.pub >> /path/to/archive/authorized_keys, so I don't add any extra characters or line breaks.
Now, I can login from the the first linux:
```
juan@calisto:~$ ssh root@192.168.1.30 -vvv
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.1.30 is address
debug2: ssh_connect_direct
debug1: Connecting to 192.168.1.30 [192.168.1.30] port 22.
debug1: Connection established.
debug1: identity file /home/juan/.ssh/id_rsa type 0
debug1: identity file /home/juan/.ssh/id_rsa-cert type -1
debug1: identity file /home/juan/.ssh/id_dsa type -1
debug1: identity file /home/juan/.ssh/id_dsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa type -1
debug1: identity file /home/juan/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519 type -1
debug1: identity file /home/juan/.ssh/id_ed25519-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_xmss type -1
debug1: identity file /home/juan/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.2p1 Ubuntu-4ubuntu0.4
debug1: Remote protocol version 2.0, remote software version dropbear_2022.82
debug1: no match: dropbear_2022.82
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to 192.168.1.30:22 as 'root'
debug3: hostkeys_foreach: reading file "/home/juan/.ssh/known_hosts"
debug3: record_hostkey: found key type ECDSA in file /home/juan/.ssh/known_hosts:6
debug3: load_hostkeys: loaded 1 keys from 192.168.1.30
debug3: order_hostkeyalgs: prefer hostkeyalgs: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group16-sha512,diffie-hellman-group18-sha512,diffie-hellman-group14-sha256,ext-info-c
debug2: host key algorithms: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,sk-ecdsa-sha2-nistp256-cert-v01@openssh.com,ssh-ed25519-cert-v01@openssh.com,sk-ssh-ed25519-cert-v01@openssh.com,rsa-sha2-512-cert-v01@openssh.com,rsa-sha2-256-cert-v01@openssh.com,ssh-rsa-cert-v01@openssh.com,sk-ecdsa-sha2-nistp256@openssh.com,ssh-ed25519,sk-ssh-ed25519@openssh.com,rsa-sha2-512,rsa-sha2-256,ssh-rsa
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com
debug2: MACs ctos: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: MACs stoc: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256,curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group14-sha256,diffie-hellman-group14-sha1,kexguess2@matt.ucc.asn.au
debug2: host key algorithms: ssh-ed25519,ecdsa-sha2-nistp256,rsa-sha2-256,ssh-rsa,ssh-dss
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes256-ctr
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes256-ctr
debug2: MACs ctos: hmac-sha1,hmac-sha2-256
debug2: MACs stoc: hmac-sha1,hmac-sha2-256
debug2: compression ctos: none
debug2: compression stoc: none
debug2: languages ctos: 
debug2: languages stoc: 
debug2: first_kex_follows 0 
debug2: reserved 0 
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ecdsa-sha2-nistp256
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug3: send packet: type 30
[... 245 lines cut ...]
debug2: we sent a publickey packet, wait for reply
debug3: receive packet: type 51
[... cut ...] d.
If you need I can continue to help debugging this, or we can close it (it's working somehow :) )

I've arrived to your project because I was trying to use my wyze cams with Frigate, and I couldn't to use the telnet hack on my cams, the project from [Vendo232/wyze_v3_frigate_http_stream](https://github.com/Vendo232/wyze_v3_frigate_http_stream) uses a HTTP stream hack installed from a telnet session. Do you know if this HTTP stream hack is working with wz_mini_hacks? 

Thanks for the hard work!

gtxaspec, 2022-05-13:
@vendo232 would know more 😁

closed 2022-05-13 by raelbsd

Vendo232, 2022-05-14:
the ssh and keys is a mistery to me also. I wish I can learn it one day. here is a guide for Windows/Putty it might give you some clues for the RPI4. https://devops.ionos.com/tutorials/use-ssh-keys-with-putty-on-windows/

In terms of Frigate you don`t need SSH, you have 2 options now.

1. use WZH Mini RTSP server , in this case you would follow @gtxaspec guide in this repo and add path in config.yaml like this.

`      inputs:
        - path: rtsp://admin:pass@192.168.1.6:8554/unicast`

2. Or you can use WYZE RTSP firmware with LAN dongle. I have one camera running like this and it seems very stable. Frigate would look like this

`      inputs:
        - path: rtsp://admin:pass@192.168.1.6/live`

3. You can still install WyzeHacks and HTTP stream, it is still doable. It is a pain, it will take several trials and long setup but it can be done and the HTTP stream is still the most reliable option for V3 over WIFI.

raelbsd, 2022-05-16:
Hi @Vendo232! Sadly I'm still stuck fighting with the ssh access. I was using the version commited 3 days ago, and half of the time I try, the ssh works. I managed to use the Mini RTSP server but I have no response from VLC on my desktop or my phone. I configured too in Frigate but if doesn't work from VLC the only thing I have is a green screen.
I've tried also to clone the repo with the recent additions (looks promising) but I can't manage to get the ssh prompt at all.
Also tried to mix your scripts (just a lucky shoot, I didn't expect to work) in something like your 3rd option but until I didn't solve the ssh access I'm lost, and having the hires and lowres streams it's a must for Frigate I think.

I live nearly in the end of the world, and have a chance to bought a 4-pack cams a year ago, I was looking for the more mature v2 but when the V3s was on my hands in  rush I thought that "ah ...they must to work in my Home Assistant". I didn't knew I'll be enslaved by Wyze and his app, and from then I was trying to escape that jail. My setup is completed with 2 hacked Yi-cams, and I'll like to add a few esp32 cams to this.
I came late with the Wyzehacks, I tried several firmwares, dns spoofing, isolated vlans but the updater do nothing on my cams, and then I landed in this project that gave me a lot of hope.

The steps I'm doing are: cloning the @gtxaspec repo, format the sdcard, copying the SD_ROOT files on the root of sd card, adding the keys and nothing else, I left the run_mmc.sh as is.
When I started the cam I heard the "swap initializing", the cam reboots and then the solid blue light of the wifi connection. When I tried to ssh this is what I got:
```
juan@calisto:~/Documentos$ ssh root@192.168.1.53 -vv
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.1.53 is address
debug2: ssh_connect_direct
debug1: Connecting to 192.168.1.53 [192.168.1.53] port 22.
debug1: Connection established.
debug1: identity file /home/juan/.ssh/id_rsa type 0
debug1: identity file /home/juan/.ssh/id_rsa-cert type -1
debug1: identity file /home/juan/.ssh/id_dsa type -1
debug1: identity file /home/juan/.ssh/id_dsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa type -1
debug1: identity file /home/juan/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk type -1
debug1: identity file /home/juan/.ssh/id_ecdsa_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519 type -1
debug1: identity file /home/juan/.ssh/id_ed25519-cert type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk type -1
debug1: identity file /home/juan/.ssh/id_ed25519_sk-cert type -1
debug1: identity file /home/juan/.ssh/id_xmss type -1
debug1: identity file /home/juan/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.2p1 Ubuntu-4ubuntu0.4
kex_exchange_identification: read: Connection reset by peer
juan@calisto:~/Documentos$ ssh root@192.168.1.53 -vv
OpenSSH_8.2p1 Ubuntu-4ubuntu0.4, OpenSSL 1.1.1f  31 Mar 2020
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: include /etc/ssh/ssh_config.d/*.conf matched no files
debug1: /etc/ssh/ssh_config line 21: Applying options for *
debug2: resolve_canonicalize: hostname 192.168.1.53 is address
debug2: ssh_connect_direct
debug1: Connecting to 192.168.1.53 [192.168.1.53] port 22.
debug1: connect to address 192.168.1.53 port 22: Connection refused
ssh: connect to host 192.168.1.53 port 22: Connection refused
```
Looking on the log folder on the SD, there are 3 log files, the run_mmc.sh shows "sdcard not ready yet" ... why it's stuck here? The sdcard its being used to write this, or it's locked somehow? I've tried with several sdcards, different size, even one new. I've performed surface analysis to avoid the corrupted ones, I'm running out of options.

```
welcome to run_mmc.sh
PID 89
+ HOSTNAME=WCV3
+ ENABLE_USB_ETH=false
+ ENABLE_USB_DIRECT=false
+ USB_DIRECT_MAC_ADDR=02:01:02:03:04:08
+ ENABLE_USB_RNDIS=false
+ ENABLE_IPV6=false
+ ENABLE_WIREGUARD=false
+ ENABLE_IPTABLES=false
+ ENABLE_NFSv4=false
+ REMOTE_SPOTLIGHT=false
+ REMOTE_SPOTLIGHT_HOST=0.0.0.0
+ RTSP_LOGIN=admin
+ RTSP_PASSWORD=
+ RTSP_HI_RES_ENABLED=false
+ RTSP_HI_RES_ENABLE_AUDIO=false
+ RTSP_HI_RES_PORT=8554
+ RTSP_HI_RES_MAX_BITRATE=
+ RTSP_HI_RES_TARGET_BITRATE=
+ RTSP_HI_RES_ENC_PARAMETER=
+ RTSP_LOW_RES_ENABLED=false
+ RTSP_LOW_RES_ENABLE_AUDIO=false
+ RTSP_LOW_RES_PORT=8555
+ RTSP_LOW_RES_MAX_BITRATE=
+ RTSP_LOW_RES_TARGET_BITRATE=
+ RTSP_LOW_RES_ENC_PARAMETER=
+ ENABLE_SWAP=true
+ ENABLE_USB_STORAGE=false
+ ENABLE_EXT4=false
+ ENABLE_CIFS=false
+ DISABLE_FW_UPGRADE=false
+ DEBUG_ENABLED=false
+ first_run_check
+ '[[' -e /opt/wz_mini/tmp/.run_mmc_firstrun ]]
+ wait_sdroot
+ true
+ '[[' -d /media/mmc/wz_mini ]]
+ echo 'sdcard not ready yet...'
sdcard not ready yet...
+ sleep 5
+ true
+ '[[' -d /media/mmc/wz_mini ]]
+ echo 'sdcard not ready yet...'
sdcard not ready yet...
+ sleep 5
+ true
+ '[[' -d /media/mmc/wz_mini ]]
+ echo 'sdcard not ready yet...'
sdcard not ready yet...
+ sleep 5

```
Thanks in advance for any comments!

gtxaspec, 2022-05-17:
can you post your full run_mmc.log, or is that the full log?

raelbsd, 2022-05-17:
That's the full log.

gtxaspec, 2022-05-17:
what firmware version are you running on the camera?

raelbsd, 2022-05-17:
Firmware version: 4.36.0.228

gtxaspec, 2022-05-17:
too old, that is why.  Lots of stuff has changed, upgrade to the latest official firmware, i think its 4.36.9.131

gtxaspec, 2022-05-17:
I am discovering lots of issues with very old firmware, I haven't had time to go that way back.  I think that is one of the initial shipping firmwares