gtxaspec/wz_mini_hacks #37: RTSP and firmware versions

virmaior, 2022-05-14:
wz_mini'd 4.36.2.5 running over USB direct.
with
+ RTSP_ENABLED=true
+ RTSP_ENABLE_AUDIO=true

in run_mmc.log, I'm getting:

VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device  

and no video

if I do 
+ RTSP_ENABLED=true
+ RTSP_ENABLE_AUDIO=false

the same error message and no RTSP

ffmpeg output for video only:

> ffmpeg version a4e1dd6 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib --extra-libs='-lpthread -lm -latomic' --arch=armel --enable-gmp --enable-gpl --enable-libaom --enable-libass --enable-libdav1d --enable-libdrm --enable-libfdk-aac --enable-libfreetype --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopus --enable-librtmp --enable-libsnappy --enable-libsoxr --enable-libssh --enable-libvorbis --enable-libvpx --enable-libzimg --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-mmal --enable-nonfree --enable-omx --enable-omx-rpi --enable-version3 --target-os=linux --enable-pthreads --enable-openssl --enable-hardcoded-tables
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[tcp @ 0x1f91ce0] Connection to tcp://192.168.5.103:8554?timeout=0 failed: Connection refused
rtsp://admin:test@192.168.5.103:8554/unicast: Connection refused

ffmpeg output for video+audio:
ffmpeg version a4e1dd6 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib --extra-libs='-lpthread -lm -latomic' --arch=armel --enable-gmp --enable-gpl --enable-libaom --enable-libass --enable-libdav1d --enable-libdrm --enable-libfdk-aac --enable-libfreetype --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopus --enable-librtmp --enable-libsnappy --enable-libsoxr --enable-libssh --enable-libvorbis --enable-libvpx --enable-libzimg --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-mmal --enable-nonfree --enable-omx --enable-omx-rpi --enable-version3 --target-os=linux --enable-pthreads --enable-openssl --enable-hardcoded-tables
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[udp @ 0x33bf7d0] attempted to set receive buffer to size 393216 but it only ended up set as 360448
[udp @ 0x33cfb10] attempted to set receive buffer to size 393216 but it only ended up set as 360448
Guessed Channel Layout for Input Stream #0.0 : mono
Input #0, rtsp, from 'rtsp://admin:test@192.168.5.103:8554/unicast':
  Metadata:
    title           : LIVE555 Streaming Media v2022.04.15
    comment         : LIVE555 Streaming Media v2022.04.15
  Duration: N/A, start: 0.000000, bitrate: 705 kb/s
  Stream #0:0: Audio: pcm_s16be, 44100 Hz, mono, s16, 705 kb/s

=only an audio feed.

Do I need to upgrade the base version of the firmware or is something else busted?

gtxaspec, 2022-05-14:
if you type `cmd on` do you get any output over ssh?

virmaior, 2022-05-14:
I just gave that a shot and no effect.
after that went ahead and tried
`cmd video off`
and then
`cmd video on`
and then checking for good measure.

output remains as is 

- audio there if audio on (though to be honest I haven't checked the sound itself)
- nothing at all if just rtsp enabled

gtxaspec, 2022-05-14:
can you do
```logread | grep inject```
and 
```logread | grep callback```

virmaior, 2022-05-14:
`logread | grep inject` is blank.
`logread | grep callback` is
> May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 1 cb
May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 2 cb
May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 3 cb
May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 4 cb
May 14 18:50:08 syslog: [sdk,0479]dbg: (local_sdk_set_major_error_callback) Doing...
May 14 18:50:08 iCamera: [sdk,0479]dbg: (local_sdk_set_major_error_callback) Doing...
May 14 18:50:12 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 5 cb
May 14 18:50:12 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 6 cb
May 14 18:50:12 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 7 cb
May 14 15:44:05 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:03:02 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:04:00 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:05:00 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:06:00 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:07:00 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:08:00 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:09:00 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:10:00 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:11:00 iCamera: [record.c:1216](record_stop_finish_callback) in...

on a fresh reboot with just the RTSP enabled (no audio),
`logread | grep inject` is blank again and `logread | grep callback` is
> May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 1 cb
May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 2 cb
May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 3 cb
May 14 18:50:07 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 4 cb
May 14 18:50:08 syslog: [sdk,0479]dbg: (local_sdk_set_major_error_callback) Doing...
May 14 18:50:08 iCamera: [sdk,0479]dbg: (local_sdk_set_major_error_callback) Doing...
May 14 18:50:12 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 5 cb
May 14 18:50:12 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 6 cb
May 14 18:50:12 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 7 cb
May 14 16:03:02 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:13:51 iCamera: [record.c:1216](record_stop_finish_callback) in...
May 14 16:14:01 iCamera: [record.c:1216](record_stop_finish_callback) in...

gtxaspec, 2022-05-14:
strange, i never tested with that firmware version.  i would recommend upgrading to at least 4.36.3.19.

i will do more research on that older version later

virmaior, 2022-05-14:
I will probably upgrade them soon ... but for right now, I'm just glad to have it running without wifi.

Just got the app running again on that camera. will test it against the cron jobs I've got set up for tonight and then move the other cameras to wz_mini 

it's amazing how much easier it is to use them via SSH than telnet

gtxaspec, 2022-05-14:
ssh = good :)

gtxaspec, 2022-05-14:
can you share the output of ``ps w``` with rtsp on?

virmaior, 2022-05-14:
here it is:

>     1 root      0:00 {v3_init.sh} /bin/sh /opt/wz_mini/etc/init.d/v3_init.sh
    2 root      0:00 [kthreadd]
    3 root      0:01 [ksoftirqd/0]
    5 root      0:00 [kworker/0:0H]
    7 root      0:04 [rcu_preempt]
    8 root      0:00 [rcu_bh]
    9 root      0:00 [rcu_sched]
   10 root      0:00 [watchdog/0]
   11 root      0:00 [khelper]
   12 root      0:00 [kdevtmpfs]
   13 root      0:00 [writeback]
   14 root      0:00 [bioset]
   15 root      0:00 [kblockd]
   16 root      0:00 [khubd]
   18 root      0:00 [cfg80211]
   19 root      0:00 [rpciod]
   20 root      0:09 [kworker/0:2]
   21 root      0:00 [kswapd0]
   22 root      0:00 [fsnotify_mark]
   23 root      0:00 [nfsiod]
   24 root      0:00 [crypto]
   40 root      0:00 [deferwq]
   45 root      0:01 [mmcqd/0]
   48 root      0:00 [kworker/0:1H]
   59 root      0:00 [jffs2_gcd_mtd6]
   84 root      0:00 /opt/wz_mini/bin/dropbear -R -s -g
   89 root      0:00 {linuxrc} init
  103 root      0:00 sh /tmp/wyze_hack/run/main.sh run
  104 root      0:00 /sbin/getty -L console 115200 vt100
  138 root      0:03 [irq/37-isp-m0]
  140 root      0:00 [irq/38-isp-w02]
  196 root      0:00 [ksdioirqd/mmc1]
  215 root      0:00 dmon --stderr-redir --max-respawns 20 --environ LD_PRELOAD=libsetunbuf.so /sbin/syslogd -C2048 -n -S -- dslog --priority DEBUG --facility USER sysl
  218 root      0:00 /sbin/syslogd -C2048 -n -S
  219 root      0:00 dslog --priority DEBUG --facility USER syslogd
  222 root      0:00 dmon --stderr-redir --max-respawns 20 --environ LD_PRELOAD=libsetunbuf.so /system/bin/assis -- dslog --priority DEBUG --facility USER assis
  224 root      0:00 dmon --stderr-redir --max-respawns 20 --environ LD_PRELOAD=libsetunbuf.so /system/bin/hl_client -- dslog --priority DEBUG --facility USER hl_client
  226 root      0:01 /system/bin/assis
  227 root      0:00 dslog --priority DEBUG --facility USER assis
  228 root      0:00 dmon --stderr-redir --max-respawns 20 --environ LD_PRELOAD=libsetunbuf.so /system/bin/sinker -- dslog --priority DEBUG --facility USER sinker
  230 root      0:05 /system/bin/hl_client
  231 root      0:00 dslog --priority DEBUG --facility USER hl_client
  232 root      0:00 dmon --stderr-redir --max-respawns 0 --environ LD_PRELOAD=libsetunbuf.so /system/bin/iCamera -- dslog --priority DEBUG --facility USER iCamera
  234 root      0:00 /system/bin/sinker
  235 root      0:00 dslog --priority DEBUG --facility USER sinker
  236 root      9:57 /system/bin/iCamera
  237 root      0:00 dslog --priority DEBUG --facility USER iCamera
  238 root      0:00 dmon --stderr-redir --max-respawns 20 --environ LD_PRELOAD=libsetunbuf.so /system/bin/dumpload -- dslog --priority DEBUG --facility USER dumpload
  240 root      0:00 /system/bin/dumpload
  241 root      0:00 dslog --priority DEBUG --facility USER dumpload
  242 root      0:00 dmon --stderr-redir --max-respawns -1 --environ LD_PRELOAD=libsetunbuf.so /system/bin/timesync -- dslog --priority DEBUG --facility USER timesync
  244 root      0:00 /system/bin/timesync
  245 root      0:00 dslog --priority DEBUG --facility USER timesync
  287 root      0:00 {sysMonitor.sh} /bin/sh /system/bin/sysMonitor.sh
  309 root      1:28 [isp_fw_process]
  358 root      0:00 [RTW_XMIT_THREAD]
  359 root      0:00 [RTW_CMD_THREAD]
  360 root      0:00 [RTWHALXT]
  417 root      0:00 wpa_supplicant -D nl80211 -i wlan0 -c /tmp/wpa_supplicant.conf -B
  488 root      0:00 busybox telnetd
  548 root      0:00 udhcpc -i wlan0 -x hostname WCV3Z3 -p /var/run/udhcpc.pid -b
  580 root      0:00 /opt/wz_mini/bin/dropbear -R -s -g
  581 root      0:00 -sh
  624 root      0:00 /system/bin/kvs_stream
 2517 root      0:00 [kworker/u2:2]
 2905 root      0:00 [kworker/u2:0]
 3037 root      0:00 [kworker/0:0]
 3334 root      0:00 sleep 60
 3358 root      0:00 [kworker/u2:1]
 3399 root      0:00 sleep 2
 3400 root      0:00 ps

gtxaspec, 2022-05-14:
it's possible that the old version of wyzehacks running is interfering, because the iCamera executable is not running the way it should... you should have ```  262 root      1h11 /opt/wz_mini/tmp/.storage/iCamera``` in there too.  Are you using the recent version of wz_mini?

virmaior, 2022-05-14:
I'm using the version 2022-05-12 of wz_mini

maybe it is wyzehacks that is the problem ...

or maybe it's the firmware version

```
ps | grep iCamera
  232 root      0:00 dmon --stderr-redir --max-respawns 0 --environ LD_PRELOAD=libsetunbuf.so /system/bin/iCamera -- dslog --priority DEBUG --facility USER iCamera
  236 root     11:03 /system/bin/iCamera
  237 root      0:00 dslog --priority DEBUG --facility USER iCamera
 3714 root      0:00 grep iCamera
```

gtxaspec, 2022-05-14:
i dont have that FW version and its no longer posted, do you still have the demo.bin?

virmaior, 2022-05-14:
[demo_wcv3_4.36.2.5.bin.zip](https://github.com/gtxaspec/wz_mini_hacks/files/8692523/demo_wcv3_4.36.2.5.bin.zip)

gtxaspec, 2022-05-14:
ok flashing now to test

gtxaspec, 2022-05-14:
can you ```cat /opt/wz_mini/tmp/.storage/rcS``` and share it?

virmaior, 2022-05-14:
#!/bin/sh

# Set mdev
echo /sbin/mdev > /proc/sys/kernel/hotplug
/sbin/mdev -s && echo "mdev is ok......"

echo " __________________________________
|                                  |
|                                  |
|                                  |
|                                  |
| _   _             _           _  |
|| | | |_   _  __ _| |     __ _(_) |
|| |_| | | | |/ _| | |  _ / _| | | |
||  _  | |_| | (_| | |_| | (_| | | |
||_| |_|\__,_|\__,_|_____|\__,_|_| |
|                                  |
|                                  |
|_____2020_WYZE_CAM_V3_@HUALAI_____|
"
set -x

# create console and null node for nfsroot
#mknod -m 600 /dev/console c 5 1
#mknod -m 666 /dev/null c 1 3

# Set Global Environment
export PATH=/bin:/sbin:/usr/bin:/usr/sbin:/opt/wz_mini/bin
export PATH=/system/bin:$PATH
export LD_LIBRARY_PATH=/system/lib:/opt/wz_mini/lib
export LD_LIBRARY_PATH=/thirdlib:$LD_LIBRARY_PATH

# networking
ifconfig lo up
#ifconfig eth0 192.168.1.80

# Set the system time from the hardware clock
#hwclock -s

# Mount driver partition
mount -t squashfs /dev/mtdblock3 /system

/opt/wz_mini/etc/init.d/v3_post.shn
# Mount configs partition
mount -t jffs2 /dev/mtdblock6 /configs

# Run init script
if [ -f /configs/wyze_hack.sh ]; then
    /configs/wyze_hack.sh &
elif [ -f /system/init/app_init.sh ]; then
    /system/init/app_init.sh &
fi

# Run liteOta upgrade app
#/liteOta &

gtxaspec, 2022-05-14:
looks like theres a bug in the v3_init.sh script, one sec

gtxaspec, 2022-05-14:
looks like that fw version is broken, the scripts don't run correctly because the busybox version doesn't work as it should.  the scripts should run even on an old version of busybox...

so the solution for now is to upgrade the FW

gtxaspec, 2022-05-14:
add this under GPIO=63
```mount --bind /opt/wz_mini/bin/busybox /bin/busybox```

virmaior, 2022-05-14:
fully operational wz_mini'd camera !

gtxaspec, 2022-05-14:
note that with that fix, wifi doesn't work because iCamera needs an older busybox.  I'll try figure out how to include this into the release to make it work across all fw versions...

virmaior, 2022-05-14:
fortunately since I'm doing usb direct, I don't actually want/need the wifi support! 

But it seems like we should start invoicing wyze

gtxaspec, 2022-05-15:
latest version has the fixes built in.  hope the new version with bit rate  settings works well for you

closed 2022-05-15 by gtxaspec