gtxaspec/wz_mini_hacks #39: List of working LAN adapters

Vendo232, 2022-05-15:
if you have working adapter with this hack please share direct link on Amazon, eBay etc.. thx

gtxaspec, 2022-05-15:
make sure they are USB 2.0, asix based.  USB 3.0 does not work reliably.  I I recently got this as a spare and it works fine.

https://www.amazon.com/dp/B00MYT481C

wehbs, 2022-05-16:
> make sure they are USB 2.0, asix based. USB 3.0 does not work reliably. I I recently got this as a spare and it works fine.
> 
> https://www.amazon.com/dp/B00MYT481C

So how would you connect this with power? Thanks!

gtxaspec, 2022-05-16:
https://www.amazon.com/dp/B083M1S6QT
or
https://www.amazon.com/dp/B07VVNNQ8Q

wehbs, 2022-05-16:
Do you think this one would work for a poe solution?

https://www.amazon.com/gp/product/B0876N3PMZ

gtxaspec, 2022-05-16:
tough to say, it doesn't mention the chipset used inside the adapter.   If you can find one that says ASIX chipset, and it's usb 2.0, it most likely should work, or try asking the seller for chipset information.

gtxaspec, 2022-05-16:
https://www.amazon.com/ANVISION-Splitter-Compliant-Raspberry-AV-PS05-1/dp/B07W9DQ4BC/
this one has a review from a user that says it works on a pan cam, so maybe it will work

evanheckert, 2022-05-16:
This works great for both power and Ethernet. I haven't tested both via Poe yet, but separate Ethernet and microusb power. UCTRONICS for Raspberry Pi Zero... https://www.amazon.com/dp/B0876N3PMZ

Vendo232, 2022-05-16:
The must have bad batch or something, ordered 5 all of them don't have Ethernet working.

I do have one working but can't get another one

gtxaspec, 2022-05-17:
what is the device and vendor IDs of the non working ones? can you plug into a PC to find out?

evanheckert, 2022-05-17:
One thing to check, the one I link above wasn't working, but then I tested it without using the micro USB extension I bought, and it worked perfectly. I hadn't realized that not all micro USB cables deliver data.

wehbs, 2022-05-17:
Ok so it actually arrived in the mail about an hr ago (https://www.amazon.com/dp/B0876N3PMZ) and I tested with straight POE and with Usb power + ethernet and not working just blue blinking light on cam. Going to try and connect it directly to my mac to see if I can any details on the vendor ID.

Vendo232, 2022-05-17:
> what is the device and vendor IDs of the non working ones? can you plug into a PC to find out?

the device is not showing up at all, this has been the case with all units I got from Amazon. I emailed their support to improve functional testing at their manufacturing sites.

here I`m using old Stream HP 8 tablet with Lubuntu. You can see that USB + OTG works and shows up on the list of devices but the UCTRONICS is not. So it will not be a change of chipset situation.

[image]

[image]

Vendo232, 2022-05-17:
even with power injection it is dead

[image]

Vendo232, 2022-05-17:
the only option right now to use POE  would be this combo

POE Splitter: https://www.amazon.com/UCTRONICS-802-3af-Splitter-Ethernet-Raspberry/dp/B01MDLUSE7

+

The USB 2.0 you confirmed it is working: https://www.amazon.com/dp/B00MYT481C

PanchitoSolo, 2022-05-19:
Got 3 cameras working with a TP-Link USB 2.0 UE200 Ethernet Adapter (Realtek RTL8153 Chipset) https://www.amazon.com/dp/B01M2VGO8I/ (easier for me to get from stores here in Mexico) by enabling cdc_ether.ko and rndis_host.ko in run_mmc.sh

gtxaspec, 2022-05-20:
interesting that it uses ncm... I'll get one to test

skilau, 2022-05-30:
This one does NOT work, (not sure why yet):
https://www.amazon.com/gp/product/B00LLUEJFU/

I had this one from an existing install of a Amazon Fire Stick 4K, where I wanted Ethernet instead of Wifi.
It works fine on the Fire stick, but for some reason does not work here, even though it is based on the AX88179 Chipset.

Since the Wifi is disabled if Ethernet is enabled, I can't hop onto the Cam over ssh to check out the log messages/dmesg to see what might be wrong.

Is there any way to have the Cam preserve boot logs from a previous boot?

gtxaspec, 2022-05-30:
good catch on the logging, I'll have a fix for this soon.

gtxaspec, 2022-05-31:
@skilau latest version saves all logs properly now.

skilau, 2022-05-31:
Awesome, thanks!
I will hook it back up tonight, to investigate it further.

I also ordered a few of these yesterday:
https://www.amazon.com/gp/product/B01N1X28F1/

Which appear to be the same chipset (AX88772) as the one that is known to work.
(It has the Y splitter cable built into it, which makes for a little cleaner look)
I will post here once I get it and can verify that it works.

skilau, 2022-06-01:
@gtxaspec : OK, did a bit of a crash course on this tonight!

The dmesg logs show:
[    1.315084] usb 1-1.1: new high-speed USB device number 3 using dwc2
[    1.450554] usb 1-1.1: device v0b95 p1790 is not supported

That does correlate device-id-wise to what I saw in lsusb:
Bus 001 Device 003: ID 0b95:1790

So I grabbed the atomcam_tools repo, and got the docker up and running, and building, so I could check out the kernel sources, etc.

Looking at:
linux-custom/drivers/net/usb/ax88179_178a.c 

static const struct usb_device_id products[] = {
{
        /* ASIX AX88179 10/100/1000 */
        USB_DEVICE(0x0b95, 0x1790),
        .driver_info = (unsigned long)&ax88179_info,
},

So in theory anyway, this usb device should work I think.

I am not yet familiar with how you are building the kernel for the Wyze cam (I am using the v3).

Do you have a specific and/or custom configs for building the kernel that you are using in the docker?
Or are you using one the existing ones he provides?

Thanks for all your help!

gtxaspec, 2022-06-01:
looks like there is a bug, we forget to load all the kernel modules.  ax88179_178a.ko is the kernel module that needs to be loaded for your adapter, and currently.

```
        insmod $KMOD_PATH/kernel/drivers/net/usb/usbnet.ko
        insmod $KMOD_PATH/kernel/drivers/net/usb/asix.ko
```
in `wz_user.sh` we only load two modules, instead of all of them for the various adapters.  ad the following under those lines:

```
        insmod $KMOD_PATH/kernel/drivers/net/usb/r8152.ko
        insmod $KMOD_PATH/kernel/drivers/net/usb/ax88179_178a.ko
```

and see if your adapter works.

skilau, 2022-06-01:
@gtxaspec :  Yup, you got it. :)

That does indeed get my adapter up and running!

[   20.321104] ax88179_178a 1-1.1:1.0 eth0: register 'ax88179_178a' at usb-dwc2-1.1, ASIX AX88179 USB 3.0 Gigibit Ethernet,

And I can confirm for others that this adapter works now:
https://www.amazon.com/gp/product/B00LLUEJFU/

Thanks for all your help!

PanchitoSolo, 2022-06-01:
> looks like there is a bug, we forget to load all the kernel modules. ax88179_178a.ko is the kernel module that needs to be loaded for your adapter, and currently.
> 
> ```
>         insmod $KMOD_PATH/kernel/drivers/net/usb/usbnet.ko
>         insmod $KMOD_PATH/kernel/drivers/net/usb/asix.ko
> ```
> 
> in `wz_user.sh` we only load two modules, instead of all of them for the various adapters. add the following under those lines:
> 
> ```
>         insmod $KMOD_PATH/kernel/drivers/net/usb/r8152.ko
>         insmod $KMOD_PATH/kernel/drivers/net/usb/ax88179_178a.ko
>         insmod $KMOD_PATH/kernel/drivers/net/usb/cdc_ether.ko
> ```
> 
> and see if your adapter works.

Is there any issue in loading more modules than we needed? I was thinking that because every time I update I need to edit wz_user.sh to load the modules for my RTL8153, so maybe if there is no issue in loading all of them can you add them all on the  wz_user.sh by default?

gtxaspec, 2022-06-01:
we could have something like 
`ENABLE_USB_ETH_MODULES=r8152,asix,cdc_ether"

gtxaspec, 2022-06-01:
@PanchitoSolo which modules are you using for your 8153? 

On mine i need usbnet, cdc_ether, and r8152.

PanchitoSolo, 2022-06-01:
> we could have something like `ENABLE_USB_ETH_MODULES="r8152,asix,cdc_ether"`
> 
> to save some memory and load only the modules we need.

This would be amazing! Thank you

> @PanchitoSolo which modules are you using for your 8153?
> 
> On mine i need usbnet, cdc_ether, and r8152.

Yep, those 3 are the ones that I needed to load. I posted my adapter here https://github.com/gtxaspec/wz_mini_hacks/issues/39#issuecomment-1132069126.

I'm getting a new one tomorrow which doesn't mention the chipset but I think it could work, I will need to open it first and post if there is any difference with the ones already mentioned here.

gtxaspec, 2022-06-01:
latest version out now includes `ENABLE_USB_ETH_MODULES=` in the config.

Vendo232, 2022-06-01:
can I list all modules?
ENABLE_USB_ETH_MODULES="six, ax88179_178a, cdc_ether, r8152"

gtxaspec, 2022-06-01:
yeah!

gtxaspec, 2022-06-01:
https://www.amazon.com/dp/B09VC9584F works fine with the V2, haven't tested with v3.

skilau, 2022-06-02:
I can confirm, this product works as well:
https://www.amazon.com/dp/B01N1X28F1

It uses the "asix" driver, and shows as:
asix 1-1:1.0 eth0: register 'asix' at usb-dwc2-1, ASIX AX88772B USB 2.0 Ethernet

PanchitoSolo, 2022-06-02:
> I can confirm, this product works as well: https://www.amazon.com/dp/B01N1X28F1
> 
> It uses the "asix" driver, and shows as: asix 1-1:1.0 eth0: register 'asix' at usb-dwc2-1, ASIX AX88772B USB 2.0 Ethernet
> 
> (The nice thing about this USB Ethernet adapter, is that it includes the "Y" cable built-in. So it is less cable mess to deal with)

You beat me to it. I've just received an adapter like yours (but no brand) and it is exactly an ASIX AX88772B USB 2.0 Ethernet. Works by enabling usbnet.ko and asix.ko drivers. 

[image]

Got it on a mexican e-commerce site but it seems to me that this is the one on Amazon US https://www.amazon.com/dp/B07M5X9795

gtxaspec, 2022-06-04:
has anyone found a working POE adapter that is one unit?

Vendo232, 2022-06-04:
with the latest update got this RTL8153  based working.
https://www.amazon.com/Yizhet-Ethernet-Adapter-Gigabit-Compatible/dp/B08ZN5TMB4

gtxaspec, 2022-06-04:
users from the atomcam_tools repo report this poe adapter working:
https://www.amazon.com/dp/B0876N3PMZ

Vendo232, 2022-06-04:
> users from the atomcam_tools repo report this poe adapter working: https://www.amazon.com/dp/B0876N3PMZ

yes, I have one working and 5 not working.

gtxaspec, 2022-06-11:
converting to discussion for visibility