gtxaspec/wz_mini_hacks #40: RTSP - modify libcallback to inject both streams, 1080p and 360p

gtxaspec, 2022-05-15:
investigating

gtxaspec, 2022-05-15:
app live view is broken again when more than one callback is hooked

gtxaspec, 2022-05-16:
fixed function, everything works now, including with next release.

closed 2022-05-16 by gtxaspec

Vendo232, 2022-05-16:
unable to make it work
[image]

seems like stream is missing video
[image]

gtxaspec, 2022-05-16:
which stream? both?

gtxaspec, 2022-05-16:
just tested, working on latest release for me, both streams.

Vendo232, 2022-05-16:
I can't get any working. Hmm , must be doing something wrong. Im using RTSP FW with RTSP off will check Standard FW

gtxaspec, 2022-05-16:
haven't tested latest with the rtsp firmware...

I will test

gtxaspec, 2022-05-16:
definitely worth upgrading to latest fw, there's no advantage to using the old rtsp one anymore from what I see, unless you have some feedback regarding that.

the latest fw has numerous performance related fixes tok

Vendo232, 2022-05-16:
it works PERFECT!

gtxaspec, 2022-05-16:
awesome! enjoy