gtxaspec/wz_mini_hacks #42: Official RTSP firmware + wz_mini_hacks

shamlord, 2022-05-16:
I didn't bother reflashing my v3 cams to a non-RTSP firmware before attempting to use wz_mini_hacks. I found that in this case the camera will run both the Wyze RTSP server on port 558, as well as the wz_mini_hacks RTSP server on 8558. Both streams function, though performance may not be the best.

I don't know if anyone would intentionally do this, but I figured I'd share my experience. After realizing this, I went around and flashed the cameras back to a non-RTSP Wyze firmware. and confirmed the "official" RTSP server was no longer running on port 558.

shamlord, 2022-05-16:
Yes, the default ports.

gtxaspec, 2022-05-26:
latest release works better than rtsp firmware according to reports, closing for now.

closed 2022-05-26 by gtxaspec