gtxaspec/wz_mini_hacks #44: RTSP High + Low Resolution Streams Port and URL configuration

shamlord, 2022-05-16:
Reference:  https://ipcamtalk.com/threads/adding-camera-that-the-substream-are-on-a-different-ports.59539/

I'm attempting to set up Blue Iris to use the low resolution stream for detection and only switching to the high resolution stream when necessary.  According to the thread above, most IP cameras use the same port number for their low and high resolution streams, and use a different URL to separate the streams.  Is this possible to implement on wz_mini_hacks?

The URLs could look something like this:
192.168.2.17:8554/unicast
192.168.2.17:8554/unicastlow

gtxaspec, 2022-05-16:
v4l2rtspserver currently does not support this as far as I know.  perhaps you could check their repo to see if anyone has a solution or even open up an issue.

shamlord, 2022-05-16:
I've commented on an v4l2rtspserver issue from someone with a similar request. The repo owner stated that this may be possible by using v4l2loopback, but that project does not create RTSP streams itself, only a loopback device.

I'll follow up here if the v4l2rtspserver dev will be adding a multi-stream feature or explains how you might accomplish this with v4l2rtspserver and v4l2loopback.

gtxaspec, 2022-05-16:
can you link to the issue(s) you are referring to?

wz_mini currently uses both v4l2rtspserver and v4l2loopback

shamlord, 2022-05-16:
https://github.com/mpromonet/v4l2rtspserver/issues/277

Vendo232, 2022-05-17:
I can confirm nthe dual stream setup works perfect in Frigate. Low stream ( 512/256kbps ) for detection, High ( 2048/1024kbps ) .

shamlord, 2022-05-18:
v4l2rtspserver supports multiple streams on the same port according to this post: https://github.com/mpromonet/v4l2rtspserver/issues/277#issuecomment-1128731798

gtxaspec, 2022-05-18:
currently testing some stuff with the v2, i will test this on the v3 shortly.

gtxaspec, 2022-05-18:
it works, i'll release an update tonight.

gtxaspec, 2022-05-18:
```/media/mmc/wz_mini/bin/v4l2rtspserver -C 1 -a S16_LE -P 8554 /dev/video1,hw:Loopback,0 /dev/video2,hw:Loopback,1```

gtxaspec, 2022-05-18:
fixed in latest release!

closed 2022-05-18 by gtxaspec