gtxaspec/wz_mini_hacks #45: 90° Video Rotation

wehbs, 2022-05-17:
Using my Wyze cam as part of a Homekit video doorbell. Would be awesome if we could somehow rotate the video feed by 90°.

wehbs, 2022-05-18:
Wyze cam only allows 180 looks like. But this could be done with ffmpeg looks like with a filter -vf "transpose=dir=0".

gtxaspec, 2022-05-18:
ffmpeg is the easy way capturing the rtsp stream, otherwise will need to look at the T31 SDK to see how it can be done in software on the camera.

gtxaspec, 2022-06-04:
we may be able to do this by changing the i2c config on the image sensor, though, the camera does not use the standard i2c interface.  will research this method.

gtxaspec, 2022-06-09:
looks like only horizontal mirror, vertical flip, and Horizontal Mirror and Vertical Flip are supported modes by the sensor.   

I wonder how the doorbell does it...

mrlt8, 2022-06-09:
I believe they do the rotation in the app. That's why they had/have issues with the echo shows.

gtxaspec, 2022-06-09:
@mrlt8 oh awesome information, I wasn't aware of that... You've saved me some time in Ghidra digging through libraries.

gtxaspec, 2022-06-11:
closing, for now the only way to do this is to use something like ffmpeg.

closed 2022-06-11 by gtxaspec

day1118, 2023-12-04:
I tried adding `-vf "transpose=dir=0"` into the go2rtc config after the `-i /dev/video1`, but the video then shows blank. Has anyone done this successfully? Are there any other changes needed?
Thanks