gtxaspec/wz_mini_hacks #48: Unable to start RTSP

BoredMind, 2022-05-20:
Seems I am unable to get RTSP to start.  I've got this loaded and SSH enabled but when I look in v3_post I don't see RTSP starting up. I would expect to see "echo "load video loopback driver at video1"" but I don't.   Am I missing something here?

```

[root@WCV3:log]# cat v3_post.log
welcome to v3_post.sh
PID 104
+ echo 'v3_post.sh exec'
v3_post.sh exec
+ grep 'RTSP_HI_RES_ENABLED\='
+ cat /opt/wz_mini/run_mmc.sh
' '==' 'RTSP_HI_RES_ENABLED\="true"' ]]
+ grep 'RTSP_LOW_RES_ENABLED\='
+ cat /opt/wz_mini/run_mmc.sh
' '==' 'RTSP_LOW_RES_ENABLED\="true"' ]]
```

Here is my RTSP block from run_mmc.sh

```
#####VIDEO STREAM#####
RTSP_LOGIN="xxxx"
RTSP_PASSWORD="xxxx"
RTSP_PORT="8554"

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"
RTSP_HI_RES_MAX_BITRATE="2048"
RTSP_HI_RES_TARGET_BITRATE="1024"
RTSP_HI_RES_ENC_PARAMETER="2"

RTSP_LOW_RES_ENABLED="false"
RTSP_LOW_RES_ENABLE_AUDIO="false"
RTSP_LOW_RES_MAX_BITRATE=""
RTSP_LOW_RES_TARGET_BITRATE=""
RTSP_LOW_RES_ENC_PARAMETER=""

ENABLE_MP4_WRITE="false"
ENABLE_MP4_WRITE="false"

```

gtxaspec, 2022-05-20:
can you share v3_init.log too?

gtxaspec, 2022-05-20:
just tested, works OK for me on x.139.

BoredMind, 2022-05-20:
```
[root@WCV3:log]# cat v3_init.log
welcome to v3_init.sh
PID 1

 __          ________  __  __ _____ _   _ _____
 \ \        / |___  / |  \/  |_   _| \ | |_   _|
  \ \  /\  / /   / /  | \  / | | | |  \| | | |
   \ \/  \/ /   / /   | |\/| | | | | . ` | | |
    \  /\  /   / /__  | |  | |_| |_| |\  |_| |_
     \/  \/   /_____| |_|  |_|_____|_| \_|_____|
                  ______
                 |______|

+ mount --bind /opt/wz_mini/bin/busybox /bin/busybox
+ GPIO=63
+ mount -t jffs2 /dev/mtdblock6 /configs
+ grep PRODUCT_MODEL
+ cat /configs/.product_config
+ '[[' 'PRODUCT_MODEL=WYZE_CAKP2JFUS' '==' 'PRODUCT_MODEL=HL_PAN2' ]]
+ '[[' -e /opt/wz_mini/etc/.first_boot ]]
+ echo 'first boot already completed'
first boot already completed
+ mount --bind /opt/wz_mini/etc/inittab /etc/inittab
+ echo 'bind /etc/profile for local/ssh shells'
bind /etc/profile for local/ssh shells
+ mount --bind /opt/wz_mini/etc/profile /etc/profile
+ echo 'mounting tmpfs'
mounting tmpfs
+ mount -t tmpfs /tmp
+ echo 'mount system to replace factorycheck with dummy, to prevent bind unmount'
mount system to replace factorycheck with dummy, to prevent bind unmount
+ mount /dev/mtdblock3 /system
+ mount --bind /opt/wz_mini/bin/factorycheck /system/bin/factorycheck
+ touch /tmp/usrflag
+ echo 'replace stock fstab'
replace stock fstab
+ mount --bind /opt/wz_mini/etc/fstab /etc/fstab
+ echo 'mount workplace dir'
mount workplace dir
+ mount -t tmpfs /opt/wz_mini/tmp
+ echo 'install busybox applets'
install busybox applets
+ mkdir /opt/wz_mini/tmp/.bin
+ /opt/wz_mini/bin/busybox --install -s /opt/wz_mini/tmp/.bin
+ echo 'create workspace directory'
create workspace directory
+ mkdir /opt/wz_mini/tmp/.storage
+ echo 'copy stock rcS'
copy stock rcS
+ cp /etc/init.d/rcS /opt/wz_mini/tmp/.storage/rcS
+ echo 'add v3_post inject to stock rcS'
add v3_post inject to stock rcS
+ sed -i '/^".*/aset -x' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/^# Mount configs.*/i/opt/wz_mini/etc/init.d/v3_post.sh\n' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/sbin:/s/$/:\/opt\/wz_mini\/bin/' /opt/wz_mini/tmp/.storage/rcS
+ sed -i '/system\/\lib/s/$/:\/opt\/wz_mini\/lib/' /opt/wz_mini/tmp/.storage/rcS
+ echo 'replace stock password'
replace stock password
+ cp /opt/wz_mini/etc/shadow /opt/wz_mini/tmp/.storage/shadow
+ mount --bind /opt/wz_mini/tmp/.storage/shadow /etc/shadow
+ chmod 400 /etc/shadow
+ echo 'mount kernel modules'
mount kernel modules
+ mount --bind /opt/wz_mini/lib/modules /lib/modules
+ '[[' -e /opt/wz_mini/swap.gz ]]
+ echo 'swap archive not present, not extracting'
swap archive not present, not extracting
+ '[[' -d /opt/wz_mini/usr/share/terminfo ]]
+ echo 'terminfo already present'
terminfo already present
+ echo 'Run dropbear ssh server'
Run dropbear ssh server
+ /opt/wz_mini/bin/dropbear -R -s -g
+ grep 'DEBUG_ENABLED\='
+ cat /opt/wz_mini/run_mmc.sh
' '==' 'DEBUG_ENABLED\="true"' ]]
+ grep 'WEB_CAM_ENABLE\='
+ cat /opt/wz_mini/run_mmc.sh
' '==' 'WEB_CAM_ENABLE\="true"' ]]
+ '[[' -e /tmp/dbgflag ]]
+ /linuxrc
+ /opt/wz_mini/run_mmc.sh
/opt/wz_mini/etc/init.d/v3_init.sh: line 133: /opt/wz_mini/run_mmc.sh: not found
mdev is ok......
 __________________________________
|                                  |
|                                  |
|                                  |
|                                  |
| _   _             _           _  |
|| | | |_   _  __ _| |     __ _(_) |
|| |_| | | | |/ _| | |  _ / _| | | |
||  _  | |_| | (_| | |_| | (_| | | |
||_| |_|\__,_|\__,_|_____|\__,_|_| |
|                                  |
|                                  |
|_____2020_WYZE_CAM_V3_@HUALAI_____|

+ export 'PATH=/bin:/sbin:/usr/bin:/usr/sbin:/opt/wz_mini/bin'
+ export 'PATH=/system/bin:/bin:/sbin:/usr/bin:/usr/sbin:/opt/wz_mini/bin'
+ export 'LD_LIBRARY_PATH=/system/lib:/opt/wz_mini/lib'
+ export 'LD_LIBRARY_PATH=/thirdlib:/system/lib:/opt/wz_mini/lib'
+ ifconfig lo up
+ mount -t squashfs /dev/mtdblock3 /system
mount: mounting /dev/mtdblock3 on /system failed: Resource busy
+ /opt/wz_mini/etc/init.d/v3_post.sh
+ mount -t jffs2 /dev/mtdblock6 /configs
mount: mounting /dev/mtdblock6 on /configs failed: Resource busy
+ '[' -f /system/init/app_init.sh ]
+ /system/init/app_init.sh
#############ubootddr = 600.000MHz ########
################## to wait wifi vendor id ####################
################## [vendor:0x007a] atbm603x wifi ###################
(O _ o) welcome to the config file repair script [start part]!
(O _ o) this is valid mac!
(O _ o) set [/configs/.product_config] to read-only!
(O _ o) to mount kback...
(^ _ ^) kback mount ok!
(O _ o) found the md5 file in kback! [/mnt/MD5.dfeedd15d176b36b3380dd996bcbc146.config]
(O _ o)  real  md5 is [dfeedd15d176b36b3380dd996bcbc146]
(O _ o) record md5 is [dfeedd15d176b36b3380dd996bcbc146]
(^ _ ^) check md5 file [/mnt/MD5.dfeedd15d176b36b3380dd996bcbc146.config] is ok!
(^ _ ^) the md5 file in kback is ok!
(- _ -) to umount kback...
(- _ -) i will exit!
Updating device time to:
Fri May 13 20:22:36 UTC 2022
===========welcome to ver-comp tool=========
[ver-comp]dbg: appver:  4.36.9.139
[ver-comp]dbg: rootver: 4.36.3.19
[ver-comp]exec cmd: cp -rf /system/bin/app.ver /configs/
#######################
#   IS USER PROCESS   #
#######################
[FC] mount tfcard finish!
[FC] Test.tar no exist
[FC] umount tfcard finish!
[FC] In [user] mode!
kernel.core_pattern = |/system/bin/ucoredmp_collector.sh --pid %p --signal %s --name %e --time %t --output-dir /media/mmc/cores
kernel.core_pipe_limit = 1
net.unix.max_dgram_qlen = 128

```

I can't find a run_mmc.log file

```
[root@WCV3:log]# find / -name run_mmc.log
[root@WCV3:log]#
```

BoredMind, 2022-05-20:
I also just noticed that my hostname is set to something other than WCV3 but the system still thinks it's the default hostname

from run_mmc.sh
```
HOSTNAME="FrontYardRTSP"

```

```
[root@WCV3:wz_mini]# hostname
WCV3
[root@WCV3:wz_mini]#
```

gtxaspec, 2022-05-20:
```/opt/wz_mini/etc/init.d/v3_init.sh: line 133: /opt/wz_mini/run_mmc.sh: not found```
 is the problem

does file exist? could be corrupted, try copying it from the repo fresh and try again

BoredMind, 2022-05-20:
The file is there:

```
[root@WCV3:wz_mini]# pwd
/opt/wz_mini

[root@WCV3:wz_mini]# ls -lah
total 128M
drwxrwxrwx   10 root     root       16.0K Jan  1  1980 .
drwxrwxrwx    8 root     root       16.0K Jan  1  1970 ..
drwxrwxrwx    2 root     root       16.0K May 20 14:53 bin
drwxrwxrwx    5 root     root       16.0K Jan  1  1980 etc
drwxrwxrwx    3 root     root       16.0K May 20 14:53 lib
drwxrwxrwx    2 root     root       16.0K Jan  1  1980 log
drwxrwxrwx    2 root     root       16.0K May 20 14:53 mnt
drwxrwxrwx    2 root     root       16.0K May 20 14:53 root
-rwxrwxrwx    1 root     root       12.9K May 20 22:41 run_mmc.sh
-rwxrwxrwx    1 root     root      128.0M Jan  1  1980 swap
drwxrwxrwt    4 root     root          80 Jan  1  1970 tmp
drwxrwxrwx    4 root     root       16.0K May 20 14:53 usr
[root@WCV3:wz_mini]#

```

BoredMind, 2022-05-20:
I blew away the run_mmc.sh file in that directory and put a new one in.  That did the trick.  Thanks

gtxaspec, 2022-05-20:
great to hear! enjoy

closed 2022-05-20 by BoredMind