gtxaspec/wz_mini_hacks #53: wz_mini seems to not run on WyzeCam v3

Matthew-Whitlock, 2022-05-24:
I've just set up a new WyzeCam v3 through the Wyze app (linked it, updated firmware), and am now trying to run this on it - I'm very excited! However, the camera doesn't seem to be running the files after following the instructions. I have:
1. Confirmed with GParted that the 32GB microSD is fat-32 formatted (plus the camera makes some folders on the sd, so it seems to be capable of read/write)
2. Copied the files within SD_ROOT to the root of the sd card
3. Written my public key to the authorized users file
4. Set a password for the rtmp server "tmp_test"
5. Turned off the camera, plugged in the sd card, turned on the camera (Is this really all it takes to set up/install?)
6. Verified the IP via router (only one WyzeCam device, which disconnects when I turn off the camera)
7. Waited 10 minutes
8. Tried SSHing root@192.168.86.47, got `ssh: connect to host 192.168.86.47 port 22: Connection refused`
9. Tried connecting to the rtmp server with VLC at `rtsp://admin:tmp_test@192.168.86.47:8554/unicast` and failed with the log message:
```
live555 error: Failed to connect with rtsp://192.168.86.47:8554/unicast    
main error: connection failed: Connection refused    
print(len(view_names[0]))    
satip error: Failed to connect to RTSP server 192.168.86.47:8554
```

gtxaspec, 2022-05-24:
do you hear an audio prompt on the camera on the first boot?

"wz mini is starting up"

Matthew-Whitlock, 2022-05-24:
I do not

gtxaspec, 2022-05-24:
that means it's not loading.  can you confirm the factory file is also in the root of the sd card? do you have an alternate card to test with? make sure it's a dos partition map not gpt too

Matthew-Whitlock, 2022-05-24:
I rewrote the partition map to dos and it's now running! I got the audio prompt. Thank you!

closed 2022-05-24 by Matthew-Whitlock

gtxaspec, 2022-05-24:
I'll update the readme to indicate the partition map requirement, enjoy!

ngadhvi, 2023-03-14:
@gtxaspec  I exactly followed the same steps as the OP. Additionally, made sure that the 32 GB SD card has a DOS partition map and not gpt. I hear no prompt when the cam (Wyze Cam V3) starts. 
I get:
ssh: connect to host 192.168.4.116 port 22: Connection refused

[tcp @ 0x7f544c002fc0] Connection to tcp://192.168.4.116:8554?timeout=0 failed: Connection refused
rtsp://admin:password@192.168.4.116:8554/unicast: Connection refused.

Please help.

glarue, 2023-04-06:
I have the same issue as @ngadhvi—formatted SD card FAT32 with DOS partition map in GParted, copied files to sd card, added SSH public key. No noise on boot, loads the default firmware.

I also tried setting the `boot` flag on the sd card partition to see if that would help to no avail.