gtxaspec/wz_mini_hacks #55: 403 on Wyze 360024852172-Release-Notes-Firmware page?

FiveLeavesLeft, 2022-05-25:
In compile_image.sh I am getting an empty string for DL_URL,

wget https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware 

seems to return 403, though I can load it in a browser ok.

 wget https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware
--2022-05-25 21:16:26--  https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware
Resolving support.wyze.com (support.wyze.com)... 104.16.51.111, 104.16.53.111
Connecting to support.wyze.com (support.wyze.com)|104.16.51.111|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2022-05-25 21:16:26 ERROR 403: Forbidden.

 wget --user-agent="Mozilla/4.0 (compatible; MSIE 6.0; Windows NT 5.1; SV1)" -qO https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware

( Silently fails )

Is this just me?

gtxaspec, 2022-05-25:
try a different user agent string

gtxaspec, 2022-05-25:
change the first variable to:

```
UA_STRING="Mozilla/5.0 (Macintosh; Intel Mac OS X 12_4) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.4 Safari/605.1.15"
DL_URL=$(wget --user-agent="$UA_STRING" -qO- https://support.wyze.com/hc/en-us/articles/360024852172-Release-Notes-Firmware | grep -Eo "(http|https)://[a-zA-Z0-9./?=_-]*" | gre>
```

FYI the old one worked for me, maybe something about your version of wget is not sending the user agent.

FiveLeavesLeft, 2022-05-25:
Same thing.

Funny thing is is that it worked for me last night. 

But if it's working for you and others, must be something on my end. I'll poke around.

gtxaspec, 2022-06-04:
@FiveLeavesLeft were you able to get it working?

FiveLeavesLeft, 2022-06-04:
I worked around it. Wyze must not like me. lol.

closed 2022-06-04 by FiveLeavesLeft