gtxaspec/wz_mini_hacks #56: Maybe a FPS Problem???

capncybo, 2022-05-26:
Perhaps I'm misunderstanding or doing something wrong... but after launching:
ffplay -x 640 -y 360 -top 641 -left 646 -window_title Front-Yard -rtsp_transport tcp rtsp://myname:mypassword@192.168.50.22:8554/video1_unicast

I'm still seeing...
Metadata:
    title           : LIVE555 Streaming Media v2022.04.26
    comment         : LIVE555 Streaming Media v2022.04.26
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 1920x1080, **20 fps**, 20 tbr, 90k tbn, 40 tbc
    Stream #0:1: Audio: pcm_s16be, 16000 Hz, 1 channels, s16, 256 kb/s

I was expecting a FPS=15 With the following settings in my  [wz_mini.conf]

RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="true"
RTSP_HI_RES_MAX_BITRATE="2048"
RTSP_HI_RES_TARGET_BITRATE="1024"
RTSP_HI_RES_ENC_PARAMETER="2"
RTSP_HI_RES_FPS="15"

gtxaspec, 2022-05-26:
v2 or v3 camera?

gtxaspec, 2022-05-26:
```
chnAttr->rcAttr->outFrmRate->frmRateNum = 5(0x5) offset:size = 80:4
chnAttr->rcAttr->outFrmRate->frmRateDen = 1(0x1) offset:size = 84:4
```
from what I can see, the FPS is indeed set correctly, but is being incorrectly reported by some applications.

```
Guessed Channel Layout for Input Stream #0.1 : mono
Input #0, rtsp, from 'rtsp://x:x@x.x.x.x:8554/video1_unicast':
  Metadata:
    title           : LIVE555 Streaming Media v2022.04.26
    comment         : LIVE555 Streaming Media v2022.04.26
  Duration: N/A, start: 0.000313, bitrate: N/A
  Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 1920x1080, 20 fps, 5 tbr, 90k tbn, 40 tbc
  Stream #0:1: Audio: pcm_s16be, 16000 Hz, mono, s16, 256 kb/s
At least one output file must be specified
```

capncybo, 2022-05-26:
Sorry for the late reply... you were correct that  I'm indeed using a v3 Camera.
And forgive my ignorance but I was surprised with the manner chosen to adjust FPS within the scripts because...
I was expecting the FPS to be specified via the -F argument of v4l2rtspserver such as:
/media/mmc/wz_mini/bin/v4l2rtspserver $AUDIO_CH $AUDIO_FMT -U $RTSP_LOGIN:$RTSP_PASSWORD -P $RTSP_PORT **-F 15** $DEVICE1 $DEVICE2 &

Somewhat curious as to why it was NOT?

gtxaspec, 2022-05-26:
adjusting any v4l2 parameters on the v4l2rtspserver a command lines are ingnored since the camera forces the parameters using its own special libraries.

claudobahn, 2022-05-26:
@capncybo @gtxaspec the `tbr` value is the actual FPS of the stream. Good question as to where the 20 fps comes from. According to the `v4l2rtspserver` documentation, if anything we'd want to add `-F 0` so it doesn't try to mess w/ the FPS since as @gtxaspec mentioned the FPS is dictated by the camera's channel.

Though the `v4l2rtspserver` on my camera right now is `0.3.0-15-g917ee3e-dirty` and the help text doesn't include the `-F 0` note, whereas the documentation on https://github.com/mpromonet/v4l2rtspserver mentions it. Not sure if that was added in a later version.

gtxaspec, 2022-05-26:
this is what we are currently shipping on wz_mini:

`Version: 0.3.0-15-g917ee3e-dirty live555 version:2022.04.26`
```
v4l2rtspserver [-v[v]] [-Q queueSize] [-O file]
	          [-I interface] [-P RTSP port] [-p RTSP/HTTP port] [-m multicast url] [-u unicast url] [-M multicast addr] [-c] [-t timeout] [-T] [-S[duration]]
	          [-r] [-w] [-s] [-f[format] [-W width] [-H height] [-F fps] [device] [device]
	 -v               : verbose
	 -vv              : very verbose
	 -Q <length>      : Number of frame queue  (default 5)
	 -O <output>      : Copy captured frame to a file or a V4L2 device
	 -b <webroot>     : path to webroot
	 RTSP/RTP options
	 -I <addr>        : RTSP interface (default autodetect)
	 -P <port>        : RTSP port (default 8554)
	 -p <port>        : RTSP over HTTP port (default 0)
	 -U <user>:<pass> : RTSP user and password
	 -R <realm>       : use md5 password 'md5(<username>:<realm>:<password>')
	 -u <url>         : unicast url (default unicast)
	 -m <url>         : multicast url (default multicast)
	 -M <addr>        : multicast group:port (default is random_address:20000)
	 -c               : don't repeat config (default repeat config before IDR frame)
	 -t <timeout>     : RTCP expiration timeout in seconds (default 65)
	 -S[<duration>]   : enable HLS & MPEG-DASH with segment duration  in seconds (default 2)
	 -x <sslkeycert>  : enable RTSPS & SRTP
	 V4L2 options
	 -r               : V4L2 capture using read interface (default use memory mapped buffers)
	 -w               : V4L2 capture using write interface (default use memory mapped buffers)
	 -B               : V4L2 capture using blocking mode (default use non-blocking mode)
	 -s               : V4L2 capture using live555 mainloop (default use a reader thread)
	 -f               : V4L2 capture using current capture format (-W,-H,-F are ignored)
	 -f<format>       : V4L2 capture using format (-W,-H,-F are used)
	 -W <width>       : V4L2 capture width (default 0)
	 -H <height>      : V4L2 capture height (default 0)
	 -F <fps>         : V4L2 capture framerate (default 25)
	 -G <w>x<h>[x<f>] : V4L2 capture format (default 0x0x25)
	 ALSA options
	 -A freq          : ALSA capture frequency and channel (default 44100)
	 -C channels      : ALSA capture channels (default 2)
	 -a fmt           : ALSA capture audio format (default S16_BE)
	 Devices :
	 [V4L2 device][,ALSA device] : V4L2 capture device or/and ALSA capture device (default /dev/video0,/dev/video0)
```

claudobahn, 2022-05-26:
Interesting, looked at the latest version: https://github.com/mpromonet/v4l2rtspserver/blob/a43e094b2e04e569c1c7163f15915a85db67b575/main.cpp#L253 and it seems like the README.md in that repo adds the `-F 0` qualification, whereas the command-line output that you pasted above does not. So not sure if updating to the latest revision would make any changes to this behavior.

gtxaspec, 2022-05-26:
I'll recompile the latest rtsp server today

claudobahn, 2022-05-26:
I have my HI_RES set to 15fps and my LOW_RES set to 5fps.

`rtsp://...:8554/video1_unicast` outputs:
```
Input #0, rtsp, from 'rtsp://...:8554/video1_unicast':
  Metadata:
    title           : LIVE555 Streaming Media v2022.04.26
    comment         : LIVE555 Streaming Media v2022.04.26
  Duration: N/A, start: 1.549711, bitrate: N/A
  Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 1920x1080, 20 fps, 15 tbr, 90k tbn, 40 tbc
```

and `rtsp://...:8554/video2_unicast` outputs:
```
Input #0, rtsp, from 'rtsp://...:8554/video2_unicast':
  Metadata:
    title           : LIVE555 Streaming Media v2022.04.26
    comment         : LIVE555 Streaming Media v2022.04.26
  Duration: N/A, start: 1.749822, bitrate: N/A
  Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 640x360, 20 fps, 5 tbr, 90k tbn, 40 tbc
```

@capncybo did you reboot your camera after changing the FPS settings?

capncybo, 2022-05-26:
I think so... I tried to REBOOT the camera remotely.
I'll try a physical reboot when I'm done work & get home...
Thanks again guys

capncybo, 2022-05-27:
@gtxaspec Thank you for the updated version of the project & new binaries.
@claudobahn despite the latest updates I cannot seem to get my 'rtsp://...:8554/video1_unicast' to display as yours:
Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 1920x1080, 20 fps, **15 tbr**, 90k tbn, 40 tbc
I've tried a few different combinations but my "tbr" keeps showing 20
ex)
Would you mind sharing your RTSP section for the RTSP_HI_RES_"stuff"???
Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 1920x1080, 20 fps, 20 tbr, 90k tbn, 40 tbc
I'm completely perplexed...

capncybo, 2022-05-27:
I should have mentioned:
I can get  'rtsp://...:8554/video1_unicast2' to function as expected:
Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 640x360, 20 fps, 5 tbr, 90k tbn, 40 tbc

But for that I've only changed wz_mini.conf as:
RTSP_LOW_RES_ENABLED="true"
RTSP_LOW_RES_ENABLE_AUDIO="false"
RTSP_LOW_RES_FPS="5"
RTSP_LOW_RES_MAX_BITRATE=""
RTSP_LOW_RES_TARGET_BITRATE=""
RTSP_LOW_RES_ENC_PARAMETER=""

Totally Unsure as to why One stream seems to work while the other does not???

capncybo, 2022-05-27:
UPDATE: Some success... I just changed my "wz_mini.conf" to:
RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="false"
RTSP_HI_RES_FPS="10"
RTSP_HI_RES_MAX_BITRATE=""
RTSP_HI_RES_TARGET_BITRATE=""
RTSP_HI_RES_ENC_PARAMETER=""

And this works as expected...
Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 1920x1080, 20 fps, 10 tbr, 90k tbn, 40 tbc

capncybo, 2022-05-27:
Just re-tested & (at the risk of making a BAD Joke) 15 will get you 20 ;-)
changing my "wz_mini.conf" to:
RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="false"
RTSP_HI_RES_FPS="15"
RTSP_HI_RES_MAX_BITRATE=""
RTSP_HI_RES_TARGET_BITRATE=""
RTSP_HI_RES_ENC_PARAMETER=""

Did not work as expected...
Stream #0:0: Video: h264 (Main), yuv420p(tv, bt709, progressive), 1920x1080, 20 fps, **20 tbr**, 90k tbn, 40 tbc

However, I admit to being a novice & perhaps I'm misunderstanding some of this video tech

capncybo, 2022-05-27:
If I were to guess, perhaps the camera can only be set to "certain" numbers while others are "invalid".
If so... can anyone recommend an easy way to check?
+
If this is the case... perhaps the "valid" numbers could be documented???
Unless of course the valid numbers change due to other parameters being adjusted.

gtxaspec, 2022-05-28:
i think its a bug somewhere in the camera or the rtsp server reporting the frame rate...if you look at the video stream the rate is definitely changing if you change it while viewing the stream...

capncybo, 2022-05-28:
@gtxaspec  Alright I'll consider it a NON-Issue then & soon Close the subject.  However in your opinion, If I change the FPS of my HI-RES stream from the default of 20 to 15 because (From a viewing perspective I cant notice the difference)...
It should save some bandwidth or give me less latency when viewing remotely, correct?

capncybo, 2022-05-30:
" NON-Critical" -Issue/Bug in the camera or the rtsp server reporting the frame rate so... Closing

closed 2022-05-30 by capncybo