gtxaspec/wz_mini_hacks #58: Block Wyze streaming at camera level

Vendo232, 2022-05-27:
Create ability to block Wyze cam contacting Amazon Kinesis server "c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com" at camera level in order to reduce Internet/Lan traffic

c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com

gtxaspec, 2022-05-27:
you can enable iptables and block this host from the command line

Vendo232, 2022-05-27:
thankk you, would it be something like this?

iptables -I INPUT -s c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com -j DROP
service iptables save

Vendo232, 2022-05-27:
would this work?
[image]

endertable, 2022-05-28:
Does adding the following entry to /etc/hosts work?

127.0.0.1  c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com

gtxaspec, 2022-05-28:
@endertable yes that should work too and is probably better than iptables no kernel modules required.

claudobahn, 2022-05-29:
Also, the IP addresses for AWS DNS entries typically rotate on a regular schedule, so the `/etc/hosts` entry will likely be more effective long-term and require less maintenance.

claudobahn, 2022-05-31:
Maybe what makes more sense is having an option to not start (or kill) the Wyze processes, though? Instead of trying to block their communication.

Some probably need to be started at some point to set up the channels, though. Not sure how all of that is initialized at startup.

gtxaspec, 2022-05-31:
it's all integrated. if you don't start iCamera, you don't get video, and nothing works without it.  so that means no rtsp

mrlt8, 2022-05-31:
Doesn't the KVS/AWS IoT core stuff get started by another script or is that just the certificates/config?

gtxaspec, 2022-06-01:
A bunch of stuff runs:

```
/system/bin/assis
/system/bin/sdkshellcalltool
/system/bin/hl_client
/system/bin/iCamera
system/bin/dongle_app (v2)
/system/bin/sinker
/system/bin/timesync
/system/bin/dumpload
```
and theres more, the edge ai stuff, the web rtc stuff too.  They are launched by the app_init script, and if any are killed (except time and dumpload) , iCamera will segfault and the camera will reboot once the watchdog says so.

If you try to launch iCamera alone, it segfaults and the camera reboots...lol.

This applies to any Hualai Ingenic based cameras.  It's just how they have designed their camera system.  

The KVS/AWS stuff is inside iCamera, hl_client, webrtc, edgeai, its all intertwined from what I can tell.

mrlt8, 2022-06-01:
What about `/tmp/webrtc/launch.sh`?

gtxaspec, 2022-06-01:
that script is currently broken because it calls /bin/bash when bash is not present on the production builds pushed to the cameras

gtxaspec, 2022-06-01:
sorry, I didn't read that correctly, the web rtc client is run by launch.sh, the aws stuff is hl_client, iCamera, sinker I believe

Vendo232, 2022-06-04:
> Does adding the following entry to /etc/hosts work?
> 
> 127.0.0.1 c1ybkrkbr1j10x.credentials.iot.us-west-2.amazonaws.com

I`m getting read only error

[image]

gtxaspec, 2022-06-04:
save that file elsewhere, example to /opt/hosts
then
`mount --bind /opt/hosts /etc/hosts`

digitalpixies, 2022-10-25:
Anyone try to redirect those ips to a self hosted we server? Assuming there's error code handling, an invalid https cert or http error response should put the Wyze into a wait-and-try-again-state. It shouldn't reboot itself nor should it try to hammer their own servers (prevent ddosing themselves when they really have a server issue)

tbrausch, 2022-10-25:
> save that file elsewhere, example to /opt/hosts then `mount --bind /opt/hosts /etc/hosts`

First of all, "Thank You!".  I just saw this post and tried it on one of my cameras.  It appears to have solved ALL of the tearing issues I was having with the 1080P stream.  I am not that familiar with Linux and was hoping you could answer a quick question for me.  When I run the command:
`mount --bind /opt/hosts /etc/hosts`
it works until the camera is restarted, the mount operation apparently is reset after a reboot.  What would be the best way to perform this command at power up?  I assume a custom script, but could someone tell me what this script would look like and where to put it?

Thanks in advance!

gtxaspec, 2023-09-26:
self-hosted mode was added.

closed 2023-09-26 by gtxaspec