gtxaspec/wz_mini_hacks #59: How to play audio files triggered by Home Assistant

Vendo232, 2022-05-28:
Prerequisites

- working SSH setup
- Audio files ( Wav )
- Working Home Assistant instance

1. SSH

I use Windows and Putty and the easiest way for me to generate SSH keys is Puttyget
[image]
Copy public key to `wz_mini/etc/ssh/authorized_keys` right after that save your Private key which you will need to SSH to the camera.
Launch puttygen and load your existing private key file. From the menu, select Export OpenSSH key (force new file format). Then click Save private key and save your converted key file.

[image]

Now you should have update authorized_keys file on WYZE cam and privatekey.ppk on your PC. pprivatekey,ppk will have to be copied to your Home Assistant.

2. Home Assistant

create folder called "ssh" within your "config" folder. I mapped HA config file to my Windows so look for SMB addon to do the same.
if you do the SMB you will be able to map and access HA Conofig folder from your Windows. I highly recommend doing it.
[image]

Place your privatekey.ppk into /config/ssh or in my case I placed it in existing /.ssh folder
[image]

Now we will have to change permission to this file. Install Terminal & SSH ADDON in Home Assistant and enter your SSH folder
`cd config`
`cd ssh`
`chmod 600 privatekey.ppk`
[image]

at this point you should be able to SSH into WYZE cam

in same terminal window type
`ssh -i /config/.ssh/privatekey.ppk root@192.168.1.174`

the IP will be your IP of your Wyze cam

you should now be presented with root@WCV3 command and have successful SSH connection to your cam.

3. Executing the audio file

Wyze_mini hacked camera can play Wav files with following parameters 
[image]
these worked for me.

Copy desired audio file on WYZE camera SD Card and execute the audio playback by command 

`cmd aplay /media/mmc/wz_mini/usr/share/audio/init.wav 40`

path to the WAV file is your choice and the 40 at the end represents volume 

4. Automations in Home Assistant

to use Audio files for your automations add use following command line action.

`  - platform: command_line
    switches:
      audio_switch_a:
        command_on: "ssh -i /config/.ssh/private3.ppk -o StrictHostKeyChecking=no root@192.168.1.174 cmd aplay /media/mmc/wz_mini/usr/share/audio/b.wav 100"`

closed 2022-05-28 by Vendo232

Vendo232, 2022-05-28:
https://user-images.githubusercontent.com/44098230/170833562-b65554f6-081a-4062-859c-9358ef028cb3.mp4

gtxaspec, 2022-05-29:
very cool!

Vendo232, 2022-05-29:
@gtxaspec  would you know if it would be possible using Filezilla and sftp to access cam for easier and remote file transfer ?

gtxaspec, 2022-05-29:
hmm, no clue, not a windows guy, for me scp would be the easiest way.  I think putty has scp doesn't it?