gtxaspec/wz_mini_hacks #62: Possible to increase `cmd jpeg` quality?

tensiondriven, 2022-05-29:
wz_mini_hacks is amazing and has given me new hope for all kinds of projects with Wyze cams.  It goes so far beyond simply enabling RSTP, like so many people have posted about.  Thank you so much for this effort!

My question - I am using `cmd jpeg` with a bash script to periodically save a jpg for time lapse / security purposes.  (At the moment, I also have RTSP enabled.)

Currently the quality of the JPG's look fairly compressed.  I'm not 100% sure but downloading a JPG from `cmd jpeg` seems to be worse quality than those captured in the native Wyze app.

Is the output of `cmd jpeg` the same as that captured in the Wyze app?  What dictates the quality of images captured via `cmd jpeg`?  Is it possible to increase the quality of jpeg capture, perhaps by changing a setting in `/etc/config/.user_config`?  (I see `bitRate=120` in mine, curious what this does.  Could not find docs describing possible values in `.user_config`.)

Searching the web for more on the `cmd` command didn't do a whole lot of good, nor did looking for info about `cmd` in the repos linked in the README here.  

If anyone could provide any insight, that would be great!   Thank you again!

gtxaspec, 2022-05-29:
hmm, yes, I have never tried to use the snapshots.

```
	CHANNEL 4    1920x 1080      STOP JPEG tf:0          df:0          encdur:123602, encodingFrameCnt:0,endencodeFrameCnt=0,endrelaseFrameCnt=0,Fps:0.00,Bitrate:0.00
-------------------------------------------------
ch->index = 4
ch->releaseFrmNum = 0
ch->releaseFrmDen = 0
chnAttr->encAttr->eProfile = 67108864(0x4000000) offset:size = 0:4
chnAttr->encAttr->uLevel = 0(0x0) offset:size = 4:1
chnAttr->encAttr->uTier = 0(0x0) offset:size = 5:1
chnAttr->encAttr->uWidth = 1920(0x780) offset:size = 6:2
chnAttr->encAttr->uHeight = 1080(0x438) offset:size = 8:2
chnAttr->encAttr->ePicFormat = 392(0x188) offset:size = 12:4
chnAttr->encAttr->eEncOptions = 294952(0x48028) offset:size = 16:4
chnAttr->encAttr->eEncTools = 156(0x9c) offset:size = 20:4
chnAttr->rcAttr->rcMode = 0(0x0) offset:size = 44:4
chnAttr->rcAttr->FixQP->iInitialQP = 32(0x20) offset:size = 48:2
chnAttr->rcAttr->outFrmRate->frmRateNum = 20(0x14) offset:size = 80:4
chnAttr->rcAttr->outFrmRate->frmRateDen = 1(0x1) offset:size = 84:4
chnAttr->gopAttr->uGopCtrlMode = 2(0x2) offset:size = 88:4
chnAttr->gopAttr->uGopLength = 40(0x28) offset:size = 92:2
chnAttr->gopAttr->uNotifyUserLTInter = 0(0x0) offset:size = 94:1
chnAttr->gopAttr->uMaxSameSenceCnt = 1(0x1) offset:size = 96:4
chnAttr->gopAttr->bEnableLT = 0(0x0) offset:size = 100:1
chnAttr->gopAttr->uFreqLT = 0(0x0) offset:size = 104:4
chnAttr->gopAttr->bLTRC = 0(0x0) offset:size = 108:1
```

gtxaspec, 2022-05-29:
`echo saveraw 1 > /proc/jz/isp/isp-w02`
gives you /tmp/snap0.raw, lets find out how to convert this to png or jpg

gtxaspec, 2022-05-29:
I wonder if cmd jpeg's quality is based off the h264 stream quality?  @tensiondriven maybe you want to try to set:
```
RTSP_HI_RES_ENABLED="true"
RTSP_HI_RES_ENABLE_AUDIO="false"
RTSP_HI_RES_FPS=""
RTSP_HI_RES_MAX_BITRATE="8096"
RTSP_HI_RES_TARGET_BITRATE="8096"
RTSP_HI_RES_ENC_PARAMETER="2"
```
and take a jpg, and see if the quality has improved.

gtxaspec, 2022-05-29:
`snap0.raw` may be converted using the following program:

https://github.com/jdthomas/bayer2rgb

the examples I was able to produce look pretty clean.  let me know what you think

gtxaspec, 2022-05-29:
the images saved by the Wyze app are indeed higher quality, they are PNG...

gtxaspec, 2022-05-29:
`ffmpeg -rtsp_transport tcp -y -i rtsp://admin:password@x.x.x.x:8554/unicast -vframes 1 do.png` combined with the higher bitrate settings gets you images BETTER than the wyze app images.

tensiondriven, 2022-05-29:
Amazing.  This is far beyond my expectations.  As far as I'm concerned this can be closed!  1,000,000 thanks!

closed 2022-05-29 by tensiondriven

tensiondriven, 2022-05-29:
Ok, this is interesting.  Following up on your suggestions, I captured three versions, taken a few minutes apart: `cmd jpg`, raw + bayer2rgb, and rtsp + ffmpeg.  See below:

### Raw to Tiff using bayer2rgb

(Converted to 16bpp PNG to satisfy github) Produces an image with what looks like only infra-red, is that possible?  I tried turning off night vision on the camera and capturing this image again and got the smae result.  Also, this image looks like it has some compression, or maybe it's just the quantization of an 8-bit single-channel image:

[image]

Any idea if it's possible to either capture or decode this in full RGB?  I used the same command you posted above for bayer2rgb after building it with cmake on ubuntu.  Perhaps I did something wrong there.

### cmd jpeg

For comparison to the below.  Notice all the (expected) compression in the image.  This looks like it's been over-sharpened to me, as well, but I'm no expert.

[image]

### RTSP @ ~8000 to PNG via ffmpeg

This does indeed look the best, given that the raw was not in color:

[image]

If I look at the concrete floor or the ceiling, I see what looks like compression, for sure.  Perhaps the RTSP quality could be cranked even more, would that be insane?

If I could get the raw image in RGB, that would still probably be my ideal.  (The RAW was 4MB coming over the wire from the camera, FWIW.)  The RTSP to PNG looks good, indeed.  It seems the image could be sharper, and I wonder if this is a camera limitation or a limitation of RTSP that would be overcome by capturing raw in RGB.  Trying to squeeze the highest quality possible out of these Wyze cams is very interesting to me, and I bet that if we could find an optimal solution that others would be interested in this as well.

gtxaspec, 2022-05-29:
the raw image file is a capture straight from the image sensor.  you may want to look at the spec sheet for it, gc2053.   it will look tinted because there is no AWB post processing correction I believe.

I just guessed that it was 10bit bayer, I could be wrong though.

gtxaspec, 2022-05-29:
you could also try to push the bitrate higher, try 16032 or something lol

tensiondriven, 2022-05-29:
Is it important to keep the bitrate at a power of two or is that just a convention?

gtxaspec, 2022-05-29:
just my preference 😁

tensiondriven, 2022-05-30:
I read over https://taoic.oss-cn-hangzhou.aliyuncs.com/6446/product/XTW2020_1589249333000.pdf but couldn't make much sense of it; the bits that seemed useful:

- "The primary color GR/BG is arranged in a line-alternating way"
- The output format is listed as "Raw Bayer 10bit/8bit"

And then `bayer2rgb` lists:
```
   --first,-f     first pixel color: RGGB, GBRG, GRBG, BGGR
```

I tried decoding snap0.raw with each of the listed "first color pixels" and got output that was in different colors (screenshot below).  I was hoping that I could composite these outputs to get each channel but each image is very similar, suggesting to me that it isn't separate color channel data I'm seeing but the same color channel data being expressed in different ways in the outputs.  I could be wrong though.  The bright areas in the thumbnails were white, so maybe it makes sense that these would have lots of data in R, G, and B.

I noticed that in your example you specified 16bpp, I will try with 8 or 10 bpp.  I'm afraid I don't know much about raw image processing - can you give me any pointers, anything that point me in the direction of being able to decode one of these raw's in a meaningful way?

[image]

tensiondriven, 2022-05-30:
http://rawpixels.net/ is helpful in experimenting with different decoding methods, but doesn't yield anything better than the `bayer2rgb` example command that you provided - if anything, it's worse, it looks like the output is skipping every other/every few pixels.

gtxaspec, 2022-05-30:
Wish I could help, I'm not an image/video guy... @claudobahn would you have any idea?

tensiondriven, 2022-06-14:
@claudobahn ping - any chance you could offer a pointer for how to decode the raw image created by running `echo saveraw 1 > /proc/jz/isp/isp-w02`?  (Thanks again @gtxaspec for all the help!)

claudobahn, 2022-06-15:
@tensiondriven I'm interested in that myself, too. I'll post back here if I have any updates.