gtxaspec/wz_mini_hacks #64: I can seem to get rtsp output for wyze cam v3

jemershaw, 2022-06-02:
I'm running wyze cam v3 using firmware 4.61.0.3(offical rtsp firmware). I disabled rtsp before adding the microsd card and I can reach the rtsp endpoint but when I try to play it via VLC I only get audio(Codec PCM S16).

I tried toggling video using `cmd video on` and that didn't change anything. Let me know what sort of logs you'd need me to gather to fix this problem.

gtxaspec, 2022-06-02:
compatability with the rtsp firmware is spotty since its old.  i would share v3_init.log, wz_post.log, and wz_user.log.  blank out any passwords you have set and paste here

jemershaw, 2022-06-02:
Here's the v3_init.log
```
 PID 1

 __          ________  __  __ _____ _   _ _____
 \ \        / |___  / |  \/  |_   _| \ | |_   _|
  \ \  /\  / /   / /  | \  / | | | |  \| | | |
   \ \/  \/ /   / /   | |\/| | | | | . ` | | |
    \  /\  /   / /__  | |  | |_| |_| |\  |_| |_
     \/  \/   /_____| |_|  |_|_____|_| \_|_____|
                  ______
                 |______|

+ GPIO=63
+ mount --bind /opt/wz_mini/bin/busybox /bin/busybox
+ [ -b /dev/mtdblock9 ]
+ [[  == false ]]
+ echo not HL_PAN2
not HL_PAN2
+ [[ -e /opt/wz_mini/etc/.first_boot ]]
+ echo first boot already completed
first boot already completed
+ touch /opt/wz_mini/etc/.first_boot
+ mount --bind /opt/wz_mini/etc/inittab /etc/inittab
+ echo bind /etc/profile for local/ssh shells
bind /etc/profile for local/ssh shells
+ mount --bind /opt/wz_mini/etc/profile /etc/profile
+ echo mounting tmpfs
mounting tmpfs
+ mount -t tmpfs /tmp
+ echo mount system to replace factorycheck with dummy, to prevent bind unmount
mount system to replace factorycheck with dummy, to prevent bind unmount
+ [[ !  == true ]]
+ mount /dev/mtdblock3 /system
+ mount --bind /opt/wz_mini/bin/factorycheck /system/bin/factorycheck
+ touch /tmp/usrflag
+ echo replace stock fstab
replace stock fstab
+ mount --bind /opt/wz_mini/etc/fstab /etc/fstab
+ echo mount workplace dir
mount workplace dir
+ mount -t tmpfs /opt/wz_mini/tmp
+ echo install busybox applets
install busybox applets
+ mkdir /opt/wz_mini/tmp/.bin
+ /opt/wz_mini/bin/busybox --install -s /opt/wz_mini/tmp/.bin
+ echo create workspace directory
create workspace directory
+ mkdir /opt/wz_mini/tmp/.storage
+ echo copy stock rcS
copy stock rcS
+ cp /etc/init.d/rcS /opt/wz_mini/tmp/.storage/rcS
+ echo add wz_post inject to stock rcS
add wz_post inject to stock rcS
+ sed -i /^".*/aset -x /opt/wz_mini/tmp/.storage/rcS
+ sed -i /^# Mount configs.*/i/opt/wz_mini/etc/init.d/wz_post.sh\n /opt/wz_mini/tmp/.storage/rcS
+ sed -i /sbin:/s/$/:\/opt\/wz_mini\/bin/ /opt/wz_mini/tmp/.storage/rcS
+ sed -i /system\/\lib/s/$/:\/opt\/wz_mini\/lib/ /opt/wz_mini/tmp/.storage/rcS
+ echo replace stock password
replace stock password
+ cp /opt/wz_mini/etc/shadow /opt/wz_mini/tmp/.storage/shadow
+ mount --bind /opt/wz_mini/tmp/.storage/shadow /etc/shadow
+ chmod 400 /etc/shadow
+ [[ -e /opt/wz_mini/swap.gz ]]
+ echo swap archive not present, not extracting
swap archive not present, not extracting
+ [[ -d /opt/wz_mini/usr/share/terminfo ]]
+ echo terminfo already present
terminfo already present
+ echo Run dropbear ssh server
[... 323 lines cut ...]
+ i=0
+ '[[' -e /opt/wz_mini/log/dmesg.log.0 '||' -L /opt/wz_mini/log/dmesg.log.0 ]]
+ let i++
+ '[[' -e /opt/wz_mini/log/dmesg.log.1 '||' -L /opt/wz_mini/log/dmesg.log.1 ]]
+ let i++
+ '[[' -e /opt/wz_mini/log/dmesg.log.2 '||' -L /opt/wz_mini/log/dmesg.log.2 ]]
+ mv /opt/wz_mini/log/dmesg.log /opt/wz_mini/log/dmesg.log.2
+ DMESG_LOG=/opt/wz_mini/log/dmesg
+ touch -- /opt/wz_mini/log/dmesg.log
+ dmesg
+ sync
+ echo 3
+ '[' -f  ]
+ echo 'custom script not found'
custom script not found
log level:500
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
+ event=n
+ directory=/tmp
+ file=webrtc
+ date
Thu Jun  2 10:53:50 UTC 2022
+ '[[' webrtc '==' img ]]
```

claudobahn, 2022-06-02:
@jemershaw don't think they they ever spelled it out, but I'm pretty sure their beta RTSP firmware has a massive security hole which is why they pulled the download from the RTSP page. This security hole affected all their cameras from what I understand: https://www.bitdefender.com/blog/labs/vulnerabilities-identified-in-wyze-cam-iot-device/

To my knowledge they chose not to fix this problem in the V1 cameras and their RTSP firmware versions for V2/V3.

So you're likely better off using the currently working V2/V3 firmware that's been patched along with this mod which provides you RTSP w/o needing their RTSP firmware.

gtxaspec, 2022-06-02:
@jemershaw can you reboot the camera, once you reboot, login to ssh, wait 30 seconds, and run the commands:
`logread | grep local_sdk_video`
 `logread | grep callback`
and share the output here if you can?

gtxaspec, 2022-06-02:
@claudobahn is right, it's an old firmware with bugs and security holes, plus running double RTSP will use up more memory and make performance worse.

gtxaspec, 2022-06-04:
@jemershaw any update?

jemershaw, 2022-06-07:
Please see the output. I'm going to upgrade since I'd rather have it working and if wyze pulled the firmware then no reason to use it.
```
[root@WCV3:~]# logread | grep local_sdk_video
Jun 22 01:48:16 iCamera: [sdk,0459]dbg:(local_sdk_video_osd_set_font_bitmap) bitmap addr:[0x54f840]
Jun 22 01:48:16 syslog: [sdk,0459]dbg:(local_sdk_video_osd_set_font_bitmap) bitmap addr:[0x54f840]
Jun 22 01:48:16 iCamera: [sdk,1624]dbg:(local_sdk_video_create) apiChn:0, grp:0
Jun 22 01:48:16 syslog: [sdk,1624]dbg:(local_sdk_video_create) apiChn:0, grp:0
Jun 22 01:48:16 iCamera: [sdk,1730]dbg:(local_sdk_video_set_parameters) apiChn:0, grp:0
Jun 22 01:48:16 syslog: [sdk,1730]dbg:(local_sdk_video_set_parameters) apiChn:0, grp:0
Jun 22 01:48:16 iCamera: [sdk,0542]dbg:(local_sdk_video_osd_set_parameters) osd set params chn:0, grp:0
Jun 22 01:48:16 syslog: [sdk,0542]dbg:(local_sdk_video_osd_set_parameters) osd set params chn:0, grp:0
Jun 22 01:48:16 iCamera: [sdk,0622]dbg:(local_sdk_video_osd_set_parameters) set to paramsok, chn0, grp0
Jun 22 01:48:16 syslog: [sdk,0622]dbg:(local_sdk_video_osd_set_parameters) set to paramsok, chn0, grp0
Jun 22 01:48:16 iCamera: [sdk,1939]dbg:(local_sdk_video_start) apiChn:0 grp:0
Jun 22 01:48:16 syslog: [sdk,1939]dbg:(local_sdk_video_start) apiChn:0 grp:0
Jun 22 01:48:16 iCamera: [sdk,1976]dbg:(local_sdk_video_start) apiChn:0, encChn:0, grp:0
Jun 22 01:48:16 syslog: [sdk,1976]dbg:(local_sdk_video_start) apiChn:0, encChn:0, grp:0
Jun 22 01:48:16 iCamera: [sdk,1993]dbg:(local_sdk_video_start) apiChn:0, jpegChn:4, grp:0
Jun 22 01:48:16 syslog: [sdk,1993]dbg:(local_sdk_video_start) apiChn:0, jpegChn:4, grp:0
Jun 22 01:48:16 iCamera: [sdk,2166]dbg:(local_sdk_video_run) apiChn:0, encChn:0, grp:0
Jun 22 01:48:16 syslog: [sdk,2166]dbg:(local_sdk_video_run) apiChn:0, encChn:0, grp:0
Jun 22 01:48:16 iCamera: [sdk,1624]dbg:(local_sdk_video_create) apiChn:1, grp:1
Jun 22 01:48:16 syslog: [sdk,1624]dbg:(local_sdk_video_create) apiChn:1, grp:1
Jun 22 01:48:16 iCamera: [sdk,1730]dbg:(local_sdk_video_set_parameters) apiChn:1, grp:1
Jun 22 01:48:16 syslog: [sdk,1730]dbg:(local_sdk_video_set_parameters) apiChn:1, grp:1
Jun 22 01:48:16 iCamera: [sdk,1749]dbg:(local_sdk_video_set_parameters) apiChn:1, grp:1, enable scaler
Jun 22 01:48:16 syslog: [sdk,1749]dbg:(local_sdk_video_set_parameters) apiChn:1, grp:1, enable scaler
Jun 22 01:48:16 iCamera: [sdk,0542]dbg:(local_sdk_video_osd_set_parameters) osd set params chn:1, grp:1
Jun 22 01:48:16 syslog: [sdk,0542]dbg:(local_sdk_video_osd_set_parameters) osd set params chn:1, grp:1
Jun 22 01:48:16 iCamera: [sdk,0622]dbg:(local_sdk_video_osd_set_parameters) set to paramsok, chn1, grp1
Jun 22 01:48:16 syslog: [sdk,0622]dbg:(local_sdk_video_osd_set_parameters) set to paramsok, chn1, grp1
Jun 22 01:48:16 iCamera: [sdk,1939]dbg:(local_sdk_video_start) apiChn:1 grp:1
Jun 22 01:48:16 syslog: [sdk,1939]dbg:(local_sdk_video_start) apiChn:1 grp:1
Jun 22 01:48:16 iCamera: [sdk,1976]dbg:(local_sdk_video_start) apiChn:1, encChn:1, grp:1
Jun 22 01:48:16 syslog: [sdk,1976]dbg:(local_sdk_video_start) apiChn:1, encChn:1, grp:1
Jun 22 01:48:16 iCamera: [sdk,1993]dbg:(local_sdk_video_start) apiChn:1, jpegChn:5, grp:1
Jun 22 01:48:16 syslog: [sdk,1993]dbg:(local_sdk_video_start) apiChn:1, jpegChn:5, grp:1
Jun 22 01:48:16 iCamera: [sdk,2166]dbg:(local_sdk_video_run) apiChn:1, encChn:1, grp:1
Jun 22 01:48:16 syslog: [sdk,2166]dbg:(local_sdk_video_run) apiChn:1, encChn:1, grp:1
Jun 22 01:48:16 iCamera: [sdk,1624]dbg:(local_sdk_video_create) apiChn:2, grp:2
Jun 22 01:48:16 syslog: [sdk,1624]dbg:(local_sdk_video_create) apiChn:2, grp:2
Jun 22 01:48:16 iCamera: [sdk,1730]dbg:(local_sdk_video_set_parameters) apiChn:2, grp:2
Jun 22 01:48:16 syslog: [sdk,1730]dbg:(local_sdk_video_set_parameters) apiChn:2, grp:2
Jun 22 01:48:16 iCamera: [sdk,1749]dbg:(local_sdk_video_set_parameters) apiChn:2, grp:2, enable scaler
Jun 22 01:48:16 syslog: [sdk,1749]dbg:(local_sdk_video_set_parameters) apiChn:2, grp:2, enable scaler
Jun 22 01:48:16 iCamera: [sdk,2021]dbg:(local_sdk_video_start) apiChn:2, encChn:0, grp:2
Jun 22 01:48:16 syslog: [sdk,2021]dbg:(local_sdk_video_start) apiChn:2, encChn:0, grp:2
[root@WCV3:~]# logread | grep callback
Jun 22 01:48:15 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 1 cb
Jun 22 01:48:15 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 2 cb
Jun 22 01:48:15 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 3 cb
Jun 22 01:48:15 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 4 cb
Jun 22 01:48:15 syslog: [sdk,0479]dbg: (local_sdk_set_major_error_callback) Doing...
Jun 22 01:48:15 iCamera: [sdk,0479]dbg: (local_sdk_set_major_error_callback) Doing...
Jun 22 01:48:19 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 5 cb
Jun 22 01:48:19 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 6 cb
Jun 22 01:48:19 iCamera: [storagedevice.c:148][sdevice_can_work_set_callback]dbg: set ok! have 7 cb
```

gtxaspec, 2022-06-07:
looks like we changed something in libcallback that broke compatability with RTSP firmware.  thanks for the logs.

yerffejytnac, 2022-06-10:
Commenting to say I have the exact same problem and similar log output. Only thing that is different from OP is that I haven't flashed the custom rtsp firmware on my v3 cam. I'm using stock firmware, upgraded to the latest version.

Any idea on how much effort this would be to patch? Would I be shooting myself in the foot if I used the rtsp firmware from Wyze as a quick workaround?

gtxaspec, 2022-06-10:
@yerffejytnac can you specify which exact version you are on?  I have over 5 camera's with no issues.  The more details I receive from users the better we can nail down any issues.

rickgitdone, 2022-06-10:
@yerffejytnac  -- I would also resync all the files on the SDCARD to the latest available on this git ... then retry.. Wyze rev 4.36.2.5 seems to work for many ... 

@gtxaspec  - what rev Wyze firmware are your cams on?

gtxaspec, 2022-06-10:
@rickgitdone the latest, 4.36.9.139. old firmware including 4.36.2.5  has some bugs and security issues.

stephandie, 2022-06-11:
I had the same issue but was able to get it working. I started with the RTSP firmware (RTSP was activated) on a V3. Changed to  4.36.9.139 but couldn't get RTSP to work. I deleted all files from the SD card and started over. Worked right away. 
I assume somethings getting messed up starting up initially with the RTSP firmware and then changing to the regular firmware.

gtxaspec, 2022-06-11:
Yes, it's always recommended to start with a fresh memory card.  I will add this to the README to help others

yerffejytnac, 2022-06-11:
> @yerffejytnac can you specify which exact version you are on?  I have over 5 camera's with no issues.  The more details I receive from users the better we can nail down any issues.

Sorry @gtxaspec - got a new puppy (reason for the cam) and he's been keeping me busy. 

The current camera (Wyze Cam v3) is on firmware version 4.36.9.139 — let me know if I can provide anything else that's helpful. Currently using the latest code from the repo.

gtxaspec, 2022-06-13:
@jemershaw @yerffejytnac check out the latest version, let me know if you still have issues

yerffejytnac, 2022-06-18:
> @jemershaw @yerffejytnac check out the latest version, let me know if you still have issues

I'll give it a try this weekend and report back!

gtxaspec, 2022-06-19:
closing this issue, please open a new one if you still experience problems

closed 2022-06-19 by gtxaspec