gtxaspec/wz_mini_hacks #66: camera audio/video tuning

gtxaspec, 2022-06-07:
testing libcallback with imp_control

`cmd imp_control <command> <value>`

agc_on
agc_off
hpf_on
hpf_off
ns_on
ns_off
aec_on
aec_off

flip_mirror
flip_vertical
flip_horizontal
flip_normal

tune_contrast -128 to 128
tune_brightness -128 to 128
tune_sharpness -128 to 128
tune_saturation -128 to 128
tune_aecomp 0-255
tune_wbmode:
        ISP_CORE_WB_MODE_AUTO = 0,
        ISP_CORE_WB_MODE_MANUAL,
        ISP_CORE_WB_MODE_DAY_LIGHT,
        ISP_CORE_WB_MODE_CLOUDY,
        ISP_CORE_WB_MODE_INCANDESCENT,
        ISP_CORE_WB_MODE_FLOURESCENT,
        ISP_CORE_WB_MODE_TWILIGHT,
        ISP_CORE_WB_MODE_SHADE,
        ISP_CORE_WB_MODE_WARM_FLOURESCENT,

need help:
change framerate dynamically
change audio hz frequency dynamically
move OSD to specified region

[libcallback.so.tar.gz](https://github.com/gtxaspec/wz_mini_hacks/files/8851931/libcallback.so.tar.gz)

capncybo, 2022-06-07:
Awesome stuff, & forgive my ignorance but...
Is there already an integrated way (Or perhaps a combined terminal command) which can (play/pipe) "remote" audio over the Camera's built-in speaker.  In other words... to mimic the Microphone/Speaker feature **without** using the WYZE app.
Or to a lesser degree... stream audio which actually resides elsewhere?

gtxaspec, 2022-06-07:
not sure if there is a way to do two way audio, but streaming audio from a source other than a file... hmm I will test with ffmpeg to see if it works.

capncybo, 2022-06-07:
Thanks, I think within the WYZE API it's more like a Walkie Talkie (meaning One at a Time & Not Full Duplex) but...
It might INDEED be useful if we could stream audio from a remote source other than a file.

gtxaspec, 2022-06-07:
not familiar with the api, maybe @mrlt8 has some familiarity with the api and has some knowledge about how the two way audio works?

gtxaspec, 2022-06-07:
included in latest release.

closed 2022-06-07 by gtxaspec

mrlt8, 2022-06-08:
The app reuses the same TUTK AVAPIs that it uses to stream audio/video from the camera, but this time sets up the avserver on the mobile device and uses avSendAudioData to push the audio to the camera.

gtxaspec, 2022-06-08:
@capncybo  see https://github.com/gtxaspec/wz_mini_hacks/issues/68

sibero80, 2023-04-30:
Is there a way to read the values of the `cmd imp_control` properties?