gtxaspec/wz_mini_hacks #69: Outdoor Cam V1/V2 Compatability

gtxaspec, 2022-06-09:
research these, just got a WCO2.

gtxaspec, 2022-06-10:
V2:

command line:
`console=ttyS1,115200n8 mem=80M@0x0 rmem=48M@0x5000000 root=/dev/ram0 rw rdinit=/linuxrc mtdparts=jz_sfc:256K(boot),352K(tag),5M(kernel),5M(rootfs),2720K(recovery),2304K(system),512k(config),16M@0(all) lpj=6955008 quiet`

update rootfs:
```
[sd_update.sh] ROOTFS updateing...
[sd_state_wait.sh] sd_update.sh is running, exit
SystemCall_Dbus_ReadWrite_Thread 520 read socket data failed exit this thread, ret:0 errno:0 (Success)
SystemCall_Dbus_ReadWrite_Thread 521 maybe client is close
[sd_update.sh] copy failed
rmmod: remove 'cywdhd': Device or resource busy
umount: proc busy - remounted read-only
Sent SIGTERM to all processesr
Sent SIGKILL to all processes
Requesting system reboot
[   27.797603] Restarting system.

U-Boot 2013.07 (Nov 14 2021 - 09:40:06)

Board: ISVP (Ingenic XBurst T31 SoC)
DRAM:  128 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 441k for U-Boot at: 83f90000
Reserving 32776k for malloc() at: 81f8e000
Reserving 32 Bytes for Board Info at: 81f8dfe0
Reserving 124 Bytes for Global Data at: 81f8df64
Reserving 128k for boot params() at: 81f6df64
Stack Pointer at: 81f6df48
Now running in RAM - U-Boot at: 83f90000
MMC:   msc: 0
the manufacturer c8
SF: Detected GD25Q128

*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   ====>PHY not found!Jz4775-9161
Hit any key to stop autoboot:  0 
the manufacturer c8
SF: Detected GD25Q128

--->probe spend 4 ms
SF: 2785280 bytes @ 0xa98000 Read: OK
--->read spend 894 ms
Wrong Image Format for bootm command
ERROR: can't get kernel image!
isvp_t31# 
```
boot log:

```
Ver:20201017-Turret
od_cam Build:Mar 15 2022 05:01:34
----====>>>> come into od_cam:555(ms)
sensor name:gc2063
ERROR: serch the USER token failed!
ERROR: serch the USER token failed!
ERROR: serch the USER token failed!
!! The specified ScalingList is not allowed; it will be adjusted!!
!! The specified ScalingList is not allowed; it will be adjusted!!
[frame_pooling_thread--400 Channel:0 ]:585(ms)
[frame_pooling_thread--400 Channel:1 ]:683(ms)
----====>>>> first video frame time:697(ms)
[IMP_Encoder_GetStream--2150 Channel:0 ]:697(ms)
----====>>>> first video sub frame time:847(ms)
[IMP_Encoder_GetStream--2150 Channel:1 ]:847(ms)
open /sys/class/gpio/gpio60/direction error !
open /sys/class/gpio/gpio49/direction error !
----====>>>> first audio frame time:863(ms)
IVS Version:1.0.5 built: Sep  3 2020 14:15:52
ERROR: serch the WIFI token failed!
wakeupFlag : 0
cam_ev_init error:-1
ERROR: serch the USER token failed!
grid_info is not exist!
gridTempArray =ffff
gridTempArray =ffff
gridTempArray =ffff
gridTempArray =ffff
gridTempArray =ffff
gridTempArray =ffff
gridTempArray =ffff
gridTempArray =ffff
gridTempArray =ffff
gridTempFirstLine = 0
gridTempLastLine = 8
gridTempFirstRow = 0
gridTempLastRow = 15
binaryTempright = 0
binaryTempleft = 15
ZRT_POWER_WIFI:ZRT_Get_WIFI_Config error
ZRT_POWER_WIFI:ZRT_HL_Dual_Bind_TCP_sync Start
ZRT_CAM_DAEMON:[DUAL] TCP socket erro
>>>>>>>>>>>BATTERY_USAGE_EVENT_DROP old: 0, start: 1
Setting up swapspace version 1, size = 16773120 bytes
UUID=0df0afb8-3a67-464e-975b-e2a77add239c
write /sys/class/gpio/export error: Device or resource busy
z_cmd_disable_wdt()
CMD: head=c309, index=4012, index_n=bfed, end=55aa
resp:OK

WCO_V2 login: Not This File: /config/profiles/.reconnect.conf
Stream Cipher init time: 4703
[IMP_Encoder_GetStream--2150 Channel:2 ]:5064(ms)
Stream Cipher init success
check_pir: 0
check_time: 1
check_repower: 1
pir_sensitive: 128
mov_sensitive: 128
file_size_avg: 0KB trans_rate_avg: 0KB/S
alarm resolusion: 1080P
----====>>>> get first pir value:5108(ms)
Alarm analysis, moved frame num: 1, threshold num: 7
Alarm analysis not pass!
paracfg user has not inited
od_cam init done.
MCU Event Flag: 0 -> 0
firmware_version:4.48.4.124
hardware_version:0.0.0.2
hardware_ver2:D03F272EB7C9D03F272EB7C9F00A0000
[Real-time alarm] alarm start, get_alarm_video_flag: 0
Sleeping may corrupt here, So add log
_lostBeaconCount_statistics();
_lostBeaconCount_statistics quit
************* camera task: 0 -> 0 *************
notifyWyzeFlag = 0
go_sleep_immediately
mv: can't rename '/tmp/mnt/sdcard/Wyze_camera_log/wyze_camera_2*': No such file or directory
come into mcu check...
mcu version is right
[Real-time alarm] lower: 997, pir_min: 126, pir_max: 132, upper: 3098
[Real-time alarm] moved frame num: 57, threshold num: 7
[Real-time alarm] pir & moved filter pass, start alarm.
cond signa; done
[Real-time alarm]sleeping;quit
[pir_log] pir_up : 0  |  pir_max : 133  |  pir_min : 126 
alarm file(/tmp/alarm.info) is not find
alarm file(/tmp/alarm.info) is not find
sleep,wifi hasn't keep alive
export T31_FORCE_POWER gpio59
killall: zrt_app: no process killed
rmmod: remove 'bcmdhd': No such file or directory
killall: cat: no process killed
killall: logcat: no process killed

************* camera task: 0 -> 0 *************
notifyWyzeFlag = 0
go_sleep_immediately
diff: can't stat '/tmp/mnt/sdcard/sd_update/app.ver': No such file or directory
cat: can't open '/tmp/mnt/sdcard/sd_update/app.ver': No such file or directory
sh: 4.48: unknown operand
cp: can't stat '/tmp/mnt/sdcard/sd_update/.update.info': No such file or directory
mv: can't rename '/tmp/mnt/sdcard/Wyze_camera_log/wyze_camera_2*': No such file or directory
killall: mcu_ver_check.sh: no process killed
killall: update.sh: no process killed
killall: cam_update: no process killed
killall: daemon_wdt: no process killed
killall: telnetd: no process killed
PROBE:z_cmd_lock_exit:190
killall: kvs_stream: no process killed
z_cmd_disable_wdt()
```

gtxaspec, 2022-06-10:
this seems boot via riscv first... unable to extract firmware:
`rootfs_camera.cpio.lzo`

gtxaspec, 2022-06-10:
Able to boot the V3's t31 kernel by corrupting the rootfs update process.  

If the RISC-V updater script detects the `sd_update` dir on the root of the sd card, and a file named `rootfs_camera.cpio.lzo` inside it, it will blindly flash it.  u-boot then complains that the rootfs is broken, and dumps you to a uboot shell prompt.  From here we can load the t31 kernel from the mmc.

gtxaspec, 2022-06-10:
mtd3 is the one we need to extract, I have so far failed to extract it.  from the update.zip, `rootfs_camera.cpio.lzo` is flashed directly to mtd3.
[roofs.zip](https://github.com/gtxaspec/wz_mini_hacks/files/8876856/roofs.zip)

LouDnl, 2022-07-31:
I have a camera with the same Ingenic T31 soc that seems to be based on the Ingenic Zeratul development kit. Unfortunatly I am in the same position as you when trying to unpack the rootfs. I have dumped the Serial Flash chip and can get the squashfs and jffs2 images, but the rootfs is somehow compressed and I cannot unpack it.

gtxaspec, 2022-08-02:
@LouDnl not yet, haven't found the docs or sdk.  Most of the information is here:

https://caibiao-lee.blog.csdn.net/?type=blog

LouDnl, 2022-08-02:
Do you have discord?

LouDnl, 2022-08-02:
> @LouDnl not yet, haven't found the docs or sdk. Most of the information is here:
> 
> https://caibiao-lee.blog.csdn.net/?type=blog

I have found this indeed, CSDN doesn't allow reading. Here's readable links:
BLOG:
君正T31硬件设计手册-爱代码爱编程 
https://icode.best/i/41780336027339

君正Zeratul开发(0)——序言及目录-爱代码爱编程 
https://icode.best/i/19347139642873

君正Zeratul开发(1)——分区启动分析-爱代码爱编程 
https://icode.best/i/19349839198032

君正Zeratul开发(2)——uboot启动分析-爱代码爱编程 
https://icode.best/i/19351039601500

君正Zeratul开发(3)——升级回滚-爱代码爱编程 
https://icode.best/i/19352439643973

君正Zeratul开发(4)——图像效果调试-爱代码爱编程 
https://icode.best/i/19353339647262

君正Zeratul开发(5)——快速启动优化-爱代码爱编程 
https://icode.best/i/19354339660886

Translated:
Junzheng T31 Hardware Design Manual - Love Code Love Programming
https://icode.best/i/41780336027339

Junzheng Zeratul Development (0) - Preface and Contents - Love Code Love Programming
https://icode.best/i/19347139642873

Junzheng Zeratul development (1) - partition startup analysis - Programmer Sought
https://icode.best/i/19349839198032

Junzheng Zeratul development (2) - uboot startup analysis - Programmer Sought
https://icode.best/i/19351039601500

Junzheng Zeratul development (3) - upgrade and rollback - Programmer Sought
https://icode.best/i/19352439643973

Junzheng Zeratul development (4) - image effect debugging - Programmer Sought
https://icode.best/i/19353339647262

Junzheng Zeratul development (5) - quick start optimization - Programmer Sought
https://icode.best/i/19354339660886

LouDnl, 2022-08-02:
I cracked the protection, send me a message on Discord if you're interested -> LouD#4930
[image]

monkollek, 2023-03-10:
@gtxaspec you probably already sorted this out with @LouDnl but for prosterity and those who are interested but don't have time to hang out on discord 😉

The rootfs_camera.cpio.lzo file cannot be simply decompressed using lzop because someone was tricksy and changed the magic number.

```
hexyl rootfs_camera.cpio.lzo | head -10
┌────────┬─────────────────────────┬─────────────────────────┬────────┬────────┐
│00000000│ e8 7b 4c 00 00 0d 0a 1a ┊ 0a 10 40 20 a0 09 40 03 │×{L00__•┊_•@ ×_@•│
│00000010│ 09 03 00 00 01 00 00 81 ┊ a4 62 30 1d f6 00 00 00 │_•00•00×┊×b0•×000│
```

The standard magic number for lzo header is 
```
static const unsigned char lzop_magic[9] =
    { 0x89, 0x4c, 0x5a, 0x4f, 0x00, 0x0d, 0x0a, 0x1a, 0x0a };
```
taken from: https://github.com/mirror/lzop/blob/master/src/lzop.c#L622

So if you change the magic number either in the code or the file itself to match then you should be golden.
It currently is: `e8 7b 4c 00 00 0d 0a 1a 0a` and what is expected is `89 4c 5a 4f 00 0d 0a 1a 0a`

```
lzop -x rootfs_camera.cpio.lzo
cpio -idv < rootfs_camera.cpio

ls -lh
total 17M
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 bin
drwxr-xr-x 3 ubuntu ubuntu 4.0K Mar  9 23:39 config
drwxr-xr-x 3 ubuntu ubuntu 4.0K Mar  9 23:39 config_bak
drwxr-xr-x 4 ubuntu ubuntu 4.0K Mar  9 23:39 dev
drwxr-xr-x 4 ubuntu ubuntu 4.0K Mar  9 23:39 etc
drwxr-xr-x 4 ubuntu ubuntu 4.0K Mar  9 23:39 lib
lrwxrwxrwx 1 ubuntu ubuntu   11 Mar  9 23:39 linuxrc -> bin/busybox
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 mnt
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 proc
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 root
-rw-r--r-- 1 ubuntu ubuntu  12M Mar 15  2022 rootfs_camera.cpio
-rw-r--r-- 1 ubuntu ubuntu 4.8M Mar  9 18:39 rootfs_camera.cpio.lzo
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 run
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 sbin
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 sys
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 system
drwxr-xr-x 2 ubuntu ubuntu 4.0K Mar  9 23:39 tmp
drwxr-xr-x 5 ubuntu ubuntu 4.0K Mar  9 23:39 usr
drwxr-xr-x 3 ubuntu ubuntu 4.0K Mar  9 23:39 var
```
**Note** you'll have to remember to put back in the **custom magic number** when packaging something back up.

LouDnl, 2023-03-10:
> @gtxaspec you probably already sorted this out with @LouDnl but for prosterity and those who are interested but don't have time to hang out on discord 😉
>
> The rootfs_camera.cpio.lzo file cannot be [...]

This is indeed correct :) Don't forget to use the proper lzop version.
Here is my unpack pack routine:
```shell
# change UBIA LZO header to LZO header
echo -ne \\x89\\x4C\\x5A\\x4F | dd conv=notrunc count=1 of=rootfs_camera.cpio.lzo 
# unpack
mkdir /tmp/ramdisk
cd /tmp/ramdisk
sudo cpio -idmv < rootfs_camera.cpio
# do what ever you changes want to do
# repack
find . | cpio -o -H newc > /tmp/rootfs_camera.cpio  # pack the main filesystem
cd /tmp
lzop -9 -o rootfs_camera.cpio.lzo rootfs_camera.cpio  # repack with lzop 1.03
#  change LZO header to UBIA LZO header
echo -ne \\xC8\\xA0\\x27\\x00 | dd conv=notrunc count=1 of=rootfs_camera.cpio.lzo 
```

monkollek, 2023-03-10:
> Able to boot the V3's t31 kernel by corrupting the rootfs update process.
> 
> If the RISC-V updater script detects the `sd_update` dir on the root of the sd card, and a file named `rootfs_camera.cpio.lzo` inside it, it will blindly flash it. upon reboot, u-boot complains that the rootfs is broken, and dumps you to a uboot shell prompt. From here we can load the t31 kernel from the mmc.

@gtxaspec @LouDnl what extra work have you done since on this? I just got the WCO v2 and wanna start playing with it. I don't quite get the next steps that needs to happen that's detailed above. Many thanks!

LouDnl, 2023-03-10:
> @gtxaspec @LouDnl what extra work have you done since on this? I just got the WCO v2 and wanna start playing with it. I don't quite get the next steps that needs to happen that's detailed above. [...]

I have a different camera with the same soc, so I have done nothing for the WCO v2. The camera I have is some white label Tuya PTZ thing. The wifi chip died on mine so I tossed it in a corner for now.

gtxaspec, 2024-11-05:
zeratul platform not supported, planned future support in thingino.

closed 2024-11-05 by gtxaspec