gtxaspec/wz_mini_hacks #70: Wyze Alarm Question/Suggestion

capncybo, 2022-06-09:
So my Wife informed me Last-night we had some (sketchy people) outside our house attempting to open our front-door.
She was (reluctant/scared) to make her presence known & wanted to know...
-Could she have triggered an alarm/siren?
Obviously inside the WYZE application... It can be done.
However IMO there are too many steps because the feature is not directly accessible & therefore takes too long to trigger.
You Have to wait, for the app to load... Then click on the camera... wait for the Video to stream... Click on the... ...More & finally Siren.
"Ridiculous" 

But I already have a Raspberry Pi of which I use for home automation & was going to simply use:
**ssh root@192.168.50.22 "cmd aplay /opt/wz_mini/usr/share/audio/swap.wav 90"**
But hearing "Initializing Swap" Just isn't very intimidating LOL (Only Joking), of course I was going to use a proper "siren.wav" file...
However I thought I'd ask, maybe you have already (ripped/created) the siren sound already used as the WYZE-Siren.
If not, perhaps adding a siren.wav into: **/opt/wz_mini/usr/share/audio/** Might make some-sense. (Just an idea/suggestion)

Regardless, that leads to the other question.
Perhaps there's an (easier/better) method to trigger a WYZE Cam's Siren via scripting?
I already looked among the options of... cmd <arg>
Nothing siren related... Yet???

Hint-Hint
LOL

gtxaspec, 2022-06-09:
the siren wav file is built in to the camera's filesystem, check in the directory `/system/local`

capncybo, 2022-06-09:
> 

Well now... that's embarrassing, how the heck did I miss that.  Thanks
In the meantime... I had found a 30-second police siren.wav file & uploaded it via scp