gtxaspec/wz_mini_hacks #71: Wyze Cam Pan v1 support

mwhdc, 2022-06-09:
Support original Pan camera if possible. Please label as enhancement.

I own 3 but unfortunately all are on current stock firmware (4.10.8.1002) which AFAIK doesn't have a working hack to get a root shell. @gtxaspec would dmesg output from Xiaomi-Dafang-Hacks custom firmware be helpful? I believe it's still usable on the older cams. It requires flashing a modified bootloader and I forget which distro it's based on.

gtxaspec, 2022-06-09:
@mwhdc, I can create a modified stock firmware for the pan v1 with telnet enabled.  you can flash it, telnet in and grab some logs, if you are up to testing it

gtxaspec, 2022-06-09:
[demonew.bin.zip](https://github.com/gtxaspec/wz_mini_hacks/files/8874531/demonew.bin.zip)
you can flash this, just rename the bin to demo.bin.  root password is blank, and telnet should be enabled once network is up on the camera... Since i don't have a panv1 to test, if something goes wrong you can always flash the stock demo.bin and all will be well again.

mwhdc, 2022-06-10:
@gtxaspec Just flashed and the camera did not like that file... Shutter clicked only once on boot, solid yellow light, not responding to pings.

gtxaspec, 2022-06-10:
hmm... strange.    The next thing I can think of is, try to flash the wz_mini demo.bin that is for the V2, and see if that boots...

mwhdc, 2022-06-10:
> 

```
dev:    size   erasesize  name
mtd0: 00040000 00008000 "boot"
mtd1: 00200000 00008000 "kernel"
mtd2: 00350000 00008000 "root"
mtd3: 000a0000 00008000 "driver"
mtd4: 004a0000 00008000 "appfs"
mtd5: 00200000 00008000 "backupk"
mtd6: 000a0000 00008000 "backupd"
mtd7: 00200000 00008000 "backupa"
mtd8: 00040000 00008000 "config"
mtd9: 00040000 00008000 "para"
mtd10: 00010000 00008000 "flag"
```

gtxaspec, 2022-06-10:
couple more if you can:

`cat /etc/init.d/rcS`
`cat /system/init/app_init.sh`

mwhdc, 2022-06-10:
Looks like everything in app_init.sh is from WyzeHacks.

```
[root@WyzeCamPAN-32B4:~]# cat /etc/init.d/rcS
#!/bin/sh

# Set mdev
echo /sbin/mdev > /proc/sys/kernel/hotplug
/sbin/mdev -s && echo "mdev is ok......"

# create console and null node for nfsroot
#mknod -m 600 /dev/console c 5 1
#mknod -m 666 /dev/null c 1 3

# Set Global Environment
export PATH=/bin:/sbin:/usr/bin:/usr/sbin
export PATH=/system/bin:$PATH
export LD_LIBRARY_PATH=/system/lib
export LD_LIBRARY_PATH=/thirdlib:$LD_LIBRARY_PATH

# networking
ifconfig lo up
#ifconfig eth0 192.168.1.80

# Start telnet daemon
telnetd &

# Set the system time from the hardware clock
#hwclock -s

#set the GPIO PC13 to high, make the USB Disk can be use
cd /sys/class/gpio
echo 77 > export       #申请GPIO
cd gpio77
echo out > direction   #设置为输出模式
echo 0 > active_low    #value是0,表示低电平。value是1,表示高电平
echo 1 > value         #设置电平（输出模式）

# Mount driver partition
mount -t squashfs /dev/mtdblock3 /driver

# Mount system partition
mount -t jffs2 /dev/mtdblock4 /system

# Mount backup partition
#mount -t jffs2 /dev/mtdblock5 /backupk

# Mount backup partition
#mount -t jffs2 /dev/mtdblock6 /backupd

# Mount backup partition
mount -t jffs2 /dev/mtdblock7 /backupa

# Mount configs partition
mount -t jffs2 /dev/mtdblock8 /configs

# Mount params partition
mount -t jffs2 /dev/mtdblock9 /params

# Format system patition if it is invalid
if [ ! -f /system/.system ]; then
    echo "Format system partition..."
    umount -f /system
    flash_eraseall /dev/mtd4
    mount -t jffs2 /dev/mtdblock4 /system
    cd /system
    mkdir -p bin init etc/sensor lib/firmware lib/modules
    echo "#!/bin/sh" > init/app_init.sh
    chmod 755 init/app_init.sh
    touch .system
    cd /
    echo "Done"
fi

# Run init script
if [ -f /system/init/app_init.sh ]; then
    /system/init/app_init.sh &
fi

[root@WyzeCamPAN-32B4:~]# cat /system/init/app_init.sh
#!/bin/sh
export THIS_BIN=$(readlink -f $0)
export THIS_DIR=$(dirname $THIS_BIN)
export THIS_MD5=`md5sum $THIS_BIN | grep -oE "^[0-9a-f]*"`
export THIS_VER=0.5.08

ACTION=${1:-run}
[ "$#" -gt 1 ] && shift

WYZEHACK_DIR=${TMP:-/tmp}/wyze_hack/$ACTION
mkdir -p $WYZEHACK_DIR

PAYLOAD_START=`awk '/^#__PAYLOAD_BELOW__/ {print NR + 1; exit 0; }' $0`
tail -n+$PAYLOAD_START $0 | gzip -cd | tar x -f - -C $WYZEHACK_DIR

exec sh $WYZEHACK_DIR/main.sh $ACTION $@
#__PAYLOAD_BELOW__
[...]
```

gtxaspec, 2022-06-10:
i think its /system/init/app_init.orig...or something like that, check the dir

mwhdc, 2022-06-10:
```
#!/bin/sh

cd /system
if [ -f /system/.upgrade ]; then
    cd /backupa
    echo "init upgrading!!!!!!!!!!!!"
    ./upgrade.sh
    rm /system/.upgrade
fi

echo 61 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio61/direction
echo 1 > /sys/class/gpio/gpio61/value
sleep 1
echo 0 > /sys/class/gpio/gpio61/value

#insmod /lib/modules/tx-isp.ko isp_clk=100000000
#insmod /lib/modules/sensor_imx323.ko
#insmod /lib/modules/sensor_jxf22.ko
#insmod /lib/modules/sensor_ps5230.ko
#insmod /lib/modules/exfat.ko
#insmod /lib/modules/sample_motor.ko
#insmod /lib/modules/audio.ko
#insmod /lib/modules/sinfo.ko
#insmod /lib/modules/8189es.ko

insmod /driver/tx-isp.ko isp_clk=100000000
#insmod /driver/sensor_imx323.ko
#insmod /driver/sensor_jxf22.ko
#insmod /driver/sensor_ps5230.ko
insmod /driver/exfat.ko
insmod /driver/sample_motor.ko
insmod /driver/audio.ko
insmod /driver/sinfo.ko
insmod /driver/8189es.ko
insmod /driver/sample_pwm_core.ko
insmod /driver/sample_pwm_hal.ko
#insmod /driver/rtl8189ftv.ko

#wpa_supplicant -Dwext -i wlan0 -c /system/etc/wpa_supplicant.conf -B
#udhcpc -i wlan0 -s /system/etc/udhcpc.script -q

#ifconfig eth0 up
#udhcpc -i eth0 -s /system/etc/udhcpc.script -q
#ifconfig eth0 10.0.22.199
#route add default gw 10.0.22.1

# open ircut
#cp /system/bin/setir /tmp/
#config ip address

#/system/bin/carrier-server --st=imx322
/system/bin/singleBoadTest
/system/bin/iCamera &
```

gtxaspec, 2022-06-10:
great thanks i'll start working on support soon!

gtxaspec, 2022-06-25:
@mwhdc, If you can, try flashing the latest release on the PanV1.  See if it boots.  If it doesn't, can you see if there is anything in the `/opt/wz_mini/log/` directory?

thanks!

mwhdc, 2022-06-25:
> @mwhdc, If you can, try flashing the latest release's demo.bin on the PanV1. See if it boots. If it doesn't, can you see if there is anything in the `/opt/wz_mini/log/` directory?

Pan v1 doesn't seem to recognize the latest demo.bin as a flashable file. It ignores the reset button and does a standard boot. I'll try adding your v2_kernel to the latest stock Pan demo.bin (vs the v2's) in case there's a difference.

gtxaspec, 2022-06-28:
i've ordered a PanV1 (i hope it's a v1) so hopefully I can get this working quickly

gtxaspec, 2022-06-28:
the latest release includes an updated compile_image.sh script.

try `./compile_image.sh pan` to create a demo.bin for the pan.  it works on my test unit.

mwhdc, 2022-06-29:
> try `./compile_image.sh pan` to create a demo.bin for the pan. it works on my test unit.

Bonehead mistake on my part, I thought the Pan v1 had a white LED in front like the v2 cube that would indicate when the firmware was being updated. It doesn't, the orange just stays lit during flashing and boot, then blinks while the camera tries to bring up a network connection. So I've been able to flash your kernel successfully on two Pan v1s now.

Current issue is that both Pans seem to have forgotten their Wi-Fi credentials. Going back through network setup in the Wyze app hasn't helped as the cameras claim they can't find the specified (2.4) network. I'm assuming this is more PEBCAK of mine - I'll tinker further and report back.

gtxaspec, 2022-06-29:
if you ever upgraded beyond 1002 firmware, you need to add them back to wifi, they encrypt the ssid and password now in the configuration files.

mwhdc, 2022-06-29:
> if you ever upgraded beyond 1002 firmware, you need to add them back to wifi, they encrypt the ssid and password now in the configuration files.

I've tried a few times on both cams and still no luck. Is it possible that the right credentials are in place but the new Wi-Fi drivers aren't loading? Getting endless loops of this in the log...

```
+ grep 'inet addr'
+ ifconfig wlan0
+ grep 'inet addr'
+ ifconfig wlan0
+ echo ' wlan0 not ready yet...'
 wlan0 not ready yet...
+ sleep 5
+ true
```

gtxaspec, 2022-06-29:
when you push the setup button, do you hear the ready to connect prompt?

gtxaspec, 2022-06-29:
I don't have the new wifi-drivers in place for the panv1 yet, so wifi drivers shouldn't be an issue...

gtxaspec, 2022-06-29:
can you share the wz_init, wz_post, and dmesg logs?

mwhdc, 2022-06-29:
> when you push the setup button, do you hear the ready to connect prompt?

Yes, and the cams are reading the QR code, but then claiming they can't find the SSID.

I had the new drivers enabled in wz_mini.conf, but disabling them doesn't seem to make a difference.

Logs are attached from a boot with drivers disabled. There wasn't a dmesg.log on the card.

[wz_init.log](https://github.com/gtxaspec/wz_mini_hacks/files/9006577/wz_init.log)
[wz_post.log](https://github.com/gtxaspec/wz_mini_hacks/files/9006578/wz_post.log)

gtxaspec, 2022-06-29:
it seems I can't connect this wyze pan to my wifi, even flashing back to stock.  It refuses to connect to my wifi router... have you tried to flash your back to stock?

I'm not sure if its because I accidentally deleted a .product_config file on mine though

gtxaspec, 2022-06-29:
It works when I configure it manually, flashing stock again...

gtxaspec, 2022-06-29:
looks like the stock wifi driver does not like our kernel.  I will update the release to include a new wifi driver which does work with the pan.

gtxaspec, 2022-06-29:
latest release includes a new v2_kernel.bin and SD_ROOT including new modules.

mwhdc, 2022-06-29:
Success! Had to re-add the cam one last time after first wz_mini boot, now online and working well.

By the way, I've been making one change to fw_tool.sh after each revision to allow it to run under darwin - "stat -c %s" changes to "stat -f %z". If you add a uname -s to fork that command, the script will work for Mac users. (Homebrew install of u-boot-tools and any dependencies is also required.)

gtxaspec, 2022-07-02:
fixed in https://github.com/gtxaspec/wz_mini_hacks/commit/f4245747f3ff597b1ba94bb0f87be93e887637a1
thanks for the testing @mwhdc

closed 2022-07-02 by gtxaspec

mwhdc, 2022-07-02:
> fixed in [f424574](https://github.com/gtxaspec/wz_mini_hacks/commit/f4245747f3ff597b1ba94bb0f87be93e887637a1) thanks for the testing @mwhdc

Thanks for buying the outdated device to debug! Don't forget to add Pan v1 to README.md.