gtxaspec/wz_mini_hacks #72: Wishlist:  Ethernet/Wifi Bonding driver

skilau, 2022-06-10:
Wishlist item, definitely not a bug!

Would love to see the bonding driver brought over, and add support for fallback/failover with Ethernet -> Wifi.

https://wiki.debian.org/Bonding

Specifically Example 2:

Configuration - Example 2 ("Laptop-Mode")

Tie cable and wireless network interfaces (RJ45/WLAN) together to define a single, virtual (i.e. bonding) network interface (e.g. bond0).

As long as the network cable is connected, its interface (e.g. eth0) is used for the network traffic. If you pull the RJ45-plug, ifenslave switches over to the wireless interface (e.g. wlan0) transparently, without any loss of network packages.

After reconnecting the network cable, ifenslave switches back to eth0 ("failover mode").

From the outside (=network) view it doesn't matter which interface is active. The bonding device presents its own software-defined (i.e. virtual) MAC address, different from the hardware defined MACs of eth0 or wlan0.

The dhcp server will use this MAC to assign an ip address to the bond0 device. So the computer has one unique ip address under which it can be identified. Without bonding each interface would have its own ip address.

skilau, 2022-06-10:
I see our kernel source do include the "drivers/net/bonding" directory/driver.
I may try compiling it up this weekend, bringing it over and loading it, just to see if it has any chance of working.

gtxaspec, 2022-06-10:
i can compile the `bonding` driver as a module for you to test, what device are you using? v2/v3?

gtxaspec, 2022-06-10:
[bonding.ko.zip](https://github.com/gtxaspec/wz_mini_hacks/files/8881825/bonding.ko.zip)
for v3

skilau, 2022-06-10:
Yup, v3.  Thanks!
I will muck with it this weekend for a bit, if I get a chance.
PS: I spent some time setting up atomcam_tools, and was able to successfully generate a kernel and modules based on your kernel.config file.  Thanks for having the config in your repo!

skilau, 2022-06-15:
Hi!  I submitted a pull request on this change, however, it isn't fully complete, but I am a bit stuck on the Wifi side of things.
It is properly set up, and does the right thing for Ethernet, but I do not know how wpa_supplicant works well enough, to feed the wpa info into the 2nd slave (wlanold).
Maybe someone else might know a bit more about how to do it, and finish the patch?

skilau, 2022-06-21:
The #89 pull request was merged, and thus this request was been resolved!
Closing out!
Thanks again!

closed 2022-06-21 by skilau