gtxaspec/wz_mini_hacks #78: don't know if it's useful or not but one of my cameras picked up a partial OTA update from wyze

virmaior, 2022-06-11:
4.36.2.5 on to /.ota/ on the SD card
[1e26fd4b.gz](https://github.com/gtxaspec/wz_mini_hacks/files/8883440/1e26fd4b.gz)

presumably the prevent upgrades clobbered the process before it could complete the download or install it.

gtxaspec, 2022-06-11:
.ota stuff is related to the stuff sinker downloads, like the webrtc server for live view on the web, and also the AI detection stuff.   These programs are updated frequently from wyze, and are not stored on the camera or present in the firmware, they are always downloaded via OTA download.  Not a firmware update, just programs.

gtxaspec, 2022-06-11:
need to figure out what host they download this stuff from to allow user to blcok it

gtxaspec, 2022-06-11:
@virmaior trying to get an atomcam2 to the states, wondering if you would be willing help get one to me...

virmaior, 2022-06-11:
@gtxaspec can do. The main question is whether my wife will be pissed if I also buy one for myself. It will have two step in terms of shipping but that can't really be helped.

virmaior, 2022-06-15:
@gtxaspec cameras have been ordered. They should arrive to me in about 2-3 days.

gtxaspec, 2022-06-17:
awesome! how can I contact you off github? email, telegram,etc?

gtxaspec, 2022-06-21:
ok, so it looks like `sinker` tried to download to `/media/mmc/.ota`, and if it can't download here, then it uses `/tmp/webrtc`
So to block kvsWebrtcClientMaster we have to block both of these paths.

gtxaspec, 2023-09-26:
need webrtc for streaming, wontfix for now.

closed 2023-09-26 by gtxaspec