gtxaspec/wz_mini_hacks #79: Running rtsp on V3.

rani-pinchuk, 2022-06-11:
This is a splendid project. Thanks a lot for doing this!

I was trying to get the rtsp to run, but this did not work out of the box. 

Looking at the code, I realized that `/opt/wz_mini/etc/init.d/wz_user.sh` actually tries to start `v4l2rtspserver` but looks for it in `/media/mmc/wz_mini/bin/v4l2rtspserver`. However, when I SSH to the V3, inside `/media/mmc/` I do not have `wz_mini`. It can be found in `/opt/wz_mini/bin/v4l2rtspserver`. 

And indeed, changing in `wz_user.sh` the line:
```
LD_LIBRARY_PATH=/media/mmc/wz_mini/lib /media/mmc/wz_mini/bin/v4l2rtspserver $AUDIO_CH $AUDIO_FMT -U $RTSP_LOGIN:$RTSP_PASSWORD -P $RTSP_PORT $DEVICE1 $DEVICE2 &
```
to:
```
LD_LIBRARY_PATH=/opt/wz_mini/lib /opt/wz_mini/bin/v4l2rtspserver $AUDIO_CH $AUDIO_FMT -U $RTSP_LOGIN:$RTSP_PASSWORD -P $RTSP_PORT $DEVICE1 $DEVICE2 &
```
fixed it, and now rtsp is running on V3.

I have tested it with two cams, one is totally new. Note that I had to upgrade it first to the latest  firmware (4.36.3.18).

gtxaspec, 2022-06-11:
the sd card is actually mounted in two places.  wz_mini mounts the sd card at /opt during boot inside the initramfs.  

the camera, mounts the memory card at /media/mmc, just like it would if there was no wz_mini... So if it's not mounting /media/mmc then something is wrong.  Is your memory card FAT32, DOS partition mapped?

claudobahn, 2022-06-11:
@gtxaspec I have run into `/media/mmc` not getting mounted, as well. When that happened, `/opt/wz_mini/log/wz_user.log` had endless entries saying
```
+ first_run_check
+ '[[' -e /opt/wz_mini/tmp/.run_mmc_firstrun ]]
+ wait_sdroot
+ true
+ '[[' -d /media/mmc/wz_mini ]]
+ '[[' -d /media/mmcblk0p1/wz_mini ]]
+ echo 'sdcard not ready yet...'
sdcard not ready yet...
+ sleep 5
+ true
+ '[[' -d /media/mmc/wz_mini ]]
+ '[[' -d /media/mmcblk0p1/wz_mini ]]
+ echo 'sdcard not ready yet...'
sdcard not ready yet...
+ sleep 5
```

This was using firmware 4.36.9.139 and a recent version of wz_mini_hacks (maybe a week ago?) on a freshly FAT32 formatted sd card. Eventually it started working, after a few reboots and I also went back to an older version of wz_mini_hacks and then used the upgrade script. Not sure what got it working in the end.

All that said, I currently have 5x V3 cameras all running stock 4.36.9.139 firmware (unmodified) w/ wz_mini_hacks streaming RTSP

gtxaspec, 2022-06-11:
yes I've also experienced that, both with wz_mini and on the stock firmware.  

Seems to be a bug in iCamera.  unfortunately there doesn't seem to be a way to fix it, I can't force mount /media/mmc, iCamera doesn't like that.  I'll do someore research soon.

gtxaspec, 2022-06-12:
I've added a log gathering script to wz_mini, which gathers useful system data covering the audio, video, mmc subsystems to make it easier for users to share logs to debug... coming soon.

gtxaspec, 2022-06-12:
I've eliminated the wait for the sd card to mount, and changed the paths to ignore /media/mmc.  Hopefully that will solve some issues! Latest release out now.

closed 2022-06-17 by gtxaspec