gtxaspec/wz_mini_hacks #84: Update readme to link to wz_mini_utils, wz_mini_debian?

tensiondriven, 2022-06-14:
The readme mentions two related projects, wz_mini_debian and wz_mini_utils, however I can't find these repos anywhere on the internet, either by searching Google or looking through other repos on the accounts of contributors to wz_mini_hacks - Are these real projects?  If so, I'd love to see the readme updated with links.

### A little more context

(actually, a separate issue, i guess)

As I work with wyze_mini_hacks, I want to find out what tools are available to me.   The wyze_mini_hacks repo and docs get me up and running fast, but it isn't clear what resources/binaries are available to play with.  I saw that wz_mini_hacks leverages atomcam_tools, which looks to have all the binaries, but the documentation is in Japanese and makes it almost useless for me to explore.  A translation of the https://github.com/gtxaspec/atomcam_tools docs, perhaps in the wz_mini_hacks repo, would go a long way to helping people like me get more out of the project.

gtxaspec, 2022-06-14:
i merged wz_mini_utils into the wz_mini_hacks, so everything you see in SD_CARD/wz_mini/bin are the utilities.  

wz_mini_debian is debian in a chroot, it works, though its really, slow since everything is run on the camera.  I'll add this repo shortly in case you still want to give it a spin.   There are some things I need to update so it works ok in order to publish it.

As far as the atomcam_tools, I used google translate to translate the website, I haven't translated anything from the repo yet, it's just a mirrored fork for now.  

I recommend using google translate on the atomcam_tools docs, and get the docker image up and running, from here you can explore some of the utilities and inner workings, it includes the toolchain and kernel, so you can also play with that... though it's not wz_mini exactly.

Another tool I leverage for this project is buildroot.  I'll update the documentation soon on how I set everything up.

gtxaspec, 2022-06-14:
updated https://github.com/gtxaspec/wz_mini_hacks/tree/master/documentation

gtxaspec, 2022-06-14:
also, the SDK is worth a look too https://github.com/cgrrty/Ingenic-SDK-T31-1.1.1-20200508

skilau, 2022-06-14:
Yeah, I had to google translate atomcam_tools as well. 
But google did translate it well, and I was able to get the Docker env up and running just fine.

tensiondriven, 2022-06-15:
>  updated https://github.com/gtxaspec/wz_mini_hacks/tree/master/documentation

This is amazing, a huge step forward.  Thanks for taking the time, I think it'll really benefit others (like me!)

> ...  I'll add this repo shortly in case you still want to give it a spin. 

I don't need this repo, I mainly want to resolve the ambiguity in the readme about what these repos are. What would satisfy this ticket for me is if the readme was updated with a little more context for that section, or if the section was simply removed.  If the repos aren't accessible anywhere, simply removing references to them would solve the ambiguity.

You could preserve the info about debian/chroot with an addition like:

~Note: There is a build of this project that includes a debian chroot install - if you'd like to see it, please open a ticket~

Since wz_mini_utils doesn't seem to exist anywhere on the public internet, I think it can be removed entirely.  I don't see any value in referencing it in the docs and it only creates more confusion.

Actually, thinking through it a little more, since you added the docs, I think that the references to Related Projects could be removed entirely.  The docs can be improved over time and more added there.

If at some point I reach a sufficient level of blood caffeine saturation, I'd love to contribute by:

1) Updating the readme & docs you created with more formatting
2) Google Translate the athom_tools and provide that to this project in some form, either as a link to a repo this project could reference, or additional documentation that could be merged into this project's documentation.

TLDR; This ticket would be solved for me if the readme was updated to remove the Related Projects section, which would remove the confusion I faced that caused me to create the ticket.

gtxaspec, 2022-06-17:
updated the readme

closed 2022-06-17 by tensiondriven